zigpy/zha-device-handlers #3196: [BUG] TZE204_yjjdcqsq TS0601 (sensor for temperature and huminity, using ts0601_sensor.py): firmware update UNKNOWN

HelgeGitHub, 2024-06-09:
### Bug description

update.tze204_yjjdcqsq_ts0601_firmware is always "unknown"
No firmware information and update possibilities are available.

### Steps to reproduce

1. Go to Developer Tools --> States --> update.tze204_yjjdcqsq_ts0601_firmware

### Expected behavior

There should be the normal update handling for HA

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_yjjdcqsq",
  "model": "TS0601",
  "class": "ts0601_sensor.TuyaTempHumiditySensorVar04"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.6.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
[... 369 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

giggio, 2024-11-26:
I'm also getting unknown on the battery sensor of the device.

[image]

alerton123, 2024-12-03:
me too, no battery state or firmware  (both unknown)