zigpy/zha-device-handlers #3200: [Device Support Request] Niko Zigbee Connected RollerShade Control   552-72301

nikplas, 2024-06-12:
### Problem description

I got this Niko zigbee motor control , 552-723, (https://www.niko.eu/en/products/niko-home-control/products-for-traditional-wiring/connected-motor-control-3-a-zigbee-productmodel-niko-7a97247a-5e33-5eaf-b21e-893ba9f15378) ; in order to operate rollershades; 
I got HAS with latest version 24.6.2 and ZHA skyconnect; Device easily installed, but there is no support for calibration, or how to set the up/down limit in seconds; This device comes preinstalled with a 90sec up/down running time, wich is practically too long ; The visualisation of the state is niet correct on HA, when the roller is up HA shows state down en vice versa; Support for invertion also not possible;

Operation with the buttons on device zelf is correct. When left button is pressed it goes up en when right button is pressed it goes down;

### Solution description

please advice on how to gain this extra funcionality, invertion om cover state and setting up/down time limit
i am novice with ZHA and cluster settings but would like to try it

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4703, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0102",
        "0x0b05",
        "0xfc00",
        "0xfc01"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "NIKO NV",
  "model": "Connectable motor control,3A",
  "class": "zigpy.device.Device"
]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
[... 592 lines cut ...]
  }
}]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

nikplas, 2024-06-12:
i also see the following lines in the configuration details, i assume that 900 is in ms the preset running time of the motor 90sec, does someboby know how to edit this?

WindowCovering    Attribute         min/max/change  
current_position_lift_percentage:  0/900/1
current_position_tilt_percentage:  0/900/1

JFMous, 2024-06-14:
I just purchased the same control and facing the same issues.
I've already been able to figure out that the opening/closing time in seconds are zigbee attributes 0xfcc1 (opening seconds) and 0xfcc2 (closing seconds) in cluster 0x0102 (window_covering).
But the reversed reporting for the open/closed state is really annoying.

nikplas, 2024-06-15:
> I just purchased the same control and facing the same issues. I've already been able to figure out that the opening/closing time in seconds are zigbee attributes 0xfcc1 (opening seconds) and 0xfcc2 [...]

JFMous, super, thnks for your feedback, i will check and replace the closing/opening attributes so as you describe

nikplas, 2024-06-15:
[image]
[image]

how did you manage to get those 0xfcc1 and 0xfcc2 attributes? i dont see these 2 in my window_covering cluster, see also screenshot

JFMous, 2024-06-16:
> how did you manage to get those 0xfcc1 and 0xfcc2 attributes? i dont see these 2 in my window_covering cluster, see also screenshot

I used [ZHA toolkit](https://github.com/mdeweerd/zha-toolkit). After you've installed the toolkit, you can call the service as follows:
```
service: zha_toolkit.attr_write
data:
  ieee: '60:b6:47:ff:fe:07:8a:5d' #replace with IEEE of NIKO motor control
  cluster: 0x0102
  attribute: 0xfcc1
  manf: 4703
  attr_val: 26 #seconds to lift
```

Just replace the ieee with the address of your motor controller and the attr_val with the time in seconds to fully open the roller shade. Then change attribute to 0xfcc2 and attr_val to the time it takes to close the shade (in my case that was 2 seconds shorter, because gravity helps a bit when closing) and issue the command again.

nikplas, 2024-06-16:
> > how did you manage to get those 0xfcc1 and 0xfcc2 attributes? i dont see these 2 in my window_covering cluster, see also screenshot
> 
> I used [ZHA toolkit](https://github.com/mdeweerd/zha-toolkit). After you've installed the toolkit, you can call the service as follows from the Development tools >> services (YAML mode):
> 
> ```
> service: zha_toolkit.attr_write
> data:
>   ieee: '60:b6:47:ff:fe:07:8a:5d' #replace with IEEE of NIKO motor control
>   cluster: 0x0102
>   attribute: 0xfcc1
>   manf: 4703
>   attr_val: 26 #seconds to lift
> ```
> 
> Just replace the ieee with the address of your motor controller and the attr_val with the time in seconds to fully open the roller shade. Then change attribute to 0xfcc2 and attr_val to the time it takes to close the shade (in my case that was 2 seconds shorter, because gravity helps a bit when closing) and issue the command again.

JFMous, thank you very much, I did exactly what you described and it worked!
I have now installed the correct opening and closing times.

if we can still figure out, how to fix the reversed repported roller state on HA, that would be perfect.
Maybe someone who uses the Niko Home Control and their zigbee hub can tell us if we still miss  more functionality, for example the feedback leds maybe activate/deactivate

nikplas, 2024-10-08:
still not 100% compayible with ZHA

tados000, 2024-12-12:
I have de same problem. You only use zigbee2mqtt or have find a solution?

nikplas, 2024-12-16:
still same issues here, invertion of roller state is actually the biggest problem; It states Closed when it is Open and vice versa;

Hopefully can someone bring an solution for this too; 
I wonder if someone knows another way workaround to invert the Roller state within HAOS

i am in generally very satisfied with ZHA and I am nog going to switch to zb2mqtt only for this;

saintfrater, 2025-02-15:
Hello, I'm facing exactly the same issue... "open" means 'close' and 'close' perform 'open'; the invert command in ZHA, seems to not stay... I'm more than happy to share any log or test.

tados000, 2025-02-22:
Do you think you can resolve the problem?

closed 2025-08-30 by github-actions[bot]