zigpy/zha-device-handlers #3204: [Device Support Request] NEO NAS-WV03B2 (_TZE204_4fblxpma, TS0601) Smart Irrigation Valve

bbadrignans, 2024-06-13:
### Problem description

I need some help to develop the quirk for the NEO NAS-WV03B2 irrigation timer valve, with the vendor ID _TZE204_4fblxpma and model TS0601.
[image]

https://www.szneo.com/en/products/show.php?id=273

I could find a quirk that support TS0601 valves : https://github.com/zigpy/zha-device-handlers/issues/1556

Changing only the VID in [ts0601_valve.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_valve.py) didn't work well. I tried for all the valve supported.
The best result is : valve states is correctly reported to HA but I can't set valve state from HA.

I haven't a Tuya gateway which make it difficult to find the good DP.
Does anyone got the same devices and can help me ? 

Additionally does someone has a good explanation on this "Tuya magic spell" (or a link) ?

### Solution description

I would like at least to be able to change valve state from ZHA.
Having battery status and measured water consumption would be great.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_4fblxpma",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_valve.TuyaValve"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
[... 936 lines cut ...]
              "access": "READ",
              "access_acl": 1,
              "attribute_value": 0
            },
            "0xfffd": {
              "attribute_id": "0xfffd",
              "attribute_name": "cluster_revision",
              "value_type": [
                "0x21",
                "uint16_t",
                "Analog"
              ],
              "access": "READ",
              "access_acl": 1,
              "attribute_value": 3
            }
          },
          "commands_received": {},
          "commands_generated": {}
        }
      }
    }
  ]
}
```

bbadrignans, 2024-06-13:
Just find out this model was added in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/giex.ts few days ago.
Anyone knows how to adapt from zigbee-herdsman-converters to zha quirks ?

bbadrignans, 2024-07-02:
I succeed to set the valve state using home assistant by modifying ts0601_valve.py as follow.
So, it seems : 
- on/off datapoint is 1 and not 2
- TuyaOnOffNM must be used instead of TuyaOnOff (to be honest I don't know what is the difference)

So I'm not sure this device behave like giex valve, despite code from [giex.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/giex.ts)

```
diff --git a/zhaquirks/tuya/ts0601_valve.py b/zhaquirks/tuya/ts0601_valve.py
index 6958afb..7b11cc2 100644
--- a/zhaquirks/tuya/ts0601_valve.py
+++ b/zhaquirks/tuya/ts0601_valve.py
@@ -314,11 +314,11 @@ class GiexValveManufCluster(TuyaMCUCluster):
     )
 
     dp_to_attribute: dict[int, DPToAttributeMapping] = {
-        1: DPToAttributeMapping(
+        2: DPToAttributeMapping(
             TuyaMCUCluster.ep_attribute,
             "irrigation_mode",
         ),
-        2: DPToAttributeMapping(
+        1: DPToAttributeMapping(
             TuyaOnOffNM.ep_attribute,
             "on_off",
         ),
@@ -381,7 +381,7 @@ class GiexValve(CustomDevice):
     """GiEX valve device."""
 
     signature = {
-        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"), ("_TZE200_a7sghmms", "TS0601"), ( "_TZE204_7ytb3h8u", "TS0601")],
+        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"), ("_TZE200_a7sghmms", "TS0601"), ( "_TZE204_7ytb3h8u", "TS0601"), ( "_TZE204_4fblxpma", "TS0601")],
         ENDPOINTS: {
             # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
             # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
```

vothar, 2024-07-05:
i have the same model and from the JSON diagnostic i get  :
             "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE204_a9ojznj8"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
it's the exact same valve but the 'green' colored version (same model as you, but not the same manufacturer) .
the valve is recognized and something is installed by HA (under zigbee ZHA) but nothing can be controlled

@bbadrignans , I didn't understand all the steps you took to make it work in HA, so could you make a short 'how-to' manual? (please :D )

bbadrignans, 2024-07-06:
@vothar 
Find the file ts0601_valve.py in your HASS server (mine is here : /srv/homeassistant/lib/python3.12/site-packages/zhaquirks/tuya/ts0601_valve.py).
Then replace it by the one below, remove your zigbee device from HASS, restart HASS, pair again your zigbee device. You should be able to control the valve.
Additional tip : don't use rechargeable battery, it reduces the range and the reliability : rechargeable are 1.2V, non rechargeable are 1.5V.
```

"""Collection of Tuya Valve devices e.g. water valves, gas valve etc."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import EnchantedDevice, TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffNM,
    TuyaPowerConfigurationCluster,
)

class TuyaValveWaterConsumed(Metering, TuyaLocalCluster):
    """Tuya Valve Water consumed cluster."""

    VOLUME_LITERS = 0x0007
    WATER_METERING = 0x02

    """Setting unit of measurement."""
    _CONSTANT_ATTRIBUTES = {
        0x0300: VOLUME_LITERS,
        0x0306: WATER_METERING,
    }

class TuyaValveManufCluster(TuyaMCUCluster):
    """On/Off Tuya cluster with extra device attributes."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("time_left", t.uint32_t, True),
            0xEF02: ("state", t.enum8, True),
            0xEF03: ("last_valve_open_duration", t.uint32_t, True),
            0xEF04: ("dp_6", t.uint32_t, True),
            0xEF05: ("valve_position", t.uint32_t, True),
        }
    )

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        5: DPToAttributeMapping(
            TuyaValveWaterConsumed.ep_attribute,
            "current_summ_delivered",
        ),
        6: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
[... 323 lines cut ...]
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffNM,
                    TuyaPowerConfigurationCluster,
                    TuyaValveWaterConsumed,
                    GiexValveManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

vothar, 2024-07-06:
@bbadrignans  THANK YOU! !! now the valve seems to react... I need to do more tests to make sure everything is ok.
I couldn't find the path you indicated on my HASS RPI4 server ( /srv/homeassistant/lib/python3.12/site-packages/zhaquirks/tuya/ ) but I found another way that 'seems' to be more convenient in my case (it's the official way to add custom zha quirks) than overwriting the original ts0601_valve.py file.

1-    Create a custom quirk dir in HA, e.g., /config/custom_zha_quirks
2-    In configuration.yaml,  add this 2 lines to point to this directory:
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/

```
then put the NEW modified ts0601_valve.py in this directory (-> custom_zha_quirks )

reboot HASS, and add the new ZHA device... (so we just need to remove this file in the custom directory if an official update add it)

thank you again for you help...

kageurufu, 2025-04-14:
I wrote a new quirk based on the herdsman converter, it seems to be working for me

```python
# Based on https://github.com/Koenkk/zigbee-herdsman-converters/blob/96efe558ebe43ba52a70a2a049d157a27cad15df/src/devices/neo.ts#L234C4-L271C7

from zigpy.quirks.v2 import EntityPlatform, EntityType
from zigpy.quirks.v2.homeassistant import UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t

from zhaquirks.const import BatterySize
from zhaquirks.tuya.builder import TuyaQuirkBuilder, TuyaValveWaterConsumed

class TuyaValveStatusExtended(t.enum8):
    Disabled = 0x00
    Active = 0x01
    Enabled = 0x02
    AppManual = 0x03
    KeyControl = 0x04

(
    TuyaQuirkBuilder('_TZE284_a9ojznj8', 'TS0601')
    .tuya_onoff(dp_id=1)
    .tuya_enum(
        dp_id=3, 
        attribute_name='valve_status', 
        enum_class=TuyaValveStatusExtended, 
        entity_platform=EntityPlatform.SENSOR,
        translation_key='valve_status',
        fallback_name='Valve status'
    )
    .tuya_number(
        dp_id=101, 
        attribute_name='valve_countdown', 
        type=t.uint16_t, 
        device_class=SensorDeviceClass.DURATION, 
        unit=UnitOfTime.MINUTES,
        min_value =1,
        max_value=240,
        step=1,
        translation_key='valve_countdown',
        fallback_name='Irrigation time'
    )
    .tuya_sensor(
        dp_id=6, 
        attribute_name='valve_duration', 
        type=t.uint32_t, 
        state_class=SensorStateClass.MEASUREMENT, 
        device_class=SensorDeviceClass.DURATION,
        unit=UnitOfTime.MINUTES, 
        entity_type=EntityType.STANDARD, 
        translation_key='valve_duration', 
        fallback_name='Irrigation duration'
    )
    .tuya_switch(
        dp_id=104,
        attribute_name="child_lock",
        translation_key="child_lock",
        fallback_name="Child lock",
    )
    .tuya_battery(dp_id=11, battery_type=BatterySize.AA, battery_qty=4)
    .skip_configuration()
    .add_to_registry()
)
```

I'm happy to clean this up and submit a PR, but I want some sort of validation it's up to quality (as it's my first quirk)

vothar, 2025-04-14:
hi.. i copied all your code in a brand new ts00601_valve.py file. i think there is a little error at the line  _TuyaQuirkBuilder('_TZE284_a9ojznj8', 'TS0601')_ : in my case, it's _TZE2**0**4_a9ojznj8  (TZE204 , not TZE284)

result of my test:
the switch work (valve on /off) 

the sensor **Irrigation time** didn't work ( i get unknown value)

In the configuration of the device, i can change the slider of the **Irrigation duration** from 1 min to 240 min, but i get no effect and also with child lock on or off (i put the slider on minimum 1 minute, and the water was still running after 2 minutes...)

so basically, it,s the same result as the first working quirk above (only the switch work)

vothar, 2025-04-15:
i add that i have this errors this morning in my system/log : 
`
Device: a4:c1:38:f9:ff:54:ab:4a-_TZE204_a9ojznj8 TS0601 has an entity with details: {'cluster_details': (1, 61184, <ClusterType.Server: 0>), 'entity_metadata': ZCLEnumMetadata(entity_platform=<EntityPlatform.SENSOR: 'sensor'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, unique_id_suffix=None, translation_key='valve_status', fallback_name='Valve status', primary=None, enum=<enum 'TuyaValveStatusExtended'>, attribute_name='valve_status', reporting_config=None)} that does not have an entity class mapping - unable to create entity`