zigpy/zha-device-handlers #3207: [BUG] Failed to convert attribute 0x0200 from OppleOperationMode.Decoupled (<class 'str'>) to type <enum 'OppleOperationMode'>:

Briadark, 2024-06-15:
### Bug description

Since the 2024.6.X (just tested .3) update I seem to have some issues with changing the OppleSwitchCluster->OppleOperationMode settings of my Aqara switches.
Both the n2aeu1 and l2aeu1 have this issue.
Loaded quirk for n2aeu1: zhaquirks.xiaomi.aqara.switch_h1_double.AqaraH1DoubleRockerSwitchWithNeutral4 
Loaded quirk for l2aeu1:
zhaquirks.xiaomi.aqara.switch_h1_double.AqaraH1DoubleRockerSwitchNoNeutral

LOG:
Logger: zigpy.zcl
Source: components/zha/core/device.py:789
First occurred: 10:44:48 AM (1 occurrences)
Last logged: 10:44:48 AM

[0x8EEC:1:0xfcc0] Failed to convert attribute 0x0200 from OppleOperationMode.Decoupled (<class 'str'>) to type <enum 'OppleOperationMode'>: 'OppleOperationMode.Decoupled' is not a valid OppleOperationMode

Reading the operation_mode works fine, the writing of the operation mode is where the issue presents itself.
Switches that where already set to decoupled still work in that mode and keep their settings, but I can't change these settings in the new release.

### Steps to reproduce

1. Add these switches.
2. Try to change the OppleOperationMode.
3. Look at the home assistant logs.

### Expected behavior

For the OppleOperationMode to change to Coupled/Decoupled

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0012",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    },
    "31": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    },
    "41": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "42": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "51": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "61": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n2aeu1",
  "class": "zhaquirks.xiaomi.aqara.switch_h1_double.AqaraH1DoubleRockerSwitchWithNeutral4"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
Logger: zigpy.zcl
Source: components/zha/core/device.py:789
First occurred: June 15, 2024 at 11:12:28 PM (3 occurrences)
Last logged: June 15, 2024 at 11:13:05 PM

[0x9137:1:0xfcc0] Failed to convert attribute 0x0200 from OppleOperationMode.Coupled (<class 'str'>) to type <enum 'OppleOperationMode'>: 'OppleOperationMode.Coupled' is not a valid OppleOperationMode
[0x9137:1:0xfcc0] Failed to convert attribute 0x0200 from None (<class 'str'>) to type <enum 'OppleOperationMode'>: 'None' is not a valid OppleOperationMode
[0x9137:1:0xfcc0] Failed to convert attribute 0x0200 from (<class 'str'>) to type <enum 'OppleOperationMode'>: '' is not a valid OppleOperationMode
```

</details>

bakovari, 2024-11-01:
Hi there,
Is there any fix for this issue?
I have the same problem, I cannot change the Operation mode to decoupled.  :(
Thanks for your help!

Torrenator, 2024-11-10:
I'm also running into this issue. On a fairly new install since 2024.10

Torrenator, 2024-11-10:
For the people still waiting for a fix. Try using the direct translations to configure the 'Decoupled' == 0 or 'Coupled' == 1.
[image]

carlosjustogithub, 2024-11-14:
I am having the same issue after October/24 update. Whenever I try to change a setting, I get this message:
Failed to convert attr_value=3.0 from type <class 'zigpy.types.basic.Single'> to int | str

Has anyone found a fix for this? 
Thanks!!

dmpcc, 2024-11-15:
Try putting a '0' or a '1' in this field
[image]