zigpy/zha-device-handlers #3209: Add Tuya Giex GX02 water valve support

dethegeek, 2024-06-16:
### Problem description

Similar device aded to solve the following issue #3124 

My device is detected as TS0601 _TZE200_7ytb3h8u

Bought here: https://fr.aliexpress.com/item/1005006548852233.html

### Solution description

I succesfully aded the device by adding the above model in https://github.com/zigpy/zha-device-handlers/blob/c6ed94a52a469e72b32ece2a92d528060c7fd034/zhaquirks/tuya/ts0601_valve.py#L384

I'm rather new with domotic and Home Assistant, I'm not skilled enough to get a zigbee to MQTT dump for now, then I did a try after roughly understanding how the ZHA quirks works.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_7ytb3h8u",
  "model": "TS0601",
  "class": "ts0601_valve.GiexValve"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2024.6.2",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
[... 806 lines cut ...]
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffNM,
                    TuyaPowerConfigurationCluster,
                    TuyaValveWaterConsumed,
                    GiexValveManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

```

</details>

jaggaer-alocatel, 2024-07-20:
Hi @dethegeek I bought the same valve but unfortunately I cannot integrate it successfully as suggestedabove.  I copied the quirk file adding the model and  I see it in HA with only switch e battery sensor.  the problem is that the switch doesn't work (like the valve ignores it) an battery is unknown. 
same behaviour with 2 identical valves.
have I missed anything?

liksvette, 2024-07-30:
I added `_TZE200_7ytb3h8u` to line 384. No idea how to report this, so I'll just comment it here, and hope a grown-up can assist. It's for this unit: https://www.aliexpress.com/item/1005006547328147.html?spm=a2g0o.order_list.order_list_main.4.787f18021Qrnq9

`        MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"), ("_TZE200_a7sghmms", "TS0601"), ( "_TZE200_7ytb3h8u", "TS0601"), ( "_TZE204_7ytb3h8u", "TS0601")],
`

dethegeek, 2024-08-05:
Hi @jaggaer-alocatel 

On my side I can switch the state of my device. 

However I noticed some issues
- Battery level is often reported 200% or sometimes 400%. It usually drops to 100% when I switch on the device, and since a few days it dropped to 90% (semms legit, depite the battery is drained quite fast).
- Instant water delivery is empty
- Summation of delivered water is reset to zero when I switch on the device again.

dethegeek, 2024-08-05:
Hi @liksvette 

My device looks the same as yours.

jaggaer-alocatel, 2024-08-05:
> Hi @jaggaer-alocatel
> 
> On my side I can switch the state of my device.
> 
> However I noticed some issues
> 
> * Battery level is often reported 200% or sometimes 400%. It usually drops to 100% when I switch on the device, and since a few days it dropped to 90% (seems legit, depite the battery is drained quite fast).
> * Instant water delivery is empty
> * Summation of delivered water is reset to zero when I switch on the device again.

sorry, it was by bad... wrong change in the file. Now my devices look as ours. Thanks a lot

liksvette, 2024-08-05:
> Hi @liksvette
> 
> My device looks the same as yours.

Your code snippet had _TZE204 as opposed to _TZE200, so it didn't work on mine.

dethegeek, 2024-08-08:
@liksvette the code snippet is the original line (this is a reference to the line in the repository). You just need to add the device in it.

The result should look like this
```
MODELS_INFO: [("_TZE200_sh1btabb", "TS0601"), ("_TZE200_a7sghmms", "TS0601"), ( "_TZE204_7ytb3h8u", "TS0601"), ( "_TZE200_7ytb3h8u", "TS0601")],
```

closed 2024-10-20 by TheJulianJES

nicktrigger, 2024-11-03:
> ### Problem description
> Similar device aded to solve the following issue #3124
> 
> My device is detected as TS0601 _TZE200_7ytb3h8u
> 
> Bought here: https://fr.aliexpress.com/item/1005006548852233.html
> 
> ### Solution description
> I succesfully aded the device by adding the above model in
> 
> https://github.com/zigpy/zha-device-handlers/blob/c6ed94a52a469e72b32ece2a92d528060c7fd034/zhaquirks/tuya/ts0601_valve.py#L384
> 
> I'm rather new with domotic and Home Assistant, I'm not skilled enough to get a zigbee to MQTT dump for now, then I did a try after roughly understanding how the ZHA quirks works.
> 
> ### Screenshots/Video
> Screenshots/Video
> ### Device signature
> Device signature
> ### Diagnostic information
> Diagnostic information
> ### Logs
> Logs
> ### Custom quirk
> ts0601_valve.py
> ### Additional information
> Enable / disable test done
> 
> ![valve](https://private-user-images.githubusercontent.com/5788361/340119617-6e4b5c9e-957d-4360-84fb-bedeb4ba5a08.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA2NDYyNDYsIm5iZiI6MTczMDY0NTk0NiwicGF0aCI6Ii81Nzg4MzYxLzM0MDExOTYxNy02ZTRiNWM5ZS05NTdkLTQzNjAtODRmYi1iZWRlYjRiYTVhMDguZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDNUMTQ1OTA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjJmYWU1ZDZkZDQ1YTFmMjFiNmQ3OGJlNGUzYjg1ZTQ1MzhmMDY4MjkxOWJmZGZhNzQ2M2M5ODVlMzA0MmE0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.e-hZFQKpZ_2V_LTA8WOCg6K4hikQfVj6WiyFr_-JUhw) [ ![valve](https://private-user-images.githubusercontent.com/5788361/340119617-6e4b5c9e-957d-4360-84fb-bedeb4ba5a08.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA2NDYyNDYsIm5iZiI6MTczMDY0NTk0NiwicGF0aCI6Ii81Nzg4MzYxLzM0MDExOTYxNy02ZTRiNWM5ZS05NTdkLTQzNjAtODRmYi1iZWRlYjRiYTVhMDguZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDNUMTQ1OTA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjJmYWU1ZDZkZDQ1YTFmMjFiNmQ3OGJlNGUzYjg1ZTQ1MzhmMDY4MjkxOWJmZGZhNzQ2M2M5ODVlMzA0MmE0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.e-hZFQKpZ_2V_LTA8WOCg6K4hikQfVj6WiyFr_-JUhw) ](https://private-user-images.githubusercontent.com/5788361/340119617-6e4b5c9e-957d-4360-84fb-bedeb4ba5a08.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA2NDYyNDYsIm5iZiI6MTczMDY0NTk0NiwicGF0aCI6Ii81Nzg4MzYxLzM0MDExOTYxNy02ZTRiNWM5ZS05NTdkLTQzNjAtODRmYi1iZWRlYjRiYTVhMDguZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDNUMTQ1OTA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjJmYWU1ZDZkZDQ1YTFmMjFiNmQ3OGJlNGUzYjg1ZTQ1MzhmMDY4MjkxOWJmZGZhNzQ2M2M5ODVlMzA0MmE0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.e-hZFQKpZ_2V_LTA8WOCg6K4hikQfVj6WiyFr_-JUhw) [ ](https://private-user-images.githubusercontent.com/5788361/340119617-6e4b5c9e-957d-4360-84fb-bedeb4ba5a08.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA2NDYyNDYsIm5iZiI6MTczMDY0NTk0NiwicGF0aCI6Ii81Nzg4MzYxLzM0MDExOTYxNy02ZTRiNWM5ZS05NTdkLTQzNjAtODRmYi1iZWRlYjRiYTVhMDguZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDNUMTQ1OTA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjJmYWU1ZDZkZDQ1YTFmMjFiNmQ3OGJlNGUzYjg1ZTQ1MzhmMDY4MjkxOWJmZGZhNzQ2M2M5ODVlMzA0MmE0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.e-hZFQKpZ_2V_LTA8WOCg6K4hikQfVj6WiyFr_-JUhw)
> 
> _No response_

I'm sure this is a dumb question but where and how can I access this file on my home syarat instance to add the code?
I tried to add as a chain quirk but it had errors and didn't load. 
There seems to be no documenting or explanation anywhere about how to actually do this!

dethegeek, 2024-11-03:
> <https://private-user-images.githubusercontent.com/5788361/340119617-6e4b5c9e-957d-4360-84fb-bedeb4ba5a08.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA2NDYyNDYsIm5iZiI6MTczMDY0NTk0NiwicGF0aCI6Ii81Nzg4MzYxLzM0MDExOTYxNy02ZTRiNWM5ZS05NTdkLTQzNjAtODRmYi1iZWRlYjRiYTVhMDguZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDNUMTQ1OTA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjJmYWU1ZDZkZDQ1YTFmMjFiNmQ3OGJlNGUzYjg1ZTQ1MzhmMDY4MjkxOWJmZGZhNzQ2M2M5ODVlMzA0MmE0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.e-hZFQKpZ_2V_LTA8WOCg6K4hikQfVj6WiyFr_-JUhw> [image:

Hi,

This issue has been fixed 2 weeks ago, and I see a tag has been created 1
week ago.

Then instead of matching files yourself you should first ensure that you
use the latest release of Home Assistant. It may already contain the patch
now.

As I'm answering from a smartphone and don't have a computer available, I
may be wrong. I could only check the dates of resolution and last tag.

nicktrigger, 2024-11-03:
Many thanks my devices have different names but are basically the same. I just want to test to see if they work by inserting the product codes into this quirk but i have no idea where to start.  There seems to be something missing in my setup. Here is the latest error:

Logger: zhaquirks
Source: /usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py:464
First occurred: 23:07:39 (1 occurrences)
Last logged: 23:07:39

Unexpected exception importing custom quirk 'Valves'
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py", line 462, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/config/custom_zha_quirks/Valves.py", line 27, in <module>
    from zhaquirks.tuya.builder import TuyaQuirkBuilder
ModuleNotFoundError: No module named 'zhaquirks.tuya.builder'

nicktrigger, 2024-11-03:
My devices are ts0601_TZE200_akjefhj5 and ts0601_TZE204_qtnjuoae