zigpy/zha-device-handlers #3212: [Device Support Request] Thermostat Avatto TS0601 _TZE204_lzriup1j

lenkeybalis, 2024-06-17:
### Problem description

Avatto ZWT100 smart thermostat 3A. 

This thermostat is not compatible with ZHA. When I integrated this device into the HA ( current version is 2024.6.4), I can see only the firmware version.

I would like to manage a simple gas heater via this tool. 

Tool type:

TS0601
by _TZE204_lzriup1j
Firmware: 0x0000004a 

### Solution description

Temperature  control would be minimum required, to able to manage the temperature in the flat.
- Set the desired temperature value
- Read the current 

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 6388,
    "manufacturer": "_TZE204_lzriup1j",
    "model": "TS0601",
    "name": "_TZE204_lzriup1j TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 220,
    "rssi": -45,
    "last_seen": "2024-06-17T22:08:52",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "_TZE204_lzriup1j",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "sensor.tze204_lzriup1j_ts0601_rssi",
        "name": "_TZE204_lzriup1j TS0601"
      },
      {
        "entity_id": "sensor.tze204_lzriup1j_ts0601_lqi",
        "name": "_TZE204_lzriup1j TS0601"
      },
      {
        "entity_id": "update.tze204_lzriup1j_ts0601_firmware",
        "name": "_TZE204_lzriup1j TS0601"
      }
    ],
```

</details>

### Logs

<details><summary>Logs</summary>

```python

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

```

</details>

lenkeybalis, 2024-11-09:
Dear all,

Could you please help me how I can reach the similar solution ZHA, what I saw with Z2MQTT in this ticket?

https://github.com/Koenkk/zigbee2mqtt/issues/19459#issuecomment-1846785112

I would like to somehow integrate my thermostat device to HA with ZHA.

Thank you so much!

sunsetitaly, 2024-12-21:
Hi, have you found a solution to your issue? I have exactly the same ptoblem

prairiesnpr, 2024-12-21:
> Hi, have you found a solution to your issue? I have exactly the same problem

We just finished out a very complete [Tuya Thermostat quirk using the new v2 style](https://github.com/zigpy/zha-device-handlers/pull/3625). If you take the [DPs from z2m](https://github.com/Koenkk/zigbee2mqtt/issues/19459#issuecomment-1846785112) you should be able to build a working quirk with a little bit of work.

PR #3748 (2025-01-25): Add Avatto thermostats (Tuya)
## Proposed change

Adds Tuya ZWT198/ZWT100-BH Avatto wall thermostats

- _TZE204_lzriup1j
- _TZE200_viy9ihs7
- _TZE204_xnbkhhdr
- _TZE284_xnbkhhdr

Closes: #3627
Closes: #3212
Closes: #2589
Closes: #3482
Closes: #3326

## Additional information

Based on: https://github.com/Koenkk/zigbee2mqtt/discussions/25752
and https://github.com/Koenkk/zigbee-herdsman-converters/blob/40e72bc2515deb67ef35dddab04437342b7b3eb9/src/devices/tuya.ts#L7027
## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using [... cut]
closed 2025-01-25 by TheJulianJES