zigpy/zha-device-handlers #3226: [Device Support Request] TS0601 _TZE204_kgaxpvxr

DavidAntonin, 2024-06-22:
### Problem description

Please add Tuya smoke sensor

### Solution description

more info about this Tuya smoke sensor: https://github.com/Koenkk/zigbee2mqtt/issues/19667

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2024.6.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "3.10.108",
    "run_as_root": true
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "dmulcahey",
      "adminiuga",
      "puddly",
      "TheJulianJES"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
[... 516 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

moz4r, 2025-01-16:
Hi I have the same sensor someone get it worky ?

DavidTre07, 2025-01-17:
Hello,
Just received this smoke sensor, detected as _TZE204_kgaxpvxr TS0601 but not working for me too.

[image]

moz4r, 2025-01-17:
Maybe we can try custom quirks and add the id, not tested yet : https://github.com/zigpy/zha-device-handlers/blob/e8a1570203012ba12777fecd18df82ecd1ff4883/zhaquirks/tuya/ts0601_smoke.py

DavidTre07, 2025-01-18:
OK yes we can try add the device info following line 81 but I don't know to repackage ZHA and test it :/
I need to RTFM !

moz4r, 2025-01-19:
Hi I tested and get it worky but with very minimal support,  only smoke detection, no siren shutdown or batterie level . But it is a good start ! To test , add this to configuration.yaml

`zha:
     database_path: /config/zigbee.db
     enable_quirks: true
     custom_quirks_path: /config/zha_quirks/`

then put moded ts0601_smoke.py file into \config\zha_quirks\ folder. Reboot ( you can install samba plugin to access your ha files easily

[image]

DavidTre07, 2025-01-26:
Hello @moz4r,
Sorry I was travelling last week, no time to have a look on this.
Great to see you got a basic support, but I don't see the modded file "ts0601_smoke.py".
Could you provide it ?
Thanks.

PR #3794 (2025-01-27): Add Tuya smoke sensor `_TZE204_kgaxpvxr`
## Proposed change

Add Tuya smoke _TZE204_kgaxpvxr

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/db7595cf52b55ff47c2132d0952d7444cc0f1b91/src/devices/onenuo.ts#L10
Closes: #3226

## Additional information

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-01-27 by TheJulianJES