zigpy/zha-device-handlers #3228: Tuya Siren TS0601 by _TZE204_hcxvyxa5

Martijnsx, 2024-06-25:
### Problem description

Hi, i have this new tuya siren. Is it possible to add support to it to ZHA?

It is a cheap and small siren but for its size loud.

TS0601 by _TZE204_hcxvyxa5

### Solution description

A quirk? Or a normal implementation?

I do not know how to make a quirk.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheOddPerson, 2024-07-23:
Same as issue #3255 
However more information provided the other device request

closed 2024-09-24 by TheJulianJES

JAKAMI99, 2024-09-25:
Working!

At first, I enabled custom quirks in the configuration.yaml in home assistant because I didn't want to wait for the :main update
`zha:
   enable_quirks: true
   custom_quirks_path: /config/zha_quirks/`

Then I downloaded the custom quirk:
`➜  zha_quirks wget https://raw.githubusercontent.com/zigpy/zha-device-handlers/5c993588c79b7708f9ecfc8ad893df6e51d97924/zhaquirks/tuya/ts0601_siren.py`

I rebooted, removed and added the  `_TZE204_hcxvyxa5` device.

Now the device has a switch. If turned on, the device makes a siren noise for ~500ms and turns off again,

Trough "manage zigbee device" I could change some settings by choosing cluster "TuyaMCUSiren" through the attributes

- `alarm_duration (id: 0x0267)` sets the alarm duration in seconds.  
- The volume can be changed trough `volume (id: 0x0474)` by setting the value to `0`(`low`) , `1`(`medium`) or to `2`(`high`)  
- The melody can be set trough `melody (id: 0x0466)` where the value can be from `0` `36` (`37` seems to be a duplicate with 0) which varies from classical songs like "Für Elise" (`24`) to notification ish/doorbell sounds (`11`) to the three alarm siren sounds (`0`-`2`) where `0` is the most alarm one, but they really have a big cheap toy vibe 👶

_Melody (dropdown-menu), volume (dropdown-menu) and duration (slider) should be added to the GUI, so everyone can access it._

Thanks for the quick implementation! ✅

PR #3361 (2024-09-24): Add Tuya Siren TS0601 `_TZE204_hcxvyxa5` variant
## Proposed change

This PR adds siren TS0601 _TZE204_hcxvyxa5 into the existing quirk of `TuyaSirenGPP_NoSensors`, and fixes #3255 and fixes #3228.

I have confirmed locally that it enables the siren to be toggled like other similar sirens, and I am able to adjust melody and duration via "Manage zigbee device".

## Additional information

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works