zigpy/zha-device-handlers #3229: [BUG] TS0601 TZE200_a8sdabtg does not report temperature correctly.

Shintaro000000, 2024-06-26:
### Bug description

I bought two temperature sensors on Aliexpress.
Once the zigbee devices are associated with ZHA, they start working correctly for the first 15/20 minutes, then they always report 0 and only give the temperature once an hour. The same goes for humidity. The batteries are new and the sensor is next to the coordinator I am using (ZBdongle-E).
I have tried reconfiguring them multiple times without success.

### Steps to reproduce

Simply adding the device, the problem appears shortly after.

### Expected behavior

It should give the correct temperature without showing -0

### Screenshots/Video

[image]
[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405"
      ],
      "output_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "_TZE200_a8sdabtg",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.6.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Rome",
    "os_name": "Linux",
    "os_version": "6.6.33-haos",
    "supervisor": "2024.06.2",
    "host_os": "Home Assistant OS 12.4",
    "docker_version": "26.1.4",
[... 423 lines cut ...]
2024-06-26 01:44:15.838 DEBUG (bellows.thread_0) [bellows.ash] Sending data  44f921a9a52af0df7e
2024-06-26 01:44:15.854 DEBUG (bellows.thread_0) [bellows.ash] Received data 45f9a1a9a52a78b266947425b25593499c4e27abedce738bfcc66389fb7d5e3ca7ebcdde6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c267d33b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a2519048245cae7e
2024-06-26 01:44:15.854 DEBUG (bellows.thread_0) [bellows.ash] Received frame DataFrame(frm_num=4, re_tx=0, ack_num=5, ezsp_frame=b'\xbb\x80\x01\xf1\x00m\x00?\x00>\x00\x18\x00\x01\x00\x00\x00\x00\x00\x00\x00\x14\x00\x01\x00\x00\x00\x07\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-06-26 01:44:15.854 DEBUG (bellows.thread_0) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=5) + FLAG
2024-06-26 01:44:15.854 DEBUG (bellows.thread_0) [bellows.ash] Sending data  8520dd7e
2024-06-26 01:44:15.855 DEBUG (MainThread) [bellows.ezsp.protocol] Received command readCounters: [[109, 63, 62, 24, 1, 0, 0, 0, 20, 1, 0, 7, 3, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]]
2024-06-26 01:44:15.857 DEBUG (MainThread) [bellows.ezsp.protocol] Sending command  getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2024-06-26 01:44:15.857 DEBUG (bellows.thread_0) [bellows.ash] Sending frame DataFrame(frm_num=5, re_tx=False, ack_num=5, ezsp_frame=b'\xbc\x00\x01\xaa\x00\x03') + FLAG
2024-06-26 01:44:15.857 DEBUG (bellows.thread_0) [bellows.ash] Sending data  55fe21a9fe2a16ab8c7e
2024-06-26 01:44:15.862 DEBUG (bellows.thread_0) [bellows.ash] Received data 56fea1a9fe2a15b3a5e07c7e
2024-06-26 01:44:15.862 DEBUG (bellows.thread_0) [bellows.ash] Received frame DataFrame(frm_num=5, re_tx=0, ack_num=6, ezsp_frame=b'\xbc\x80\x01\xaa\x00\x00\x01\xfc')
2024-06-26 01:44:15.862 DEBUG (bellows.thread_0) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=6) + FLAG
2024-06-26 01:44:15.862 DEBUG (bellows.thread_0) [bellows.ash] Sending data  8610be7e
2024-06-26 01:44:15.863 DEBUG (MainThread) [bellows.ezsp.protocol] Received command getValue: [<EzspStatus.SUCCESS: 0>, b'\xfc']
2024-06-26 01:44:15.865 DEBUG (MainThread) [bellows.zigbee.application] Free buffers status EzspStatus.SUCCESS, value: 252
2024-06-26 01:44:15.865 DEBUG (MainThread) [bellows.zigbee.application] ezsp_counters: [MAC_RX_BROADCAST = 784, MAC_TX_BROADCAST = 413, MAC_RX_UNICAST = 970, MAC_TX_UNICAST_SUCCESS = 141, MAC_TX_UNICAST_RETRY = 4, MAC_TX_UNICAST_FAILED = 0, APS_DATA_RX_BROADCAST = 1, APS_DATA_TX_BROADCAST = 1, APS_DATA_RX_UNICAST = 434, APS_DATA_TX_UNICAST_SUCCESS = 60, APS_DATA_TX_UNICAST_RETRY = 0, APS_DATA_TX_UNICAST_FAILED = 7, ROUTE_DISCOVERY_INITIATED = 5, NEIGHBOR_ADDED = 2, NEIGHBOR_REMOVED = 0, NEIGHBOR_STALE = 1, JOIN_INDICATION = 0, CHILD_REMOVED = 0, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 252]
```

</details>

### Additional information

I don't know if it helps, but I saw that this model is present in the quirks, yet it doesn't seem to recognize the sensor correctly or there might be another problem. Maybe I made a mistake. I also tried to manually add ts0601__sensor as per the guide, but nothing changed.

[image]

Shintaro000000, 2024-06-27:
Good morning,

Based on what I found online and in the original TS0601_sensor file, I tried creating my own file specifically for the sensor that was giving me issues. I made a few modifications to the original file to better fit this model. Since I'm not very skilled in these things, I'm not sure if the changes I made actually helped, but I know that with my modifications, it has been working without problems for about 4 hours now. I'm sharing the code in case it might be useful.

If my modifications are useless or incorrect, I would like to know so that I can resolve these issues more easily in case I encounter similar problems with other devices.

Thank you very much, and have a great day.

[image]

```py
from typing import Any

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, Identify
from zigpy.zcl.clusters.measurement import (
    RelativeHumidity,
    SoilMoisture,
    TemperatureMeasurement,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaSoilMoisture(SoilMoisture, TuyaLocalCluster):
    """Tuya local SoilMoisture cluster with a device RH_MULTIPLIER factor if required."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster with a device RH_MULTIPLIER factor."""

    def update_attribute(self, attr_name: str, value: Any) -> None:
        """Apply a correction factor to value."""

        if attr_name == "measured_value":
            value = value * (
                self.endpoint.device.RH_MULTIPLIER
                if hasattr(self.endpoint.device, "RH_MULTIPLIER")
                else 100
            )
        return super().update_attribute(attr_name, value)

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            # converter=lambda x: x * 10,  --> move conversion to TuyaRelativeHumidity cluster
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            converter=lambda x: x * 2,  # double reported percentage
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
    }  # Aggiunta la chiusura della parentesi graffa

class TuyaTempHumiditySensor_Square(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    # RelativeHumidity multiplier
    # RH_MULTIPLIER = 100

    signature = {
        MODELS_INFO: [
            ("_TZE200_a8sdabtg", "TS0601"),  # Variant without screen, round
        ],
        ENDPOINTS: {
            1: {
                # "profile_id": 260, "device_type": "0x0302",
                # "in_clusters": ["0x0000","0x0003","0x0001","0x0402","0x0405"],
                # "out_clusters": ["0x000a","0x0019","0x0003"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    TuyaPowerConfigurationCluster2AAA.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id],
            }
        },
    }
    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster2AAA,
                    TemperatureHumidityManufCluster,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id],
            }
        },
    }
```

lvillani, 2024-07-05:
Hi @Shintaro000000, would you consider re-opening this issue? I have the same sensor and observed the same behaviour. I tried your custom ZHA quirk and I confirm that it works for me. However, I would love to see this addressed upstream so that I can stop having to carry a custom quirk file with my own Home Assistant installation.

Shintaro000000, 2024-07-05:
@lvillani I'm glad it's working for you. I reopened it as requested.
Have a good day :)

j2gl, 2024-07-23:
Hi I have the same issue and saw it posted in several places, it will be great if this can be fixed. 

Thanks for the great work. 

[image]

[image]

j2gl, 2024-08-02:
Hi @myromeo,  

How did you make it to work? Did you add the script or program from this [comment](https://github.com/zigpy/zha-device-handlers/issues/3229#issuecomment-2194627391) in HomeAssistant?  or maybe just point me to the documentation.

Thanks!

chrisism, 2024-08-07:
I have the same issue. I have 4 of these sensors. 2 of them work correctly, 2 of them have this issue as described above.
Where do I need to place the python script file above?

Shintaro000000, 2024-08-08:
@chrisism I followed the guides I found (though I might have made a mistake or forgotten a step). I created a folder called zha_quirks in /homeassistant, and I placed the script inside it. 
Then, in the configuration.yaml, I added the following:

zha:
  database_path: /config/zigbee.db
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks
  
After restarting the device, it recognized the quirk, and it shows up in the device's description. Of course, the device you have must have the exact code TZE200_a8sdabtg, otherwise, it won't work.

Since I made the post, it's been working without any issues for me.

I hope this helps you. Have a great day!

[image]

chrisism, 2024-08-08:
Thanks, got the quirk working. I have exactly the same sensor. Still two of my sensors show -0.0. Other two work fine just as before. Lets see if it starts picking it up.

Shintaro000000, 2024-08-08:
@chrisism You can also try removing and re-adding the devices after adding the quirk.

chrisism, 2024-08-10:
@Shintaro000000 that did the trick. Looks like it works. Only restarted the service before but that isnt enough.

j2gl, 2024-09-03:
@Shintaro000000 many thanks for the code, I put it in my HA and now it's working good since Saturday.

moebis, 2024-09-07:
@Shintaro000000 thank you so much! This fixed it. What a frustration. ZigBee2MQTT had no issues with this device and I found endless threads with folks having this problem (and similar ones) with ZHA for a long time. I'm amazed they either don't work together or share some kind of database of ZigBee devices. Really bad user experience with ZHA, almost went back to Z2M but I stuck it out because I wanted to use something that was out of the box supported. Z2M also has some strange issues of its own, but I have yet to find a device it can't handle, ZHA has improved slightly over the last 6 months with device support, but the developers completely ignore the forums (where all of these reports are) and only focus on GitHub repo which I (and many others) did not know about. Anyways, thanks again for sharing!

lvillani, 2025-01-07:
I recently revisited my setup and noticed that the code to handle this sensor was refactored. Intrigued, I removed the sensor from my HA 2024.12.05 installation, removed the custom quirk and then re-added the sensor. It seems to work just fine out of the box now! The device info page reports `Quirk: zigpy.quirks.v2.CustomDeviceV2`, which should be the builtin quirk. Can someone else confirm that the issue has been fixed upstream and that the custom quirk is no longer necessary?

sspenning, 2025-02-28:
I'm new to HA and I just picked up a pair of _TZE200_a8sdabtg Zigbee sensors.  Neither is working correctly using stock HAOS on a x86_64 install.

I see the sensors as Zigbee devices but when I look at the sensor values like humidity, I see the graph for data but across the top it indicates "This entity is unavailable." https://i.imgur.com/V488JOU.png

I've tried adding the quirk discussed in this bug it persists in using zigpy.quirks.v2.CustomDeviceV2.

Can anyone point me in the right direction?

danilolima, 2025-08-30:
> I have a similar issue, bought a device identifying as 'TS0601 by _TZE284_1wnh8bqp', it is not showing any temperature or humidity, and it is not picking up any quirk.
> 
> <img alt="Image" width="1028" height="632" src="https://private-user-images.githubusercontent.com/185636905/479938573-01016870-4100-46da-a153-d718df32cbf5.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTY1OTA4MjQsIm5iZiI6MTc1NjU5MDUyNCwicGF0aCI6Ii8xODU2MzY5MDUvNDc5OTM4NTczLTAxMDE2ODcwLTQxMDAtNDZkYS1hMTUzLWQ3MThkZjMyY2JmNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgzMFQyMTQ4NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kYThhNTI0NGZhNWE5NTNhMDU5OTEyNDY3ZjhhNzRhMjY0ZDlhZmYxMGEyMzA5MzQwMTgzNjVkNDE1MjNhYjA0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.YCQfKNIcyjgFY1cpHbh8_Z_oXkneDclGbPybqXQQJQQ">
> [zha-01J7N1YR6YMBZ9YXKGCRQE27PX-_TZE284_1wnh8bqp TS0601-23d8d009fbb8c6b14718c8641036cea6.json](https://github.com/user-attachments/files/21889993/zha-01J7N1YR6YMBZ9YXKGCRQE27PX-_TZE284_1wnh8bqp.TS0601-23d8d009fbb8c6b14718c8641036cea6.json)

Same here