zigpy/zha-device-handlers #3230: [BUG] Tuya TY0201 Temperature/Humidity sensor does not work

vJan00, 2024-06-26:
### Bug description

I have tested the Quirk which was merged by #3201 and it does not work with my TY0201 _TZ3000_bjawzodf!
It does not show humidity, nor the temperature, which it showed without the Quirk.
Based on other Quirks I have developed my own Quirk for TY0201 _TZ3000_bjawzodf, which works without problems.
This should be integrated as V2.

### Steps to reproduce

Try to pair the TY0201 _TZ3000_bjawzodf and read temperature or humidity sensor (doesn't show up).

### Expected behavior

Pairing the TY0201 _TZ3000_bjawzodf which results in temperature and humidity sensor showing up.

vJan00, 2024-06-26:
Opened #3231 to resolve this issue.

vJan00, 2024-06-26:
@philippedeswert would it be possible you can test the updated ty0201.py too? :)

philippedeswert, 2024-06-26:
> @philippedeswert would it be possible you can test the updated ty0201.py too? :)

Sure, I was planning to do the v2 rewrite myself this weekend as I was getting an extra HA setup so I did not need to mess with my "production" setup to test things.

Mattie112, 2024-10-09:
Do you happen to have any progress on this? I bought a couple but (in HomeAssisant) only the temperature shows up, not the humidity.

philippedeswert, 2024-10-09:
I haven't had time yet unfortunately. I managed to find a little bit of time to test things, but something is indeed still off as I added a new one, on a clean home assistant instance and I had to restart zha to have it recognize everything.

Mattie112, 2024-10-09:
Yeah, I just wanted to add: added 2more but those don't even have the temperature shown. Strange... I will see if I can dive into it but not sure where to start.

philippedeswert, 2024-10-09:
Try reconfigure in the device menu and then restart zha...

Mattie112, 2024-10-09:
Ah yes indeed, now it shows up. We'll great thanks! Let's see if it continues to work :)

phhusson, 2024-10-30:
I don't really understand. I have the same issue. I've tried reconfiguring it three times with always the same result.

But I can read the TuyaAirQualityTemperature and TuyaAirQualityHumidity clusters just fine manually?

[image]
[image]
[image]

Can someone point me to what am I doing wrong?

Migsi, 2024-12-05:
I can also confirm this fix works. @vJan00 please look forward to get your PR rebased and merged ;)

closed 2025-01-16 by TheJulianJES