zigpy/zha-device-handlers #3233: [Device Support Request] Moes Tuya sliding windows pusher

tribela, 2024-06-28:
### Problem description

I got this(https://www.aliexpress.com/item/1005006924617514.html) product and connected it to Home Assistant.
But it detected as Smoke detector, Not windows cover variant.

### Solution description

It should detected as windows cover, Not a smoke detector

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_5rta89nj",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2024.6.4",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Asia/Seoul",
    "os_name": "Linux",
    "os_version": "5.15.0-107-generic",
    "run_as_root": true
  },
  "custom_components": {
[... 729 lines cut ...]
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWindowCoverManufCluster,
                    TuyaWindowCovering,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```

</details>

AlbertoV8, 2024-07-03:
I have the same issue...

devtech-tech, 2024-07-31:
Really great product but have had the same issues on Zigbee and Zigbee2MQTT

jajaP42, 2024-09-26:
Hi, I have the same problem. Is there any way to use it urgently before getting official support?

devtech-tech, 2024-09-27:
@ jajaP42

Checkout the really great work - https://github.com/Koenkk/zigbee2mqtt/issues/22844

Working perfectly on two units for me.

jajaP42, 2024-09-27:
@devtech-tech   Thanks for the answer. Since I'm a newbie, what you mean is, should I move to Z2M? Or should I try to convert the code to ZHA?

devtech-tech, 2024-09-27:
@jajaP42 I find that both have advantages and disadvantages. I run a device for each ZHA and Z2M, lost a lot of hair trying to get both to work on a single device.

Z2M has helped me retain the small amount of hair I have left especially with motorised devices :-)

jajaP42, 2024-09-27:
@devtech-tech I understand. Thank you very much.

thecodymatthew, 2024-10-29:
Were you able to get the ZigBee Smart Sliding Window Pusher TS0601 by _TZ3210_5rta89nj model ZC-LP01 working with ZigBee Home Automation ZHA? @tribela

tribela, 2024-10-29:
No. Still not working and detected as smoke detector

aarongorka, 2024-11-02:
I've just drilled this in to my window so I guess I'm stuck with getting it integrated now. I've got a few bits of state by going off of these:

https://community.hubitat.com/t/moes-tuya-zigbee-smart-sliding-window-pusher/139006/22
https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2297321480

Working on the commands side of things now.

aarongorka, 2024-11-03:
I've got some bare minimum basic functionality working, but do note that positioning is broken. The code is also probably a mess but it works.

```python
"""
MOES Tuya Smart Sliding Window Pusher

Caveats:
  * Does not support specific positioning (and will probably crash if you try)
  * Doesn't expose current position, state, or anything else as sensors
  * Doesn't expose controls for calibration and changing "manual mode"

References:
  * https://github.com/zigpy/zha-device-handlers/issues/3233#issuecomment-2453169171
  * https://community.hubitat.com/t/moes-tuya-zigbee-smart-sliding-window-pusher/139006/22
  * https://github.com/Koenkk/zigbee2mqtt/issues/22844#issuecomment-2297321480
"""

from typing import Optional, Union
from zhaquirks import Bus
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    Ota,
    Time,
    PowerConfiguration,
    ZCLAttributeDef,
    # ZCLCommandDef,
)
from zigpy.zcl.clusters.closures import (
    WindowCovering,
)
from zigpy.quirks import CustomCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    LocalDataCluster,
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaDatapointData,
    TuyaManufCluster,
    TuyaCommand,
    TuyaNoBindPowerConfigurationCluster,
    TuyaNoBindPowerConfigurationCluster,
)
from zhaquirks.tuya.mcu import (
    TuyaMCUCluster,
)
from zigpy.zcl.clusters.security import IasZone
import logging
from zigpy.zcl import foundation
import zigpy.types as t

_LOGGER = logging.getLogger(__name__)
_LOGGER.debug("Logging intialised for pusher...")

ZONE_TYPE = 0x0001

class TuyaMoesPusherState(t.enum8):
    """Pusher state enum."""

    OPEN = 0x00
    CLOSED = 0x01
    STOP = 0x02
[... 339 lines cut ...]
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaNoBindPowerConfigurationCluster,
                    Identify.cluster_id,
                    TuyaMoesPusherIasZone,
                    TuyaMoesPusherWindowCluster,
                    TuyaMoesPusherWindowCovering,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```

tribela, 2024-11-03:
@aarongorka It works!
Opening and Closing is reversed for me. But that is not a big problem

tribakzero, 2026-02-24:
Made a new version that allows Slider functionality and makes it "easy" to invert the direction.
Oh, and it also removes the Smoke sensor :3

```python
"""
MOES Tuya Smart Sliding Window Pusher
Stable version with working manual commands
"""

from __future__ import annotations

from typing import Optional, Union
from zhaquirks import Bus
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    Ota,
    Time,
    PowerConfiguration,
)
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.security import IasZone
from zigpy.quirks import CustomCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    LocalDataCluster,
    EnchantedDevice,
    TuyaDatapointData,
    TuyaManufCluster,
    TuyaCommand,
    TuyaNoBindPowerConfigurationCluster,
    TUYA_SEND_DATA,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster
from zigpy.zcl import foundation
import zigpy.types as t
import logging

_LOGGER = logging.getLogger(__name__)

# ---------------------------
# Tuya Data Points
# ---------------------------

DP_BATTERY = 4
DP_CONTROL = 102
DP_POSITION = 104
DP_MOTOR_DIRECTION = 109

# ---------------------------
# TUYA MCU CLUSTER
# ---------------------------

class TuyaMoesPusherWindowCluster(TuyaMCUCluster):

    cluster_id = 0xEF00

    data_point_handlers = {
        DP_BATTERY: "_dp_2_attr_update",
        DP_CONTROL: "_dp_2_attr_update",
        DP_POSITION: "_dp_2_attr_update",
        DP_MOTOR_DIRECTION: "_dp_2_attr_update",
[... 225 lines cut ...]
                    TuyaMoesPusherMotorControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

```

To invert the direction just switch the comments from one block to the other like this:

```python
	    # ---- Open / Close / Stop ----
	    # if command_id == 0x00:  # Open
	    #     tuya_value = 1
	    # elif command_id == 0x01:  # Close
	    #     tuya_value = 0
	    # ---- Open / Close / Stop (INVERTED) ----
	    if command_id == 0x00:  # Open
	        tuya_value = 0
	    elif command_id == 0x01:  # Close
	        tuya_value = 1
```

Hope this helps you. I was trying to figure out the proper way to invert the direction and that's why the DP 109 was for, left it there in case someone figures the right id to use.

zsvideo, 2026-09-18:
i must be doing something wrong mine still shows up as a smoke detector