zigpy/zha-device-handlers #3243: [BUG] innr SP240 updates energy usage irregularly (power usage OK)

tvwerkhoven, 2024-07-05:
### Bug description

After using the innr SP240 plugs for a while, I notice the energy update (i.e. in kWh) is very irregular (e.g. sometimes only once a day at random times), while the power update (i.e. in kW) works almost real time (updates multiple times per minute). This is a bit annoying to diagnose energy consumption in my house.

### Steps to reproduce

1. Connect innr sp240
2. Read power/energy data
3. Get power data real time, energy data sporadically ~1x/day 

### Expected behavior

1. Connect innr sp240
2. Read power/energy data
3. Get power & energy data real time

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

Observed behaviour (energy meter updates irregularly):
[image]

Expected behaviour (other ZHA plug, energy meter updates regularly, in sync with power):
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4454, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "innr",
  "model": "SP 240",
  "class": "zhaquirks.innr.innr_sp240_plug.SP240"
}```

[... 653 lines cut ...]
            },
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}```

</details>

MrBearPresident, 2024-09-05:
I had the same problem you have. 
Then I switched to Z2MQTT  over there you have to option set some repporting settings. The standard setting for reporting are to coarse on the energy. 

It should also be possible to do this through https://github.com/mdeweerd/zha-toolkit

hhopke, 2024-09-08:
Having the same problem. Sometime `summation_delivered` is not updated for days.
Thanks for the tip @MrBearPresident, right now just a bit reluctant to spend a lot of energy into this.

MrBearPresident, 2024-09-09:
I've noticed after an firmware upgrade these thing revert back to the same settings. 

This are the setting I changed through Zigbee2MQTT 

Setting | OnOff - Minimum reporting change | Active Power  - Minimum reporting change| rmsCurrent - Minimum reporting change| cuurentSummDelivered - Minimum reporting change  
--- | --- | --- | --- |--- 
Original | 1| 5| 50| |
Corrected | 0 | 0,005| 0,005| 0,005

Jaapapa, 2024-10-30:
I didnt want to migrate to Zigbee2MQTT for this, so i tried settings the same with ZHA-toolkit. Posting here for anyone else with the same idea. I had some trouble installing due to bugs ([unrelated to the innr SP240 issues](https://github.com/mdeweerd/zha-toolkit/pull/263)) with the current HA versions but managed to get it working:

Steps:
- Install ZHA-toolkit.
- Figure out your device ieee through the device info screen for your device in HA.
- Use the HA included Developer Tools -> Actions to trigger a single action with the following yaml:
```
action: zha_toolkit.conf_report
data:
  ieee: 11:22:33:44:55:66:77:88
  cluster: 0x0702
  attribute: 0x0000
  event_success: my_read_success_trigger_event
  event_fail: my_read_fail_trigger_event
  event_done: my_read_done_trigger_event
  min_interval: 5
  max_interval: 3600
  reportable_change: 1
```

You can check the values using the following action:

```
action: zha_toolkit.conf_report_read
data:
  ieee: 11:22:33:44:55:66:77:88
  cluster: 0x0702
  attribute: 0x0000
  event_success: my_read_success_trigger_event
  event_fail: my_read_fail_trigger_event
  event_done: my_read_done_trigger_event
```
You should substitute your own ieee for your device.

epd5, 2025-01-08:
Having the same issue. I updated the min + max interval but the power summation delivered entity in HA still isnt updating, unless I manually go into ZHA and read the attribute. It seems like the quirk is preventing HA from pulling the MeteringClusterInnr values? I don't know how to begin diagnosing that though. Couldnt see anything obvious in the debug logs.

pajtaz, 2025-03-22:
I have the same issue with SP 240 latest firmware 1.9.30. I have tried everything (update firmware, changing values like suggested here, using custom quirk) but value only updates if I read it manually or upon restart of Home Assistant.

Additionally I still have the issue that ZHA is reading the total energy value 100 times larger than it actually is, i.e. 1.5kWh when it is actually 0.015kWh.

roelerps, 2025-03-25:
> I have the same issue with SP 240 latest firmware 1.9.30. I have tried everything (update firmware, changing values like suggested here, using custom quirk) but value only updates if I read it manually or upon restart of Home Assistant.
> 
> Additionally I still have the issue that ZHA is reading the total energy value 100 times larger than it actually is, i.e. 1.5kWh when it is actually 0.015kWh.
> 
> EDIT: It seems it is now working after applying this fix: [#3243 (comment)](https://github.com/zigpy/zha-device-handlers/issues/3243#issuecomment-2446995959)
> 
> I see the value updating now regularly. However it is still showing 100 times larger than it actually is.

I see the same behavior, also the current value is too high for the reported power..
The reported voltage is 235V, 0,2A and 12W. In the past 30 minutes or so I got 0,05kWh, something is wrong here..

Sunsurfer89, 2025-03-25:
Hey all, I just wanted to raise my hand, mentioning having the same problems as discussed here. For me, really a correct and automatically updated delivered energy (kWh) is necessary. (Calculation of 3D printing costs, here: energy costs)

I'm working with ZHA and do not want to Z2M just for that switch... So I tried the fix proposal by @Jaapapa but without any success. Measurement still only updates randomly on a daily basis, or if I go manually into the attribute listing... 

I'm hoping for a real fix 🙏

pajtaz, 2025-03-25:
> Hey all, I just wanted to raise my hand, mentioning having the same problems as discussed here. For me, really a correct and automatically updated delivered energy (kWh) is necessary. (Calculation [...]

I am doing exactly the same thing - trying to measure consumption of my 3D printers, just to have an idea how much they consume.

I did the fix that Jaapapa listed and it works. You have to do several things and in this order:
1. Update to latest firmware 1.9.30
2. Do the fix
3. Do "Reconfigure" for each plug.

You have to do reconfigure then it works.

Additionally, you and @roelerps if you want to fix the OTHER issue with latest 1.9.30 firmware where the actual measurement reported is 10 times larger (it was 100 times larger with older firmware, now it is 10 times larger), you can do what is suggested here: https://github.com/zigpy/zha-device-handlers/issues/3857#issuecomment-2745306464

That will create a sensor which you can then add to whatever dashboard you use and remove the "Summation delivered" from the standard one. See attached screenshots.

[image]

[image]

[image]

[image]

roelerps, 2025-03-25:
Yeah, for now I created an integral-helper calculating the energy based of the reported power to verify that the reported energy is indeed wrong.

roelerps, 2025-03-26:
For me the solution of [Jaapapa](https://github.com/Jaapapa) works for the updating.

I check the weird current/voltage/power reporting and it looks like the power factor (cos phi) is nog quite 1 (0,36). So the reporting of the current/voltage/power seem to be okay.

For the energy I see a factor 10x too big. Both the integral calculation based on the power and the energy meter show a facto 10x smaller energy usage. For now I will use the integral calculation based on the power but would like to see the reporting of the energy fixed somehow...

epd5, 2025-03-26:
I opened this issue on the OTA repository a while back in regards to the x10 problem - https://github.com/Koenkk/zigbee-OTA/issues/691

Schermbecker, 2025-05-17:
Same issue here. 🥹

closed 2025-11-22 by github-actions[bot]