zigpy/zha-device-handlers #3245: [BUG] Aqara Wireless Remote Switch H1 (WRS-R02) loses trigger `remote_button_short_press`

lindhe, 2024-07-07:
### Bug description

I have an [Aqara Wireless Remote Switch H1](https://www.aqara.com/en/product/wireless-remote-switch-h1/) (WRS-R02) paired via ZHA using a Home Assistant Connect ZBT-1 (previously known as Skyconnect). When I pair it, it connects and seems to work reasonably well (albeit with somewhat limited functionality). So I created an automation to control my lights and it seems to work. A couple of hours later my automation fails saying that the device has no trigger `remote_button_short_press`. And indeed, there are no longer any triggers available for the device.

I've tried re-pairing 3 times with the same behavior.

I get an error message in the Repairs section of Home Assistant:

> The automation "Bedroom wall switch" (`automation.bedroom_wall_switch`) is not active because its triggers could not be set up.
> 
> Error: `device does not have trigger ('remote_button_short_press', 'left')`.

### Steps to reproduce

1. Pair the device so it joins the Zigbee network.
1. Create an automation that uses the device's triggers, e.g. like this:

    <details><summary>automation.yaml</summary>
    <p>
    
    ```yaml
    id: '1718226515081'
    alias: Bedroom wall switch
    description: ''
    trigger:
      - device_id: dad78deb552a65df0cf4e9e267deab62
        domain: zha
        platform: device
        type: remote_button_short_press
        subtype: left
        id: Left
    condition: []
    action:
      - service: light.toggle
        metadata: {}
        data: {}
        target:
          device_id:
            - 70d61452e48167554749325bc78021e5
            - 409a93a8de755e2855c32f1a403fdf13
    mode: restart
    ```
    
    </p>
    </details> 

1. Wait for a few hours, in my case it happened within 12 hours.
1. Click a button, notice it does not work. Check the automation, see errors.

### Expected behavior

I expect all triggers to stay available if they are ever registered.

### Screenshots/Video

<details><summary>Screenshots</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
[... 357 lines cut ...]
2024-07-07 10:53:24.946 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/themes", "/config/themes", True)])`; This function will be removed in 2025.7 at custom_components/hacs/base.py, line 1182: self.hass.http.register_static_path(f"{URL_BASE}/themes", self.hass.config.path("themes")), please create a bug report at https://github.com/hacs/integration/issues
2024-07-07 10:53:24.955 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/frontend", "/config/custom_components/hacs/hacs_frontend", False)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 54: hass.http.register_static_path(f"{URL_BASE}/frontend", locate_dir(), cache_headers=False), please create a bug report at https://github.com/hacs/integration/issues
2024-07-07 10:53:24.963 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/iconset.js", "/config/custom_components/hacs/iconset.js", True)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 57: hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-07-07 10:53:24.967 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-07-07 10:53:24.972 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles", "/config/www/community", True)])`; This function will be removed in 2025.7 at custom_components/hacs/base.py, line 1162: self.hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-07-07 10:53:32.676 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.12/site-packages/vobject/base.py:736: SyntaxWarning: invalid escape sequence '\-'
  patterns['name'] = '[a-zA-Z0-9\-_]+'

2024-07-07 10:53:32.681 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.12/site-packages/vobject/base.py:1219: SyntaxWarning: invalid escape sequence '\;'
  s = s.replace("\\", "\\\\").replace(";", "\;").replace(",", "\,")

2024-07-07 10:53:32.681 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.12/site-packages/vobject/base.py:1219: SyntaxWarning: invalid escape sequence '\,'
  s = s.replace("\\", "\\\\").replace(";", "\;").replace(",", "\,")

2024-07-07 10:53:45.213 WARNING (MainThread) [zigpy.application] Zigbee channel 11 utilization is 95.12%!
2024-07-07 10:53:45.214 WARNING (MainThread) [zigpy.application] If you are having problems joining new devices, are missing sensor updates, or have issues keeping devices joined, ensure your coordinator is away from interference sources such as USB 3.0 devices, SSDs, WiFi routers, etc.
2024-07-07 10:53:46.664 ERROR (MainThread) [homeassistant.components.automation] Automation with alias 'Bedroom wall switch' failed to setup triggers and has been disabled: device does not have trigger ('remote_button_short_press', 'left')
2024-07-07 10:53:46.759 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.helpers.event. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from event directly at custom_components/hacs/base.py, line 630: self.hass.helpers.event.async_track_time_interval(, please create a bug report at https://github.com/hacs/integration/issues
```

</details>

### Additional information

I'm running Home Assistant Container with core version 2024.7.1.

poughkeepsee, 2024-09-19:
Hey, I randomly found this thread today. I had the same issue, what fixed it for me was to use a custom quirk: https://gist.github.com/xuxiaoqiao/6ed8dcb950f26337218d0507733970a0

May take a couple tries but eventually the device will be added using this custom quirk _aqara_h1_remote.RemoteH1DoubleRocker3_. Once mine started using the quirk it stopped dropping from the network.
[image]

lindhe, 2024-09-20:
Thank you for the tip! It started working for me too, actually. Just tried again after a recent update to Home Assistant, I'm guessing ZHA was updated. ¯\\\_(ツ)\_/¯

closed 2024-09-20 by lindhe