zigpy/zha-device-handlers #3248: [Device Support Request]

cherny-80, 2024-07-12:
### Problem description

The device is detected and connected in ZHA, but has no objects

### Solution description

Is it possible to fix this somehow

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_adkvzooy",
  "model": "TS0042",
  "class": "zhaquirks.tuya.ts0042.TuyaSmartRemote0042TO"
}

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.7.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.4",
    "docker": true,
[... 474 lines cut ...]
    }
  }
}

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2024-07-12:
Do you getting events from the switch in thee log ?
You have getting battery and the the normal LQI and switches is not getting and "objects" then they is sending events and dont have states like lights is having.

masi, 2024-09-06:
Which device?

closed 2025-08-12 by github-actions[bot]