zigpy/zha-device-handlers #3249: [Device Support Request] Tuya rainsensor TS0207_TZ3210_tgvtvdoc

wbmk78, 2024-07-12:
### Problem description

I have a TS0207 battery powered outdoor rain and light intensity sensor:
https://a.aliexpress.com/_EjJGdz9

The sensors are reported as a motion and open/close sensor. It seems like the movement sensor is triggered when rain is detected, but it seems like it is rarely updated.

### Solution description

Correctly mapped sensor output so a action can be created.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3210_tgvtvdoc",
  "model": "TS0207",
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.7.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.4",
    "docker": true,
    "arch": "aarch64",
[... 433 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Chrisni2, 2024-07-24:
I would be interested in this too

Andrik45719, 2024-07-25:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0207',
            manufacturerName: '_TZ3210_tgvtvdoc',
        },
    ],
    model: 'TS0207_rain_sensor',
    vendor: 'Tuya',
    description: 'TUYA TS0207 Rain Sensor RB-SRAN01',
    fromZigbee: [tuya.fz.datapoints, fz.battery],
    toZigbee: [],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.water_leak(), e.battery_low(), e.battery(),
        exposes.numeric('illuminance', ea.STATE).withDescription('Raw measured illuminance').withUnit('mV'),
        exposes.numeric('rainIntensity', ea.STATE).withDescription('Raw measured rain intensity').withUnit('mV')
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [101, 'illuminance', tuya.valueConverter.raw],
            [105, 'rainIntensity', tuya.valueConverter.raw]
        ],
    },
};

module.exports = definition;
`

Andrik45719, 2024-07-25:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0207',
            manufacturerName: '_TZ3210_tgvtvdoc',
        },
    ],
    model: 'TS0207_rain_sensor',
    vendor: 'Tuya',
    description: 'TUYA TS0207 Rain Sensor RB-SRAN01',
    fromZigbee: [tuya.fz.datapoints, fz.battery],
    toZigbee: [],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.water_leak(), e.battery_low(), e.battery(),
        exposes.numeric('illuminance', ea.STATE).withDescription('Raw measured illuminance').withUnit('mV'),
        exposes.numeric('rainIntensity', ea.STATE).withDescription('Raw measured rain intensity').withUnit('mV')
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [101, 'illuminance', tuya.valueConverter.raw],
            [105, 'rainIntensity', tuya.valueConverter.raw]
        ],
    },
};

module.exports = definition;
```

Chrisni2, 2024-07-25:
Please excuse my ignorance but I'm very new to HA, but where does this code go?

frankfrommelt, 2024-08-01:
This is the integration of the sensor for Zigbee2Mqtt.
For ZHA we will need a 'custom quirk'.
Sadly I have no idea how to create one.

Chrisni2, 2024-08-01:
Oops, told you I was a newbie.

pb-asbuilt, 2024-08-02:
I tried asking ChatGPT to port the Z2M converter to a ZHA quirk. I added it as a custom quirk, fixed the errors, but it doesn't work (no surprises there).

pbutterworth, 2024-08-02:
I tried asking ChatGPT to port the Z2M converter to a ZHA quirk. I added it as a custom quirk, fixed the errors, but it doesn't work (no surprises there).

hadeshimself, 2024-08-07:
There are more datapoints:
https://github.com/Koenkk/zigbee2mqtt/issues/23532

Has anyone began the work on the quirck file?

hadeshimself, 2024-08-09:
Here's a quirk that works, but does not expose the secondary measurement values to HA yet. If anyone wants to tackle that it would be great. Pretty much all DP values are DC voltages measured in mV, except for the cleaning reminder.

```
"""Quirk for TS0207 rain sensor."""

import zigpy.types as t
from typing import Any
from zigpy.profiles import zha 
from zigpy.quirks import CustomDevice, CustomCluster
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.security import IasZone
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster
from zhaquirks.tuya import TuyaManufCluster, DPToAttributeMapping

ZONE_TYPE = 0x0001

class TuyaSolarRainSensorCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            101: ("light_intensity", t.uint32_t, True),
            102: ("average_light_intensity_20mins", t.uint32_t, True),
            103: ("todays_max_light_intensity", t.uint32_t, True),
            104: ("cleaning_reminder", t.Bool, True),
            105: ("rain_sensor_voltage", t.uint32_t, True),
        }
    )

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "light_intensity",
        ),
        102: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "average_light_intensity_20mins",
        ),
        103: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "todays_max_light_intensity",
        ),
        104: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "cleaning_reminder",
        ),
        105: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "rain_sensor_voltage",
        ),
    }

    data_point_handlers = {
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        103: "_dp_2_attr_update",
        104: "_dp_2_attr_update",
        105: "_dp_2_attr_update",
    }

class TuyaIasZone(CustomCluster, IasZone):
    """IAS Zone for rain sensors."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Water_Sensor}

class TuyaRainSensor0207(CustomDevice):
    """TS0207 Rain sensor quirk."""

    signature = {
        MODELS_INFO: [("_TZ3210_tgvtvdoc", "TS0207")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    IasZone.cluster_id,
                    TuyaMCUCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaIasZone,
                    TuyaSolarRainSensorCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

pbutterworth, 2024-08-11:
Thanks for that @hadeshimself.
I've tested this out and can confirm that binary_sensor for moisture reports correctly, but only while the device is in pairing mode (green flashing light).
Once the device is out of pairing mode, we never hear from it again.
The device appears as red in the network visualisation:
[image]
Device diagnostics:
[zha-538856afef528ecfa5f3aa6173859876-_TZ3210_tgvtvdoc TS0207-aa565ac66d71ea26618fcdc8a739c403 (4).json](https://github.com/user-attachments/files/16576113/zha-538856afef528ecfa5f3aa6173859876-_TZ3210_tgvtvdoc.TS0207-aa565ac66d71ea26618fcdc8a739c403.4.json)

hadeshimself, 2024-08-12:
@pbutterworth, could you please try this one and let me know how it goes?

```python
"""Quirk for TS0207 rain sensors."""

import zigpy.types as t
from typing import Any, Type
from zigpy.profiles import zha 
from zigpy.quirks import CustomDevice, CustomCluster
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.security import IasZone
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster, TuyaEnchantableCluster
from zhaquirks.tuya import TuyaManufCluster, DPToAttributeMapping

ZONE_TYPE = 0x0001

class TuyaSolarRainSensorCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF65: ("light_intensity", t.uint32_t, True),
            0xEF66: ("average_light_intensity_20mins", t.uint32_t, True),
            0xEF67: ("todays_max_light_intensity", t.uint32_t, True),
            0xEF68: ("cleaning_reminder", t.Bool, True),
            0xEF69: ("rain_sensor_voltage", t.uint32_t, True),
        }
    )

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "light_intensity",
        ),
        102: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "average_light_intensity_20mins",
        ),
        103: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "todays_max_light_intensity",
        ),
        104: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "cleaning_reminder",
        ),
        105: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "rain_sensor_voltage",
        ),
    }

    data_point_handlers = {
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        103: "_dp_2_attr_update",
        104: "_dp_2_attr_update",
        105: "_dp_2_attr_update",
    }

class TuyaIasZone(TuyaEnchantableCluster, IasZone):
    """IAS Zone for rain sensors."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Water_Sensor}

class TuyaSolarRainSensor(CustomDevice):
    """TS0207 Rain sensor quirk."""

    signature = {
        MODELS_INFO: [("_TZ3210_tgvtvdoc", "TS0207")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    IasZone.cluster_id,
                    TuyaMCUCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaIasZone,
                    TuyaSolarRainSensorCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

hadeshimself, 2024-08-12:
@pbutterworth, could you please try this one and let me know?

```python
"""Quirk for TS0207 rain sensors."""

import zigpy.types as t
from typing import Any, Type
from zigpy.profiles import zha 
from zigpy.quirks import CustomDevice, CustomCluster
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.security import IasZone
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster, TuyaEnchantableCluster
from zhaquirks.tuya import TuyaManufCluster, DPToAttributeMapping, EnchantedDevice

ZONE_TYPE = 0x0001

class TuyaSolarRainSensorCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF65: ("light_intensity", t.uint32_t, True),
            0xEF66: ("average_light_intensity_20mins", t.uint32_t, True),
            0xEF67: ("todays_max_light_intensity", t.uint32_t, True),
            0xEF68: ("cleaning_reminder", t.Bool, True),
            0xEF69: ("rain_sensor_voltage", t.uint32_t, True),
        }
    )

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "light_intensity",
        ),
        102: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "average_light_intensity_20mins",
        ),
        103: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "todays_max_light_intensity",
        ),
        104: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "cleaning_reminder",
        ),
        105: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "rain_sensor_voltage",
        ),
    }

    data_point_handlers = {
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        103: "_dp_2_attr_update",
        104: "_dp_2_attr_update",
        105: "_dp_2_attr_update",
    }

class TuyaIasZone(TuyaEnchantableCluster, IasZone):
    """IAS Zone for rain sensors."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Water_Sensor}

class TuyaSolarRainSensor(EnchantedDevice):
    """TS0207 Rain sensor quirk."""

    signature = {
        MODELS_INFO: [("_TZ3210_tgvtvdoc", "TS0207")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    IasZone.cluster_id,
                    TuyaMCUCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaIasZone,
                    TuyaSolarRainSensorCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

pbutterworth, 2024-08-12:
@hadeshimself same symptoms.

hadeshimself, 2024-08-12:
Just to make sure you replaced the file and reloaded everything before adding the device again. You should see a TuyaNoBindPowerConfigurationCluster entry in the device clusters. 

Did you add this device previously to any other Zigbee hub? Tuya or zigbee2mqtt?

pbutterworth, 2024-08-12:
@hadeshimself, ah yes, excuse me; my mistake. Failed to save the file (I think) 🤦‍♂️.
Take 2:
* delete zha device
* update quirk file (and save it!)
* restart HA
* re-add the rain sensor to zha.

Outcome: sensor does not update at all, even when the device is in pairing mode (green flashing light).

PS: this device has not been added to any other hub. Tuya: don't have one. Z2M: I'm only using ZHA.

frankfrommelt, 2024-08-12:
Hey @hadeshimself ,

first of all thank you for that quirk and I guess it is almost the solution (at least for me...).

For me it looks different after adding the quirk, removing the device, restarting HA an re-adding the device it is working quite well.
Battery and Rain sensors are exposed.
And I have the "TuyaSolarRainSensorCluster".

But I do not see all the other attributes.

The diagnostics shows them as "unsupported attributes".

[image]
[image]

  "data": {
    "ieee": "**REDACTED**",
    "nwk": 6580,
    "manufacturer": "_TZ3210_tgvtvdoc",
    "model": "TS0207",
    "name": "_TZ3210_tgvtvdoc TS0207",
    "quirk_applied": true,
    "quirk_class": "Tuya_rain.TuyaSolarRainSensor",
    "quirk_id": null,
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": 176,
    "rssi": -56,
    "last_seen": "2024-08-12T09:08:46",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": {
        "logical_type": 2,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 128,
        "manufacturer_code": 4417,
        "maximum_buffer_size": 66,
        "maximum_incoming_transfer_size": 66,
        "server_mask": 10752,
        "maximum_outgoing_transfer_size": 66,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0402",
          "input_clusters": [
            "0x0000",
            "0x0001",
            "0x0004",
            "0x0005",
            "0x0500",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZ3210_tgvtvdoc",
      "model": "TS0207"
    },
    "active_coordinator": false,
    "entities": [
[... 158 lines cut ...]
            },
            "unsupported_attributes": {}**
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {
              "0x0002": {
                "attribute_name": "current_file_version",
                "value": 67
              }
            },
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

pbutterworth, 2024-08-12:
Quick update:
I dusted off an old Tuya gateway and added the rain sensor to that. Check for updates (none), version is V1.0.3.
I then removed it from the Tuya gateway, and added it back to ZHA, and observed no change.
Note: when adding the device to the Tuya gateway, once the device was added, the pairing mode terminated immediately (unlike ZHA where the pairing mode persists after the handshake is complete).
Also, it appears that the rain detection event in the device is an immediate push (tuya app responds immediately). Other attributes update periodically.

hadeshimself, 2024-08-12:
@pbutterworth did you use the code you received by email or did you copy it from the GitHub page? Because I edited the post with some updated code a few minutes later.

@frankfrommelt yeah, I don’t know how to expose those other attributes. There’s no standard to represent that kind of data, except from some energy related stuff. You would get 4 sensors for Voltage. How to tell them apart? If I had to pick one I would use the rain sensor voltage as it may indicate the intensity of the rain (?).

pbutterworth, 2024-08-12:
@frankfrommelt when you paired the device, did you notice whether the green light stopped flashing once the ZHA handshake was complete? Or, did the green light keep flashing for another couple of minutes?

hadeshimself, 2024-08-12:
@pbutterworth, make sure you are using that last code I posted, which includes the enchanted device code (it will have a TuyaNoBindPowerConfigurationCluster cluster shown on the UI). When you trigger your rain sensor do you see a red LED blink on the back? Also, make sure you reload everything and even clear the __pycache__ directory where you have your custom quirk files before adding the device again.

@frankfrommelt, you could try https://github.com/mdeweerd/zha-toolkit meanwhile to read the other values. Setup an action every minute or so to feed a sensor, something like:

```yaml
action: zha_toolkit.attr_read
data:
  ieee: binary_sensor.rain_sensor_moisture
  cluster: 61184
  attribute: light_intensity
  state_id: sensor.rain_sensor_light_intensity
  allow_create: true
  use_cache: true
```

Replace "rain_sensor" accordingly.

pbutterworth, 2024-08-12:
OK, another round of this. BTW @hadeshimself, feel free to step away from this; you've helped a lot already!

Re: red light on moisture detection: yes, once pairing mode has timed out, the red light flashes once when moisture is detected.

* Confirmed latest quirk, including `TuyaNoBindPowerConfigurationCluster`.
* Removed rain sensor from ZHA.
* Blitzed the __pycache__
* Restarted HA.
* Added the rain sensor (note: at this point the green light is still flashing):
[image]

* Moisture sensor updates state while the device is still in pairing mode (~30s latency):
[image]

* Once pairing mode times out, not further comms from the device.

[image]
[image]

Here's a log file that covers (sorry very noisy log file):
* adding the new device to ZHA
* triggering moisture while in pairing mode
* moisture cleared while in pairing mode
* pairing mode timed out
* tigger moisture again
[home-assistant_zha_2024-08-12T21-43-09.080Z.log](https://github.com/user-attachments/files/16592279/home-assistant_zha_2024-08-12T21-43-09.080Z.log)

hadeshimself, 2024-08-13:
@pbutterworth, also check if you have the latest coordinator firmware. At one point when debuging I couldn't pair it anymore and it only got back to normal after I updated the firmware. I'm using the Sonoff 3.0 dongle.

pbutterworth, 2024-08-13:
> @pbutterworth, also check if you have the latest coordinator firmware. At one point when debuging I couldn't pair it anymore and it only got back to normal after I updated the firmware. I'm using the Sonoff 3.0 dongle.

@hadeshimself nailed it! silabs firmware was 6.10, updated to 7.4, problem solved. 🤜🤛
Thanks for the quality assistance.

frankfrommelt, 2024-08-13:
> @frankfrommelt, you could try https://github.com/mdeweerd/zha-toolkit meanwhile to read the other values. Setup an action every minute or so to feed a sensor, something like:
>
> ```yaml
> action: [...]

That is a good workaround.

And this is how I create statistics from the States. As soon as I have a min/max value I will add some calculations to the state.

[image]

 - sensor:
   - name: sensor.regensensor_helligkeit_OK
     unique_id: sensor.regensensor_helligkeit_OK_13082024
     device_class: illuminance
     state: "{{ states('sensor.regensensor_helligkeit') }}"
     state_class: measurement
     unit_of_measurement: "lx"

hadeshimself, 2024-08-13:
@frankfrommelt yes, please let me know the max value you can get. I topped at around 5500.

frankfrommelt, 2024-08-13:
[image]

My max value was 4983 and now it is dropping rapidly as some heavy thunderstorms are approaching my area...

frankfrommelt, 2024-08-15:
OK, I had a maximum of 5492.

In comparison to a real light sensor you can see that the values delivered by the rain sensor are not realistic.
I guess it is only calculating a value from the current the solar panel is producing.

[image]

hadeshimself, 2024-08-15:
@frankfrommelt, my peak was 5880. But it's really not a good light measurement device. I've also compared it to a real light sensor and it's not even close to the same precision.

[image]

wbmk78, 2024-08-17:
@hadeshimself thanx a lot!
The second quirk seems to work nicely for rain measurement!

manuscho, 2024-08-25:
I struggle with the same problem for a water leak sensor ([Model](https://zigbee.blakadder.com/Neo_NAS-WS05B.html)) which also uses the TS0207 chip.

Same behavior: Works great with generic Tuya hub, connects smoothly with ZHA (through SkyConnect; water leak is correctly showing the first minutes after initial setup), but then never updates again.

The water leak sensor is explicitly mentioned [here](https://zigbee.blakadder.com/Neo_NAS-WS05B.html) as compatible with ZHA, which makes the whole thing a bit odd.

pbutterworth, 2024-08-25:
> I struggle with the same problem for a water leak sensor ([Model](https://zigbee.blakadder.com/Neo_NAS-WS05B.html)) which also uses the TS0207 chip.
>
> Same behavior: Works great with generic Tuya [...]

Updating the zigbee radio firmware solved these symptoms for me.

HaareistNase, 2024-08-26:
OMG.
Für einen Zigbee-Anfänger alles Bömische Dörfer^^
For a Zigbee-Beginner is all this bohemian Villages .. 

but thanx for your efford. I have the same Sensor and will try to get it work.

Manuito83, 2024-08-27:
Apologies, but here's a question from an new HA user that doesn't have the knowledge to properly follow what's being done here. All I know is that I bought this same rain/light sensor and can't see any details other than connection status and such (useless for the purpose I bought it for).

So here's the question: is this issue, and its solution, something that will be merged and eventually reach the end-user of Home Assistant? Is there a way I can follow the process? Or should I really look into making manual changes to the configuration (as explained above) to make this work in the medium-term future?

Thanks!

HaareistNase, 2024-08-27:
Maybe (I have not tryed) you can reach the sensor over the Cloud-Based Tuya Integration for Home-Assistant.
Very simple. Just install the Tuya-App on your phone/tablet, connect the sensor, and use the official Tuya-Integration in Home-Assist to reach the sensor.

The Drawback: Cloud-Based. If the Tuya-App does not work, because the Servers of Tuya failed. You get no Data.
If the Tuya-App decides to Update the Sensor every 30 Minutes, than you get Data every 30 Minutes, not every 5 Minutes.
And if Tuya decides to remove the support for this device: Elektro-Schrott.

Thats the reason we do the complicated way. No Cloud. No addiction to someone else.

frankfrommelt, 2024-08-27:
OK,, I was lucky and the device did what it was intended to (telling me if rain is falling or not)
out of the box with my Skyconnect ZigBee stick.
And, looking atz the data the device provides, from my perspective the rest of the information is mostly useless...
The light intensity reported by the device is depending on the amount of electricity that is being generated by the solar cell.
and, looking at my screen shot two weeks ago is telling you something like "it is night or day" (in comparison to a real light sensor).
The rain intensity is pretty similar as it is based on the resistance between the two metal wires on the outside.
This ist not rain intensity at all.
So, except it is great to have the ability to extract all these bits of information from a ZigBee device, from my opinion most information ist useless as it is VERY inaccurate.

hadeshimself, 2024-08-27:
I'm not happy with the moisture sensor not clearing fast enough (sometimes remains 'wet' hours after the rain stops, even with the sensor tilted more than 45 degrees). I was trying to correlate the rain sensor voltage with the moisture sensor to create a new binary sensor that is more accurate. That's only a plan, I don't have enough data yet.

I've noticed that during the night condensation might build on the sensor and that raises the sensor voltage considerably.

pbutterworth, 2024-08-27:
> I'm not happy with the moisture sensor not clearing fast enough (sometimes remains 'wet' hours after the rain stops, even with the sensor tilted more than 45 degrees). I was trying to correlate the [...]

What you need to combat this situation is a heated rain sensor, which is what would be used in an industrial situation. 
The aren't any wireless ones of these given their increase power demands.

HaareistNase, 2024-08-31:
After the hassle with the cheap Zigbee-Sensor, which data is only good for guesses (or when to open or close windows), I changed for a Cheap Ecowitt-Clone.

Its not that much more money, and you have installed it into Home-Assistant in minutes.
And it has a working Rain-Sensor^^

https://www.expondo.de/steinberg-systems-wifi-wetterstation-aussensensor-innensensor-10030583

[image]

budcalloway, 2024-08-31:
Hi ! I'm the new owner of the TS0207 and have data on rain but nothing on illuminance. It remains at 0 mV. Do you have encountered this issue ?
Thx

wsegatto, 2024-09-05:
ChatGPT told me that the quirk was not exposing all parameters mapped in the quirk as sensors. It even gave me a new code to try out but it didn't work, back to square one where we see the generic "Motion/Opening" entities. 
I was able to read the values through the "Manage zigbee device" option when I had the original quirk (non-GenAI'ed).
Perhaps it's just a matter of enhancing it a little bit to get all the other values?
I'm running HA Yellow with the Silicon Labs radio, but I'm new to zigbee so not sure the radio should impact the mapping of entities here since the attributes are readable... 

Any help is appreciated.

wsegatto, 2024-09-05:
> After the hassle with the cheap Zigbee-Sensor, which data is only good for guesses (or when to open or close windows), I changed for a Cheap Ecowitt-Clone.
>
> Its not that much more money, and you [...]

Do you have any plans to come to Brazil and bring me one? Wünderbar! :)

hadeshimself, 2024-09-05:
I can't believe the ChatGPT code didn't work! :-)

I didn't bother exposing those values because they are pretty much useless, but it's rather easy to do that. Most (all?) values are mV measurements, but I wouldn't bother using an electricity class for that. I would just implement them as AnalogInput and let the user deal with values as they wish.

I will try to work on that in the future.

wsegatto, 2024-09-05:
Hahhaha. :)
If you could provide some directions on how I could do it myself, it would
be much appreciated.
I look at the code and I don't see where the Wet/Dry parameter gets exposed
at all. :(
A kick-start and I could maybe make this work for us.
I think (only) a binary sensor wet/dry is not so appropriate sometimes e.g.
if you have dew in the night it would trigger the sensor and would stay
like that for a long time until the sun dries it out middle morning next
day. :D

hadeshimself, 2024-09-05:
The wet/dry sensor is exposed using a standard TuyaIasZone cluster. I only change the zone type to IasZone.ZoneType.Water_Sensor to make it appear as a water sensor. The other parameters are exposed on the Tuya manufacturer cluster (0xef00) and are exposed as-is for now.

simonk0606, 2024-09-18:
Hello, 
Sorry to arrive a bit late in that discussion, but I'm struggling to with those sunflowers sensor. 
I had it on ZHA, it's there but with motion and open sensor. I read all the post above, and I think I need to add one of this quirk somewhere, but I'm lost... Where ? How to do that ? 
Thanks by advance. 

Simon

frankfrommelt, 2024-09-18:
Hi Simon,

 �

well, from my perspective it is not worth the effort using a quirk.
One of the sensors you get out of the box (if I remember correctly, it is the motion one)
will tell you if the sensor is wet (so it is raining) or not.

All the additional values that you can get with the quirks (like brightness)
are very inaccurate and, from my perspective, not worth the effort.

Take care

 �

 � � � � � � � � � � � � � � � � � � � � � Frank

 �

 �

hadeshimself, 2024-09-18:
Exactly, the wet sensor is presented as a standard IasZone, not proprietary stuff from Tuya. I'm not exactly sure if you need this device to go through the "enchanted" device ritual to unlock. I remember mine only started working after I paired it to zigbee2mqtt (which performs the magic sequence). Later I modified the quirk to include that magic sequence, but I didn't have the chance to verify with a brand new device.

If yours is pairing but not blinking the green LED on the back when it triggers then you might want to try the quirk or pair it to zigbee2mqtt.

simonk0606, 2024-09-19:
Ok thanks you for your time. 
I choose the zigbee2mqtt, and after a little trouble to start that integration (thanks all those forum), it's all working... I just have to migrate all my ZigBee stuff from zha to z2m !

simonk0606, 2024-09-22:
And I'm back... It's in fact not working ! I loose the connection after few min or hour when lucky, than I struggled to pair it again... I think that a never have that famous green light flashing, expect when looking for pairing. 
So, I'm up for any tips, now I'm on z2m, I'm all listening! 

Thanks by advance

wsegatto, 2024-09-24:
> And I'm back... It's in fact not working ! I loose the connection after few min or hour when lucky, than I struggled to pair it again... I think that a never have that famous green light flashing, [...]

Not sure what radio you are using, but I have HA Yellow and I couldn't make it pair with Zigbee2MQTT, so I rolled back my HA backup and bought a Sonoff ZBDongle-E. Now I have ZHA with the embedded radio and Sonoff as with MQTT. It was a very different behavior.

[image]

[image]

Couple of things I had to do:
- Update the firmware of the dongle via Silicon Labs Flasher Add-on in HA:
Information: https://www.zigbee2mqtt.io/devices/ZBDongle-E.html
Firmware: https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl
Maybe you can do it here as well: https://darkxst.github.io/silabs-firmware-builder/

- After that for some reason I kept getting authentication errors and connection refused errors. It finally resolved and authenticated when I added the IP address of HA instead of localhost/127.0.0.1 or any other value. It worked straightaway with an user/pass maintained in the MQTT configuration tab, but I had troubles with user/pass from the Users in HA.

- Reboot HA.

- This did the trick. The battery took a while to appear, not sure why it was not there. Not all entities were there in the beginning, but it somehow "settled" after some hours. I have no clue what happened. All I did was to press the button at the back a couple of times (short press) to see what would happened and when I went to check the device multiple entities were there. Still little messy but much better than before with ZHA+Quirk.

[image]

aarongorka, 2024-10-26:
I updated the above quirk so that it exposes the `light_intensity` attribute as a new cluster, which will show up in HA as a sensor:

[image]

Note the `CALIBRATION_FACTOR` variable which I'm not sure is necessary or correct but the values seem better with it.

```python
"""Quirk for TS0207 rain sensors."""

import zigpy.types as t
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement
from zigpy.zcl.clusters.security import IasZone
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster
from zhaquirks.tuya import (
    TuyaLocalCluster,
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaNoBindPowerConfigurationCluster,
)
from zhaquirks import Bus
import logging

ZONE_TYPE = 0x0001

LOGGER = logging.getLogger(__name__)

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

    attributes = IlluminanceMeasurement.attributes.copy()

class TuyaSolarRainSensorCluster(TuyaMCUCluster, CustomCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF65: ("light_intensity", t.uint32_t, True),
            0xEF66: ("average_light_intensity_20mins", t.uint32_t, True),
            0xEF67: ("todays_max_light_intensity", t.uint32_t, True),
            0xEF68: ("cleaning_reminder", t.Bool, True),
            0xEF69: ("rain_sensor_voltage", t.uint32_t, True),
        }
    )

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "light_intensity",
[... 112 lines cut ...]
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaNoBindPowerConfigurationCluster,
                    TuyaIasZone,
                    TuyaSolarRainSensorCluster,
                    TuyaIlluminanceCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

inigortega, 2024-10-26:
And this quirk will be available in next zha uodates, or I need to install manually?
Best regards,

Sanaki, 2024-10-27:
It'll be available when someone submits a PR for it and it's accepted. Thus far no one has. That said, I like the newly posted quirk, though I have no idea how accurate its light sensor is. Still, that's the fault of the device, not the quirk.

hadeshimself, 2024-10-27:
Light sensor is pretty useless as it depends on mostly direct sunlight to
trigger a seasonable value. There’s no linear behavior as it gets in the
shade, for example.

Rain intensity as well. I could not correlate values with different
patterns of rainfall.

tasomaniac, 2024-10-28:
Just used the latest quirk from @aarongorka but unfortunately I did not have any other sensor visible. 

It is only the below. Also after adding it, it did not update at all. It was stuck with the first value.

[image]

Sanaki, 2024-10-28:
Did you reboot HA?

tasomaniac, 2024-10-28:
Yes after changing the file, I rebooted the whole device.

aarongorka, 2024-10-30:
> Just used the latest quirk from @aarongorka but unfortunately I did not have any other sensor visible.
> 
> It is only the below. Also after adding it, it did not update at all. It was stuck with the first value.
> <img alt="image" width="674" src="https://private-user-images.githubusercontent.com/763339/380811641-ad406c52-1ea7-4a2c-86ce-c15684d04c6f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyODM0MDMsIm5iZiI6MTczMDI4MzEwMywicGF0aCI6Ii83NjMzMzkvMzgwODExNjQxLWFkNDA2YzUyLTFlYTctNGEyYy04NmNlLWMxNTY4NGQwNGM2Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAzMFQxMDExNDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YjNmOGRhMjAzMGQ1MTRkNWVkZjkzNmNhZjdhZWE5YzA5MTA5Nzg0OWU1Zjg2Mzk4YTk5NzYzMWMyMmYxZWZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.HgzFAXsLm2jv1ArAZebwjdjN-7W15dRWr79s9gjzE0s">

Did you get this fixed? If not, could be interesting to see if you're getting any values from Manage Zigbee device > Clusters > TuyaSolarRainSensorCluster > light_intensity

tasomaniac, 2024-10-30:
I got it fixed. Upgrading the firmware fixed it. After the upgrade, the light sensor value also came back. This may be because of a restart, I'm not sure. But it all works well now.

Btw, I do have a "calculated" virtual Illuminance sensor based on the weather forecast. So far the values match. I will report more after observing it more.

[36 more replies]

closed 2025-02-26 by TheJulianJES