zigpy/zha-device-handlers #3251: [Device Support Request] TS000F Mumubiz Smart Switch with Temperature and Humidity Sensors

cleverdevil, 2024-07-12:
### Problem description

I recently acquired this device: https://www.amazon.com/dp/B0D63CR311

It is a smart on/off switch coupled with a temperature and humidity sensor. It can be added to ZHA, but only functions as an on/off switch, and does not report temperature or humidity. I believe that this device will require a custom quirk.

### Solution description

A custom quirk to support this device would be necessary.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3218_7fiyo3kv",
  "model": "TS000F",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2024.5.4",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Los_Angeles",
    "os_name": "Linux",
[... 571 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

brianacox, 2024-07-16:
Snap ... same issue.... have you found a work around yet?   BTW, how do you power your device?   I'm only using it as a 'main' (rather than battery) Zigbee temp sensor.... but only the 5v input seems to power it up.... I had hoped that 240v (regardless of whether I used it as a switch) would provide power.

cleverdevil, 2024-07-17:
Nope, no luck yet. I got the switch for free as part of Amazon Vine, so not in a big rush to get it to work. I am sure it will be supported eventually! Currently powering it with [this switching power adapter](https://www.coolerguys.com/products/coolerguys-fan-power-supply-12v-1a-output-3pin-or-4pin-pwm-connector). Simply cut off the connector, strip the wires, and connect them directly to the switch.

brianacox, 2024-07-22:
Mine started smoking with the USB cable in ..... seemed to be fine via the 240v input.... hey ho... waiting on a replacement at the moment....

Ezzob, 2024-10-08:
Custom quirk was created by user klibro here [TS000F_TZ3218_7fiyo3kv.py](https://github.com/klibro/zha-quirks/blob/e53ec83f044628f2f1eed239a0c0f7c4d7743056/TS000F_TZ3218_7fiyo3kv.py)

brianacox, 2024-10-09:
> Custom quirk was created by user klibro here [TS000F_TZ3218_7fiyo3kv.py](https://github.com/klibro/zha-quirks/blob/e53ec83f044628f2f1eed239a0c0f7c4d7743056/TS000F_TZ3218_7fiyo3kv.py)

Works perfectly... thank you!

tk200, 2024-10-11:
I have tried installing with the custom quirk, but the temp sensor still does not work - the diagnostics state it is unsupported. It is definately measuring it (the 2380 value is 23.8 deg, and increases as temp increases), but it doesn't display anywhere:

"0x0402": {
            "endpoint_attribute": "temperature",
            "attributes": {
              "0xfffd": {
                "attribute": "ZCLAttributeDef(id=0xFFFD, name='cluster_revision', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0002": {
                "attribute": "ZCLAttributeDef(id=0x0002, name='max_measured_value', type=<class 'zigpy.types.basic.int16s'>, zcl_type=<DataTypeId.int16: 41>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0000": {
                "attribute": "ZCLAttributeDef(id=0x0000, name='measured_value', type=<class 'zigpy.types.basic.int16s'>, zcl_type=<DataTypeId.int16: 41>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=True, is_manufacturer_specific=False)",
                "value": 2180
              },
              "0x0001": {
                "attribute": "ZCLAttributeDef(id=0x0001, name='min_measured_value', type=<class 'zigpy.types.basic.int16s'>, zcl_type=<DataTypeId.int16: 41>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0xfffe": {
                "attribute": "ZCLAttributeDef(id=0xFFFE, name='reporting_status', type=<enum 'AttributeReportingStatus'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0003": {
                "attribute": "ZCLAttributeDef(id=0x0003, name='tolerance', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              }
            },
            "unsupported_attributes": [
              0,
              "measured_value"
            ]
          }
        },

brianacox, 2024-10-11:
I downloaded the custom quirk, moved it to the correct directory and then simply did a reconfigure; that was sufficient, it seems, to reload the quirk.   It worked first time.   I wonder if you have a slightly different device?

tk200, 2024-10-11:
I think you may be right, in that it is a _slightly_ different device…

dvp-coms, 2025-05-16:
For what it's worth, I have a TYZGTH1CH-B1RF. I had the same issue where the latest quirk still wasn't showing the temp sensor but I could see it in the device settings TuyaTemperatureMeasurement > measured_value
I made a small tweak to the posted [TS000F_TZ3218_7fiyo3kv.py quirk](https://github.com/klibro/zha-quirks/blob/e53ec83f044628f2f1eed239a0c0f7c4d7743056/TS000F_TZ3218_7fiyo3kv.py) and simply replaced "102" with "402" in the two locations in the py script. For some reason this device is a bit different so the TuyaTemperatureMeasurement ID should be 402 instead of 102. After making this change and restarting HA I now see the temp sensor as an entity (I didn't need to delete/re-add the device to HA, but YMMV)

gormami, 2025-05-17:
Thank you, Thank you, Thank you!  I've been fooling around for a couple of days with this, and after I saw this I checked the signature and saw mine used 402 as well.  I changed it rather than added it, and it popped right up.  We must have been banging at the same time since you just posted this yesterday.  Fantastic timing.

gormami, 2025-05-17:
@dvp-coms You should definitely put up a PR to kilbro's repo, it is already linked all over the place.  I've come across a dozen or more threads asking about this thing and there is an "It didn't work for me" in every one.  I think there must be 2 models or firmware versions, or something, and your fix would help a lot of people.

dvp-coms, 2025-05-17:
Good call, just submitted the PR

closed 2025-11-22 by github-actions[bot]