zigpy/zha-device-handlers #3252: [Device Support Request] Tuya WHD02 (paired as TS0001 by _TZ3000_skueekg3)

Schnolle, 2024-07-14:
### Problem description

Hi,

my goal is to change the External Switch Type of TS0001 by _TZ3000_skueekg3 to _Momentary_, but it is not possible via _Manage Zigbee Device_.

Available quirks only affect other switches that already have the option to change to _Momentary_.

Could anyone help me out? Thank you!

### Solution description

Having the possibility to change external switch type

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MichalLytek, 2024-10-10:
I have a similar device `_TZ3000_zw7yf6yk` which expose via manage zigbee device a `TuyaZBExternalSwitchTypeCluster (Endpoint id: 1, Id: 0xe001, Type: in)` cluster with `external_switch_type (id: 0xd030)` attribute.

Same goes to the switch backlight: `TuyaZBOnOffAttributeCluster (Endpoint id: 1, Id: 0x0006, Type: in)` cluster and `backlight_mode (id: 0x8001)` attribute.

Would love to read some doc about writing custom quirks, so I could test those ideas 👀

BugsBunny1403, 2024-12-13:
Hello, I would be interested in the _TZ3000_skueekg3 module support to control as a switch and not as a lamp.

wise-rice, 2025-02-10:
The same problem for me with a TS000F by _TZ3000_skueekg3
it is the same device.

wenestvedt, 2025-02-27:
Same devices having the same issue here, sadly.

closed 2025-09-04 by github-actions[bot]