zigpy/zha-device-handlers #3254: [BUG] Overriding "built-in" quirks

noseshimself, 2024-07-15:
### Bug description

If there is a device specific quirk in the zigpy distribution it is impossible to override it with a local quirk file in home assistant.

### Steps to reproduce

1. Install a "TS0601 by _TZE204_nklqjk62" garage door opener
2. Watch it showing up without any entity besides its firmware version.
3. Add a local quirk (e.g. https://github.com/zigpy/zha-device-handlers/issues/2635)
4. 
[image]

### Expected behavior

my local quirk taking precedence

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

rdehuyss, 2025-01-03:
@noseshimself Were you able to resolve this?

noseshimself, 2025-01-07:
No.

Doesn't bite me anymore. I added a second radio and z2m and put everything that needs quirks in zha on z2m and it just worked out-of--the-box.

rdehuyss, 2025-01-07:
Well, I tried a little bit after I posted my question here and it worked immediately out of the box. 

Perhaps something was wrong in your home assistant config? 

For @zigpy: I can confirm this works and can be closed.

TheJulianJES, 2025-01-27:
Home Assistant Core 2025.2.0 should also improve custom quirks behavior overriding integrated quirks, though you should only really use custom quirks for development, until they're in the quirks repo and thus included with HA.

closed 2025-01-27 by TheJulianJES