zigpy/zha-device-handlers #3255: Tuya Siren (new) TS0601 by _TZE204_hcxvyxa5

Martijnsx, 2024-07-15:
### Problem description

I have a tuya zigbee siren. It is a new model i guess. How can i get this working with ZHA?
I can only see it in home assistant but it has no entities and can not be controlled (alarm fired).

TS0601
by _TZE204_hcxvyxa5

### Solution description

That i can start and stop the alarm in home assistant and read it's status.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Conchar, 2024-07-17:
I recently received this item too, from here: https://www.aliexpress.us/item/3256806983895286.html
All of my previous zigbee purchases just magically worked when paired with ZHA, so I had no idea of the constantly vigilant world of wonderful developers here until researching how to get this device to work.  Thanks to all for your hard work!

I didn't see the original poster copy & paste the device signature, so I hope this helps:

>     "signature": {
>       "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
>       "endpoints": {
>         "1": {
>           "profile_id": "0x0104",
>           "device_type": "0x0051",
>           "input_clusters": [
>             "0x0000",
>             "0x0004",
>             "0x0005",
>             "0xef00"
>           ],
>           "output_clusters": [
>             "0x000a",
>             "0x0019"
>           ]
>         },
>         "242": {
>           "profile_id": "0xa1e0",
>           "device_type": "0x0061",
>           "input_clusters": [],
>           "output_clusters": [
>             "0x0021"
>           ]
>         }
>       },
>       "manufacturer": "_TZE204_hcxvyxa5",
>       "model": "TS0601"
>     },
>

TheOddPerson, 2024-07-23:
I have a similar device
`"signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0403",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0x0006",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "_TZE204_hcxvyxa5",
      "model": "TS0601"
    },`

I added ("_TZE204_hcxvyxa5", "TS0601") to 
ts0601_siren.py in TuyaSirenGPP_NoSensors and the device redected with an on/off entity. When toggled on the siren sounds for 1 second and the toggle turns off automatically.

I'm reading through the quirks documentation to see if its within my grasp to make it work.

yurisat, 2024-07-29:
With this implementation in Node-Red it was possible to increase the tempo and change the melody. It wasn't done by me, unfortunately I don't remember where I got it. But it works, it's a way forward.

`[{"id":"0c1810def209b6a6","type":"subflow","name":"Tuya Zigbee Siren","info":"**IEEE:** Is used to identify the siren to use. It can be found in the **Zigbee Info** for the device.\r\nExample: **a1:b2:34:5c:5d:67:ef:gh**\r\n\r\n**Sound:** Chooses the sound to play.\r\n\r\n**Duration:** Defines how long the siren will sound for in seconds.\r\n\r\n**Activate Siren:** Choses whether to fire the siren or not, it can also be used to stop an active siren. If you change other settings but choose to not fire the siren, those settings will still be applied for the next event unless they are again changed.\r\n\r\n\r\n# Table of Sounds\r\n - ## ID Sound\r\n - 0\tDoorbell Chime\r\n - 1\tFur Elise\r\n - 2\tWestminster Chimes\r\n - 3\tFast double door bell\r\n - 4\tWilliam Tell Overture\r\n - 5\tTurkish March\r\n - 6\tSafe/Security Alarm\r\n - 7\tChemical Spill Alert\r\n - 8\tPiercing Alarm Clock\r\n - 9\tSmoke Alarm\r\n - 10\tDog Barking\r\n - 11\tPolice Siren\r\n - 12\tDoorbell Chime (reverb)\r\n - 13\tMechanical Telephone\r\n - 14\tFire/Ambulance\r\n - 15\t3/1 Elevator\r\n - 16\tBuzzing Alarm Clock\r\n - 17\tSchool Bell\r\n\r\n","category":"","in":[{"x":80,"y":160,"wires":[{"id":"afe31aab2b1daaab"}]}],"out":[{"x":1290,"y":160,"wires":[{"id":"920a10178ba2a0b4","port":0}]}],"env":[{"name":"siren_ieee","type":"str","value":"xx:xx:xx:xx:xx:xx:xx:xx","ui":{"icon":"font-awesome/fa-bullhorn","label":{"en-US":"Zigbee IEEE*"},"type":"input","opts":{"types":["str"]}}},{"name":"siren_sound","type":"str","value":"0","ui":{"icon":"font-awesome/fa-file-sound-o","label":{"en-US":"Sound (0-18)"},"type":"select","opts":{"opts":[{"l":{"en-US":"Doorbell Chime"},"v":"0"},{"l":{"en-US":"Fur Elise"},"v":"1"},{"l":{"en-US":"Westminster Chimes"},"v":"2"},{"l":{"en-US":"Fast double door bell"},"v":"3"},{"l":{"en-US":"William Tell Overture"},"v":"4"},{"l":{"en-US":"Turkish March"},"v":"5"},{"l":{"en-US":"Safe/Security Alarm"},"v":"6"},{"l":{"en-US":"Chemical Spill Alert"},"v":"7"},{"l":{"en-US":"Piercing Alarm Clock"},"v":"8"},{"l":{"en-US":"Smoke Alarm"},"v":"9"},{"l":{"en-US":"Dog Barking"},"v":"10"},{"l":{"en-US":"Police Siren"},"v":"11"},{"l":{"en-US":"Doorbell Chime (reverb)"},"v":"12"},{"l":{"en-US":"Mechanical Telephone"},"v":"13"},{"l":{"en-US":"Fire/Ambulance"},"v":"14"},{"l":{"en-US":"3/1 Elevator"},"v":"15"},{"l":{"en-US":"Buzzing Alarm Clock"},"v":"16"},{"l":{"en-US":"School Bell"},"v":"17"}]}}},{"name":"siren_duration","type":"num","value":"30","ui":{"icon":"font-awesome/fa-clock-o","label":{"en-US":"Duration (s)"},"type":"input","opts":{"types":["num"]}}},{"name":"siren_vol","type":"str","value":"0","ui":{"icon":"font-awesome/fa-volume-up","label":{"en-US":"Volume (0-2)"},"type":"select","opts":{"opts":[{"l":{"en-US":"Low"},"v":"0"},{"l":{"en-US":"Medium"},"v":"1"},{"l":{"en-US":"High"},"v":"2"}]}}},{"name":"activate_siren","type":"str","value":"1","ui":{"label":{"en-US":"Activate"},"type":"select","opts":{"opts":[{"l":{"en-US":"On"},"v":"1"},{"l":{"en-US":"Off"},"v":"0"}]}}}],"meta":{},"color":"#FFCC66","inputLabels":["Activate Deactivate or Change settings."],"outputLabels":["Output for debugging or just chaining."],"icon":"font-awesome/fa-bullhorn"},{"id":"e965ea08b86a79a7","type":"api-call-service","z":"0c1810def209b6a6","name":"set sound","server":"528411d506baef5f","version":5,"debugenabled":false,"domain":"zha","service":"set_zigbee_cluster_attribute","areaId":[],"deviceId":[],"entityId":[],"data":"{\t   \"ieee\": msg.siren_ieee,\t   \"cluster_type\": in,\t   \"endpoint_id\": 1,\t   \"cluster_id\": 6,\t   \"attribute\": 1126,\t   \"value\": msg.siren_sound\t}","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":460,"y":160,"wires":[["00ccd135e76b49f1"]]},{"id":"00ccd135e76b49f1","type":"api-call-service","z":"0c1810def209b6a6","name":"set duration (s)","server":"528411d506baef5f","version":5,"debugenabled":false,"domain":"zha","service":"set_zigbee_cluster_attribute","areaId":[],"deviceId":[],"entityId":[],"data":"{\t   \"ieee\": msg.siren_ieee,\t   \"cluster_type\": in,\t   \"endpoint_id\": 1,\t   \"cluster_id\": 6,\t   \"attribute\": 615,\t   \"value\": msg.siren_duration\t}","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":640,"y":160,"wires":[["92e8833d9b061a79"]]},{"id":"92e8833d9b061a79","type":"api-call-service","z":"0c1810def209b6a6","name":"set volume (0-2)","server":"528411d506baef5f","version":5,"debugenabled":false,"domain":"zha","service":"set_zigbee_cluster_attribute","areaId":[],"deviceId":[],"entityId":[],"data":"{\t   \"ieee\": msg.siren_ieee,\t   \"cluster_type\": in,\t   \"endpoint_id\": 1,\t   \"cluster_id\": 6,\t   \"attribute\": 1140,\t   \"value\": msg.siren_vol\t}","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":840,"y":160,"wires":[["920a10178ba2a0b4"]]},{"id":"920a10178ba2a0b4","type":"api-call-service","z":"0c1810def209b6a6","name":"play sound","server":"528411d506baef5f","version":5,"debugenabled":false,"domain":"zha","service":"set_zigbee_cluster_attribute","areaId":[],"deviceId":[],"entityId":[],"data":"{\t   \"ieee\": msg.siren_ieee,\t   \"cluster_type\": in,\t   \"endpoint_id\": 1,\t   \"cluster_id\": 6,\t   \"attribute\": [... cut ...] ,"statusYear":"hidden","statusMonth":"short","statusDay":"numeric","statusHourCycle":"default","statusTimeFormat":"h:m","enableGlobalContextStore":true},{"id":"4b9dca30f8055f6a","type":"subflow:0c1810def209b6a6","z":"d18f56c86b4f8991","name":"","env":[{"name":"siren_ieee","value":"a4:c1:38:38:ab:43:1e:6d","type":"str"},{"name":"siren_sound","value":"14","type":"str"},{"name":"siren_duration","value":"90","type":"num"},{"name":"siren_vol","value":"2","type":"str"}],"x":570,"y":360,"wires":[[]]}]`

JAKAMI99, 2024-08-23:
Also bought this device and would love to see it get supported

upsuper, 2024-09-14:
It seems that the existing quirk for the siren without sensors works well with this new device, so I raised a PR above to add this into the list.

Martijnsx, 2024-09-14:
How do we get or install this quirk or do we have to wait for it to be
implemented?

upsuper, 2024-09-14:
You need to setup custom quirks. You can for example follow [this guide](https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226/2).

After that, download the [file](https://github.com/zigpy/zha-device-handlers/raw/5c993588c79b7708f9ecfc8ad893df6e51d97924/zhaquirks/tuya/ts0601_siren.py) changed in the PR and put it in the directory you created for custom quirks.

Then you probably need to restart Home Assistant to take effect.

upsuper, 2024-09-15:
@Martijnsx From your description, I'd say it's working. If you want to change its alarm duration, you can update it as shown below:
[image]

Martijnsx, 2024-09-15:
Thank you very much, that worked!

closed 2024-09-24 by TheJulianJES

JAKAMI99, 2024-09-25:
Working!

At first, I enabled custom quirks in the configuration.yaml in home assistant because I didn't want to wait for the :main update
`zha:
   enable_quirks: true
   custom_quirks_path: /config/zha_quirks/`

Then I downloaded the custom quirk:
`➜  zha_quirks wget https://raw.githubusercontent.com/zigpy/zha-device-handlers/5c993588c79b7708f9ecfc8ad893df6e51d97924/zhaquirks/tuya/ts0601_siren.py`

I rebooted, removed and added the  `_TZE204_hcxvyxa5` device.

Now the device has a switch. If turned on, the device makes a siren noise for ~500ms and turns off again,

Trough "manage zigbee device" I could change some settings by choosing cluster "TuyaMCUSiren" through the attributes

- `alarm_duration (id: 0x0267)` sets the alarm duration in seconds.  
- The volume can be changed trough `volume (id: 0x0474)` by setting the value to `0`(`low`) , `1`(`medium`) or to `2`(`high`)  
- The melody can be set trough `melody (id: 0x0466)` where the value can be from `0` `36` (`37` seems to be a duplicate with 0) which varies from classical songs like "Für Elise" (`24`) to notification ish/doorbell sounds (`11`) to the three alarm siren sounds (`0`-`2`) where `0` is the most alarm one, but they really have a big cheap toy vibe 👶

_Melody (dropdown-menu), volume (dropdown-menu) and duration (slider) should be added to the GUI, so everyone can access it._

Thanks for the quick implementation! ✅

GeoffreyMcGuire, 2025-01-10:
Thanks guys for this, got mine today, and with this help and quirk file it's working fine.

ralfbergs, 2025-10-17:
I have this device since a couple of days, and I'm amazed (in a negative way) about the stupid firmware design:

- It doesn't seem possible to sound the alert for an unlimited time. The maximum duration is 380 sec.
- It doesn't seem possible to directly sound the alert with a specific melody. You first have to change the alert (which makes the siren sound the alert for a few seconds), and then you have to switch it on

So my intended usecase seems impossible:

- Short "ding" for when a door is opened.
- Real "alert" sound that sounds for an unlimited time for when something happens, like a water leakage alert.

Even if I accept that the siren can't sound for an unlimited time, my usecase still can't be implemented.

This is very disappointing, because from the hardware I like this siren. And it's actually pretty loud, I would say, comparable to the Samsung SmartThings/Sercomm sirens.

GeoffreyMcGuire, 2025-10-20:
I will dig out the code when I get a sec, but for my house alarm I do reset the ringer and duration whenever the alarm is armed.  I also have two more for door bells that are just defaulted to door bell and seem to survive a reboot,  think they are about £6 delivered so you get what you pay for lol.

PR #3361 (2024-09-24): Add Tuya Siren TS0601 `_TZE204_hcxvyxa5` variant
## Proposed change

This PR adds siren TS0601 _TZE204_hcxvyxa5 into the existing quirk of `TuyaSirenGPP_NoSensors`, and fixes #3255 and fixes #3228.

I have confirmed locally that it enables the siren to be toggled like other similar sirens, and I am able to adjust melody and duration via "Manage zigbee device".

## Additional information

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works