zigpy/zha-device-handlers #3260: [Device Support Request] TS0601 by _TZE204_cirvgep4

Keltere, 2024-07-16:
### Problem description

The device in question doesn't report any information

the device:
https://www.aliexpress.com/item/1005006861229185.html?spm=a2g0o.order_list.order_list_main.40.faef1802VfXFjX

There is already a PR merged for zigbee converter:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/7387

### Solution description

Add support for this device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_cirvgep4",
  "model": "TS0601",
  "class": "zigpy.device.Device"
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2024.7.1",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Bucharest",
    "os_name": "Linux",
    "os_version": "5.14.0-452.el9.x86_64",
[... 368 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

lipskisenator, 2024-07-26:
I  have the same issue with device FLY_ZTH08_E ("manufacturer": "_TZE204_cirvgep4",  "model": "TS0601") - I've tried to add it to ts0601_sensor_py ( class TuyaTempHumiditySensorVar04(CustomDevice) ) and than sensors of temp and hum works , but date and time on display wasn't synchronized

VasylSamoilov, 2024-08-23:
@lipskisenator you can try using quirk from https://github.com/zigpy/zha-device-handlers/issues/2304 , adding your proper id to it, but it will not tolerate backlight button press (device will hang and drain battery, probably waiting for some reply)

Keltere, 2025-04-29:
@TheJulianJES Wasn't really completed

[image]

Still no full support for it.

TheJulianJES, 2025-04-29:
@Keltere Did you reconfigure the device (or re-paired it) after updating HA to the latest version?

Keltere, 2025-04-29:
> [@Keltere](https://github.com/Keltere) Did you reconfigure the device (or re-paired it) after updating HA to the latest version?

Yes, I had another tuya device which was correctly reconfigured except this one for which the only thing working it's switching from celsius to Fahrenheit but to apply the change it seems a button press is needed.

PR #3817 (2025-01-31): Fix Tuya temp/humidity sensor quirks (`_TZE200_upagmta9`, ...)
## Proposed change

Add Tuya `tze200 cirvgep4` and combine quirks to match z2m.

See: https://github.com/Koenkk/zigbee-herdsman-converters/blob/09cfce5153d3931c7cc009aa3ae3107d9b394f5c/src/devices/tuya.ts#L1073

## Additional information

Closes: #2577
Closes: #2854 
Closes: #2304 
Closes: #3260 
Closes: #2725 
Closes: #2929 

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works