zigpy/zha-device-handlers #3261: IKEA Styrbar Remove Control N2 issues

abacao, 2024-07-17:
I have a Ikea Styrbar remote. It was not easily pair-able and only the arrow keys register.
[image]

I have tried the update without success and the device is using a "quirk" different from everyone else.

Any suggestions?

_Originally posted by @abacao in https://github.com/zigpy/zha-device-handlers/issues/2841#issuecomment-2232910191_

rcswift, 2024-08-04:
I'm also having this same issue with my Styrbar Remote:
[image]

In my case, the firmware is version `0x02040016`.

I have tried both the 'reconfigure' option from the webpage and removing and re-pairing the switch but it still shows `zigpy.quirks.v2.CustomDeviceV2` every time.

The signature shown for my device is:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc57",
        "0xfc7c"
      ],
      "output_clusters": [
        "0x0003",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "Remote Control N2",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}
```

MattWestb, 2024-08-04:
The "`class": "zigpy.quirks.v2.CustomDeviceV2"` is the current working version for the  last production version like you is having.
I getting dim up and down OK and also On/Off but i have not looking if all other commands is OK then i using it for direct controlling lights with bindings.

If not getting all commands do one reconfigure but taking out the battery and putting it back  before doing the reconfigure and look you  is getting green on all cluster (if not all commands can being missing).

Sakura-sx, 2024-09-14:
> The "`class": "zigpy.quirks.v2.CustomDeviceV2"` is the current working version for the  last production version like you is having.
>
> I getting dim up and down OK and also On/Off but i have not [...]

I have a question, did you reconfigure it? How did you sync it? I have the same issue as the other two people. For sync I have been clicking the sync button 4 times in a 5 second time window. Have you done anything different? Also, do you have the 5 button version or the 4 button version? I am using the 4 button one.

abacao, 2024-09-18:
I was able to sync it by multiple retries. Dont really understand what happened.

Current working status:
[image]

No quirk was found

closed 2024-09-18 by abacao

MattWestb, 2024-09-18:
Look in manager zigbee device (3 dots right from reconfigure) and  then signature tab you getting the quirk that is loaded  (i was not knowing it was moved from the device card).