zigpy/zha-device-handlers #3262: [Device Support Request]   NIKO 552-72004 - Wireless battery-powered switch, 4-fold, Zigbee

bdjPenguin, 2024-07-18:
### Problem description

https://www.niko.eu/en/products/niko-home-control/products-for-traditional-wiring/wireless-battery-powered-switch-4-fold-zigbee-productmodel-niko-fd091fe8-7983-4dc8-bb0a-cffaf91e3f04
Commutators are missing from the entities
This model was not found in the existing documentation.

### Solution description

I would like to see the 4 commutators in the HA interface.
Only the battery status is available.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4703, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05",
        "0x1000",
        "0xfc00",
        "0xfc01"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [],
      "output_clusters": [
        "0x0003",
        "0x0006",
        "0x0008"
      ]
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [],
      "output_clusters": [
        "0x0003",
        "0x0006",
        "0x0008"
      ]
    },
    "4": {
      "profile_id": "0x0104",
[... 530 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

klegrand, 2024-08-22:
hi bdjPingiun,

I've the same issue with a Niko Zigbee battery powered switch. For a reason I can't find the trigger when I push the button.
Did you solved it?

Thx

bdjPenguin, 2024-08-27:
Hello, I didn't have any solution up to now.
I'm still expecting a similar behaviour as the "Tradfri Remote Control" from Ikea.
Unfortunately, no success up to now.

jankdb, 2024-10-14:
Having the same issues are reported. Looking forward to the support.

closed 2025-08-12 by github-actions[bot]