zigpy/zha-device-handlers #3263: [Device Support Request] TS0603 by _TZE608_c75zqghm

G4LLY, 2024-07-18:
### Problem description

Hello, I get this garage door opener from Aliexpress and once connected to ZHA, is does not show the proper command / informations:

[image]

### Solution description

As a door opener it should have an open-close command and a sensor to show the door state (open / close)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE608_c75zqghm",
  "model": "TS0603",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.7.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
[... 392 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I'm new to HA and Zigbee, still on the learning path, any help is welcom :)

G4LLY, 2024-07-19:
Was able to make it work by adapting an existing quirks fro T601:
Probably not the most optimal code as this is a first for me, but it work ^^ 

Don't hesitate to correct / advice !

```
"""Tuya based cover and blinds."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, Ota, Scenes, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
    TuyaOnOff,
)

from zhaquirks.tuya.ts0601_dimmer import TuyaOnOffNM

ZONE_TYPE = 0x0001

class ContactSwitchCluster(TuyaLocalCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaGarageManufCluster(TuyaMCUCluster):
    """Tuya garage door opener."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF05: ("dp_5", t.uint32_t, True),
            0xEF0B: ("dp_11", t.Bool, True),
            0xEF0C: ("dp_12", t.enum8, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        # garage door trigger ¿on movement, on open, on closed?
        1: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
        ),
        2: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_2",
        ),
        3: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
            endpoint_id=2,
        ),
        4: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_4",
        ),
        5: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_5",
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_11",
        ),
        # garage door status (open, closed, ...)
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_12",
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class TuyaGarageSwitchTO(CustomDevice):
    """Tuya Garage switch."""

    signature = {
        MODELS_INFO: [
            ("_TZE608_c75zqghm", "TS0603"),
                                           
                                           
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    ContactSwitchCluster
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }
```

gs-danielnikonczuk, 2024-08-29:
I used your quirk for HA ZHA, and it worked nicely, but only for the switch to short the garage engine to send an open/stop/close signal. Unfortunately, it didn't work for the sensor. I'm not sure if I have a broken device or need additional adaptation. 
My devices is similar, but somehow a little bit different. Different Manufacturer, for sure.
Device: **TS0603**
Man: **_TZE608_fmemczv1**

G4LLY, 2024-08-29:
Hi,

for me it appear with 2 entities for the devices,  The fact it is presented as a "smart plug" andnot a door opener, the smart plug status is not linked to the sensor status, to work around this I have set 2 cards in the frontend and hide the smart plug status:

Lovelace cards:
[image]

ZHA device
[image]

gs-danielnikonczuk, 2024-08-29:
I see two entities, but the state doesn't change whether I close the sensor. 
[image]

G4LLY, 2024-08-29:
Works for me, the door sensor is properly plug on the S1 and S2 plugs ?

[image]

gs-danielnikonczuk, 2024-08-29:
Yep. When I put the magnetic sensor parts together, nothing happens. I have other similar sensors, and they work very snappy, like in an instant.
[image]
[image]

G4LLY, 2024-08-29:
seems to be S3 and S4 at your side.

I remember I had to restart HA couple of time to make it work.

gs-danielnikonczuk, 2024-08-29:
You wouldn't believe how many restarts I'm after. I wonder if this is a matter of broken sensor/device or just the converter/quirk matter in ZHA.

gs-danielnikonczuk, 2024-08-29:
Okay, I confirm that my sensor is busted. I shorted the S3 and S4, which popped up properly in the HA entity view. Sorry for bothering you. Good job with the quirk! It is sufficient for all the automations I wanted with the garage gate.

It could just come broken already. 💩 happens 😂

G4LLY, 2024-08-29:
Happy you figured it out ! 

Cheers

jlmadrugac, 2024-09-02:
Hi all.
I've just added your quirk and working (only changing signature).

The problem is the presentation of the icon... it apears as a on/off icon and I'd love have it as my other garage door opener (wifi) that has an arrow to open or close (see pics)

ZB TS0603 with G4LLY quirk
[image]

Loratap Tuya wifi connected garage opener:
[image]

Any clues of how to represent this as the second one?

MattW2, 2024-11-09:
I was able to make a functional cover control by combining the customer quirk posted by @G4LLY https://github.com/zigpy/zha-device-handlers/issues/3263#issuecomment-2238991683 and the following cover template

```
cover:
  - platform: template
    covers:
      garage_door_left:
        device_class: garage  # Optional: gives a garage door icon
        friendly_name: "Garage Door Left"
        # Commands to open the cover
        open_cover:
          - service: switch.toggle
            target:
              entity_id: switch.garage_door_left_switch
        
        # Commands to close the cover
        close_cover:
          - service: switch.toggle
            target:
              entity_id: switch.garage_door_left_switch
        
        # State of the cover based on binary sensor status
        value_template: "{{ is_state('binary_sensor.garage_door_left_status', 'on') }}"
        # Define cover position based on sensor (for example, open if status is 'on')
```

Auenlaender81, 2025-01-08:
Vielen Dank.
Funktioniert wunderbar! 👍

shecobon, 2025-04-02:
Thanks for the workaround. what a legend.

meleath, 2025-12-06:
@G4LLY 
Nicely done. This seems to work as well for **_TZE608_xkr8gep3**. Might I propose adding
`("_TZE608_xkr8gep3", "TS0603")`
to MODELS_INFO

jarzebskim, 2025-12-31:
> [@G4LLY](https://github.com/G4LLY) Nicely done. This seems to work as well for **_TZE608_xkr8gep3**. Might I propose adding `("_TZE608_xkr8gep3", "TS0603")` to MODELS_INFO

maybe make own device request?