zigpy/zha-device-handlers #3266: [Device Support Request] Tuya Soil Sensor (_TZE284_sgabhwa6 TS0601)

treyd, 2024-07-20:
### Problem description

New Tuya Soil Sensor does appear to have any clusters with temp/humidity readings after adding

https://www.aliexpress.us/item/3256806923087260.html

### Solution description

Device supported, reporting soil temp, humidity and battery level

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_sgabhwa6",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2024.7.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "3.10.0-957.27.2.el7.x86_64",
    "run_as_root": true
  },
[... 435 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

treyd, 2024-07-24:
I did a scan using `zha_toolkit` of the device.  I'm not sure which attributes are the temperatures, since they're reporting "1"

<details><summary>ZHA Toolkit Scan</summary>

```json
{
  "ieee": "REDACTED",
  "nwk": "REDACTED",
  "model": "TS0601",
  "manufacturer": "_TZE284_sgabhwa6",
  "manufacturer_id": "0x4417",
  "endpoints": [
    {
      "id": 1,
      "device_type": "0x0051",
      "profile": "0x0104",
      "in_clusters": {
        "0x0000": {
          "cluster_id": "0x0000",
          "title": "Basic",
          "name": "basic",
          "attributes": {
            "0x0000": {
              "attribute_id": "0x0000",
              "attribute_name": "zcl_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|REPORT",
              "access_acl": 5,
              "attribute_value": 3
            },
            "0x0001": {
              "attribute_id": "0x0001",
              "attribute_name": "app_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|REPORT",
              "access_acl": 5,
              "attribute_value": 77
            },
            "0x0002": {
              "attribute_id": "0x0002",
              "attribute_name": "stack_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|REPORT",
              "access_acl": 5,
              "attribute_value": 0
            },
            "0x0003": {
              "attribute_id": "0x0003",
              "attribute_name": "hw_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|REPORT",
              "access_acl": 5,
              "attribute_value": 1
            },
[... 433 lines cut ...]
              "attribute_value": 0
            },
            "0xfffd": {
              "attribute_id": "0xfffd",
              "attribute_name": "cluster_revision",
              "value_type": [
                "0x21",
                "uint16_t",
                "Analog"
              ],
              "access": "READ",
              "access_acl": 1,
              "attribute_value": 3
            }
          },
          "commands_received": {},
          "commands_generated": {}
        }
      }
    }
  ]
}
```

</details>

GeoffWy, 2024-08-06:
I have exactly the same problem.

I tried adding the quirks file at [zhaquirks/tuya/ts0601_sensor.py](zhaquirks/tuya/ts0601_sensor.py), with ("_TZE284_sgabhwa6", "TS0601") added to the MODELS_INFO for class TuyaSoilSensor, but that didn't work. The diagnostics show "quirk_applied": false.

I see one has been produced for zigbee2mqtt, see [https://github.com/Koenkk/zigbee-herdsman-converters/pull/7741](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7741), in case that helps.

abertolin, 2024-08-12:
Is it possible to convert the zigbee2mqtt configuration to ZHA ?

skypyxis, 2024-08-24:
I tried with ChatGPT and this was the result:

[_TZE284_sgabhwa6 TS0601.py.txt](https://github.com/user-attachments/files/16738278/_TZE284_sgabhwa6.TS0601.py.txt)

However it doesn't work, of course, there are errors on the log

`ImportError: cannot import name 'TuyaTemperatureMeasurement' from 'zhaquirks.tuya' (/usr/local/lib/python3.12/site-packages/zhaquirks/tuya/__init__.py)`
  
 To try it I added this to configuration.yaml:
``` 
 zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks
```

and copied the ...TS0601.py file to that folder

I'm sure someone with experience creating these zha-device-handlers can do it now that the equivalent zigbee2mqtt converter exists. I hope someone has some time to do it.

PS: Is it possible to run both ZHA and Zigbee2MQTT simultaneously on the same SkyConnect Zigbee coordinator? I assume not, but it would be ideal, as it would allow us to use this soil sensor with Zigbee2MQTT while maintaining the other devices with ZHA as we do today.
Having these two 'flavors' comes with its downsides. Development efforts are divided between ZHA and Zigbee2MQTT, spreading focus thin and impacting device support.

closed 2024-08-27 by TheJulianJES