zigpy/zha-device-handlers #3268: [Device Support Request] New variant Tuya '_Tz3000_okaz9tjs' TS011F Smart Plug found

lktechnl, 2024-07-22:
### Problem description

Smartplug _TZ3000_okaz9tjs was not detected correctly and showed light bulb switch button instead of smart plug switch button. 
Current was x1000 as large.

### Solution description

I've added an additional class for proper detection

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

    "ieee": "**REDACTED**",
    "nwk": 37520,
    "manufacturer": "_TZ3000_okaz9tjs",
    "model": "TS011F",
    "name": "_TZ3000_okaz9tjs TS011F",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4660,
    "power_source": "Mains",
    "lqi": 88,
    "rssi": -78,
    "last_seen": "2024-07-23T00:21:52",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x000a",
            "0x0702",
            "0x0b04",
            "0x1000",
            "0xe000",
            "0xe001"
          ],
          "output_clusters": []
        }
      },
      "manufacturer": "_TZ3000_okaz9tjs",
      "model": "TS011F"
    },

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>new class for ts011f_plug.py</summary>

class Plug_v4(EnchantedDevice):
    """Tuya TS011F plug. One plug is _Tz3000_okaz9tjs + DeviceType.ON_OFF_LIGHT"""

    quirk_id = TUYA_PLUG_ONOFF

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
        # "profile_id": 260,
        # "device_type": "0x0100",
        # "input_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x000a","0x0702","0x0b04","0x1000","0xe000","0xe001"],
        # "output_clusters": []
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    LightLink.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Time.cluster_id,
                    TuyaZBMeteringClusterWithUnit,
                    TuyaZBElectricalMeasurement,
                    LightLink.cluster_id,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    
</details>

LeMyst, 2024-09-05:
Duplicate of https://github.com/zigpy/zha-device-handlers/issues/3001 ?

Badbond, 2025-01-19:
Can verify that the [existing V2 quirk](https://github.com/zigpy/zha-device-handlers/blob/b7f2bc7a3becaf5c1d97d7420ff6fccc1d1e2785/zhaquirks/tuya/ts011f_plug.py#L1050-L1051) mentioning this manufacturer does not work for this new variant. When tweaking the quirk for the `device_type` and `input_clusters` mentioned in OP's post's signature, it worked for me.