zigpy/zha-device-handlers #3278: [Device Support Request] _TZE204_x9usygq1 MOES pilot wire with power comsumption

CoBouLu, 2024-08-02:
### Problem description

Hi all,
Can someone create a quirk for _TZE204_x9usygq1, model TS0601?

https://www.domadoo.fr/fr/chauffage/7153-moes-thermostat-programmable-zigbee-tuya-fil-pilote-mesure-conso.html

### Solution description

It should be usefull for me to have a quirk for this device.
Thanks in avance😉

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.7.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "6.6.33-haos",
    "supervisor": "2024.06.2",
    "host_os": "Home Assistant OS 12.4",
    "docker_version": "26.1.4",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "eufy_security": {
      "documentation": "https://github.com/fuatakgun/eufy_security",
      "version": "8.0.2",
      "requirements": [
        "websocket-client==1.4.2",
        "aiortsp==1.3.7"
      ]
    },
    "climate_group": {
      "documentation": "https://github.com/daenny/climate_group",
      "version": "0.4.2",
      "requirements": []
    },
    "heatzy": {
      "documentation": "https://github.com/cyr-ius/hass-heatzy",
[... 517 lines cut ...]
      {
        "code": "upper_temp",
        "name": "设置温度上限",
        "type": "Integer",
        "values": "{\"unit\":\"\",\"min\":50,\"max\":3000,\"scale\":1,\"step\":10}"
      },
      {
        "code": "child_lock",
        "name": "童锁",
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "temp_unit_convert",
        "name": "温标切换设置",
        "type": "Enum",
        "values": "{\"range\":[\"c\",\"f\"]}"
      },
      {
        "code": "temp_set",
        "name": "温度设置",
        "type": "Integer",
        "values": "{\"unit\":\"\",\"min\":50,\"max\":3000,\"scale\":1,\"step\":5}"
      }
    ]

CoBouLu, 2024-09-27:
> Hello @CoBouLu, could you make it work? I have the same and discovered it is not supported... Though maybe this config should be compatible? I will test [Koenkk/zigbee2mqtt#19169](https://github.com/Koenkk/zigbee2mqtt/issues/19169)

Not Yet. I bought WIFI's one and managed working with Local Tuya only

ludoviclt, 2024-10-26:
Hello @CoBouLu @raphmur , have you managed to get it  work under zigbee2mqtt or still not ?
 is it better to take the wifi version?

Merci

raphmur, 2024-10-26:
No news from me
I might stick to the handmade tuya 2 ch Switch + diodes...

CoBouLu, 2024-11-13:
> Hello @CoBouLu @raphmur @TheJulianJES , have you managed to get it work under zigbee2mqtt or still not ? is it better to take the wifi version?
> 
> Merci

Hello @ludoviclt As i've mentioned, i already use WIFI version with Local TUYA (13 nodes, one by heater)

@barchasse38 Good Job, i'll follow it 😉

Empyrz901, 2025-02-26:
Based on the differents information here and on https://github.com/Koenkk/zigbee2mqtt/issues/24635, this is my code that works at least to enable child lock, setting heating mode and getting current temperature but I've been unable to do something to manage the night led.

```python
from zigpy.quirks.v2 import EntityType, EntityPlatform
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant import (
    UnitOfElectricCurrent,
    UnitOfElectricPotential,
    UnitOfEnergy,
    UnitOfPower,
    UnitOfTemperature,
)
from zhaquirks.tuya.builder import TuyaQuirkBuilder
import zigpy.types as t

class TempUnit(t.enum8):
    """Temperature Unit Enum."""
    Celsius = 0x00
    Fahrenheit = 0x01

class HeatMode(t.enum8):
    """Heater Mode Enum."""
    Standby = 0x00
    Comfort = 0x01
    Comfort_1 = 0x02
    Comfort_2 = 0x03
    Eco = 0x04
    Antifrost = 0x05
    Program = 0x06
    Thermostat = 0x07

(
    TuyaQuirkBuilder("_TZE204_x9usygq1", "TS0601")
    .tuya_temperature(dp_id=16, scale=10)
    .tuya_sensor(
        dp_id=107,
        attribute_name="average_power",
        type=t.int24s,
        divisor=10,
        translation_key="power_30_min",
        fallback_name="Power 30 min",
        unit=UnitOfEnergy.WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL,
    )
    .tuya_switch(
        dp_id=39,
        attribute_name="child_lock",
        translation_key="child_lock",
        fallback_name="Child lock",
    )
    .tuya_dp_attribute(
        dp_id=2,
        attribute_name="heat_mode",
        type=t.enum8,
    )
    .enum(
        attribute_name="heat_mode",
        cluster_id=0xEF00,
        enum_class=HeatMode,
        translation_key="heat_mode",
        fallback_name="Heat mode",
    )
    .tuya_dp_attribute(
        dp_id=46,
        attribute_name="temp_unit_convert",
        type=t.enum8,
    )
    .enum(
        attribute_name="temp_unit_convert",
        cluster_id=0xEF00,
        enum_class=TempUnit,
        translation_key="temp_unit",
        fallback_name="Temperature Unit",
    )
    .tuya_number(
        dp_id=50,
        attribute_name="temp_set",
        type=t.int16s,
        min_value=5,
        max_value=30,
        step=0.5,
        unit="°C",
        multiplier=0.1,
        translation_key="temp_set",
        fallback_name="Temp set",
    )
    .skip_configuration()
    .add_to_registry()
)
```

I'm interested in getting info to manage the Led. ChatGPT never gave me a working solution.
ATM I got a log : 
```
Decoded ZCL frame: TuyaReplacementCluster:set_data_response(data=TuyaCommand(status=0, tsn=228, datapoints=[TuyaDatapointData(dp=114, data=TuyaData(dp_type=<TuyaDPType.RAW: 0>, function=0, raw=b'\x04\xb0\x01h\x00', *payload=b'\x04\xb0\x01h\x00'))]))
2025-02-26 22:07:33.475 DEBUG (MainThread) [zigpy.zcl] [0xD273:1:0xef00] Received command 0x02 (TSN 62): set_data_response(data=TuyaCommand(status=0, tsn=228, datapoints=[TuyaDatapointData(dp=114, data=TuyaData(dp_type=<TuyaDPType.RAW: 0>, function=0, raw=b'\x04\xb0\x01h\x00', *payload=b'\x04\xb0\x01h\x00'))]))
2025-02-26 22:07:33.475 DEBUG (MainThread) [zigpy.zcl] [0xD273:1:0xef00] No datapoint handler for TuyaDatapointData(dp=114, data=TuyaData(dp_type=<TuyaDPType.RAW: 0>, function=0, raw=b'\x04\xb0\x01h\x00', *payload=b'\x04\xb0\x01h\x00'))	
```
From what I understood, the 2 first bytes are the start time, the 2 following are the stop timer and the last byte is the led state (enable/disable) but impossible to add a control in HA to manage it.
Did someone has anything about it ?

CoBouLu, 2025-02-27:
> Based on the differents information here and on [Koenkk/zigbee2mqtt#24635](https://github.com/Koenkk/zigbee2mqtt/issues/24635), this is my code that works at least to enable child lock, setting heating mode and getting current temperature but I've been unable to do something to manage the night led.
> 
> from zigpy.quirks.v2 import EntityType, EntityPlatform
> from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
> from zigpy.quirks.v2.homeassistant import (
>     UnitOfElectricCurrent,
>     UnitOfElectricPotential,
>     UnitOfEnergy,
>     UnitOfPower,
>     UnitOfTemperature,
> )
> from zhaquirks.tuya.builder import TuyaQuirkBuilder
> import zigpy.types as t
> 
> 
> class TempUnit(t.enum8):
>     """Temperature Unit Enum."""
>     Celsius = 0x00
>     Fahrenheit = 0x01
> 
> class HeatMode(t.enum8):
>     """Heater Mode Enum."""
>     Standby = 0x00
>     Comfort = 0x01
>     Comfort_1 = 0x02
>     Comfort_2 = 0x03
>     Eco = 0x04
>     Antifrost = 0x05
>     Program = 0x06
>     Thermostat = 0x07
> 
> (
>     TuyaQuirkBuilder("_TZE204_x9usygq1", "TS0601")
>     .tuya_temperature(dp_id=16, scale=10)
>     .tuya_sensor(
>         dp_id=107,
>         attribute_name="average_power",
>         type=t.int24s,
>         divisor=10,
>         translation_key="power_30_min",
>         fallback_name="Power 30 min",
>         unit=UnitOfEnergy.WATT_HOUR,
>         device_class=SensorDeviceClass.ENERGY,
>         state_class=SensorStateClass.TOTAL,
>     )
>     .tuya_switch(
>         dp_id=39,
>         attribute_name="child_lock",
>         translation_key="child_lock",
>         fallback_name="Child lock",
>     )
>     .tuya_dp_attribute(
>         dp_id=2,
>         attribute_name="heat_mode",
>         type=t.enum8,
>     )
>     .enum(
>         attribute_name="heat_mode",
>         cluster_id=0xEF00,
>         enum_class=HeatMode,
>         translation_key="heat_mode",
>         fallback_name="Heat mode",
>     )
>     .tuya_dp_attribute(
>         dp_id=46,
>         attribute_name="temp_unit_convert",
>         type=t.enum8,
>     )
>     .enum(
>         attribute_name="temp_unit_convert",
>         cluster_id=0xEF00,
>         enum_class=TempUnit,
>         translation_key="temp_unit",
>         fallback_name="Temperature Unit",
>     )
>     .tuya_number(
>         dp_id=50,
>         attribute_name="temp_set",
>         type=t.int16s,
>         min_value=5,
>         max_value=30,
>         step=0.5,
>         unit="°C",
>         multiplier=0.1,
>         translation_key="temp_set",
>         fallback_name="Temp set",
>     )
>     .skip_configuration()
>     .add_to_registry()
> )
> I'm interested in getting info to manage the Led. ChatGPT never gave me a working solution. ATM I got a log :
> 
> ```
> Decoded ZCL frame: TuyaReplacementCluster:set_data_response(data=TuyaCommand(status=0, tsn=228, datapoints=[TuyaDatapointData(dp=114, data=TuyaData(dp_type=<TuyaDPType.RAW: 0>, function=0, raw=b'\x04\xb0\x01h\x00', *payload=b'\x04\xb0\x01h\x00'))]))
> 2025-02-26 22:07:33.475 DEBUG (MainThread) [zigpy.zcl] [0xD273:1:0xef00] Received command 0x02 (TSN 62): set_data_response(data=TuyaCommand(status=0, tsn=228, datapoints=[TuyaDatapointData(dp=114, data=TuyaData(dp_type=<TuyaDPType.RAW: 0>, function=0, raw=b'\x04\xb0\x01h\x00', *payload=b'\x04\xb0\x01h\x00'))]))
> 2025-02-26 22:07:33.475 DEBUG (MainThread) [zigpy.zcl] [0xD273:1:0xef00] No datapoint handler for TuyaDatapointData(dp=114, data=TuyaData(dp_type=<TuyaDPType.RAW: 0>, function=0, raw=b'\x04\xb0\x01h\x00', *payload=b'\x04\xb0\x01h\x00'))	
> ```
> 
> From what I understood, the 2 first bytes are the start time, the 2 following are the stop timer and the last byte is the led state (enable/disable) but impossible to add a control in HA to manage it. Did someone has anything about it ?

Seem to work well!!! Really thanks 😉

ludoviclt, 2025-02-27:
bonjour, comment on utilise le code pour que cela fonctionne sur jeedom

merci

CoBouLu, 2025-04-01:
@Empyrz901 Can you add the availability to add device "by" this device.
I have only this:

[image]
 And want this:

[image]

thanks in advance

Empyrz901, 2025-04-02:
> bonjour, comment on utilise le code pour que cela fonctionne sur jeedom
> 
> merci

Hello, aucune idée, tu es ici dans un repo dédié à Home Assistant 

> [@Empyrz901](https://github.com/Empyrz901) Can you add the availability to add device "by" this device. I have only this:
> 
> [image] And want this:
> 
> [image]
> 
> thanks in advance

Hi, I checked out and the TuyaQuirkBuilderV2 doesn't seems to allow this at the moment.
Anyway, I plan to rewrite all the code in QuirkV1 as to control the night led, it has to send a raw value that TuyaQuirkBuilderV2 doesn't allow. 
But it is harder to write and read, cf https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py

CoBouLu, 2025-04-02:
> Hi, I checked out and the TuyaQuirkBuilderV2 doesn't seems to allow this at the moment.
> Anyway, I plan to rewrite all the code in QuirkV1 as to control the night led, it has to send a raw value that [...]
Ok, let me know if u succeed 😉
Merci

closed 2025-10-08 by github-actions[bot]