zigpy/zha-device-handlers #3279: NAS-WR01B (or _TZ3000_w0qqde0g) power plug as a repeater

koverg70, 2024-08-02:
Hi everyone!

Thanks for this great integration of Zigbee under Home Assistant!

I am using this device under Home Assistant without an issue: https://forum.phoscon.de/t/new-tuya-smart-plug-nas-wr01b-or-tz3000-w0qqde0g/819

But I would like to use it as a repeater too. 

Can I set it up within HA or using any other tool?
There is an "Add devices via this device" menu in HA inside the device info card, but it doesn't work for me.

Hope someone can help...

Best regards,
Gábor

Frsco9, 2024-11-01:
Hi Gabor,

i have the same issue withe the smart plag. Works fine but do not behave as a repeter. Did you solve this already?

Best,
Marco

koverg70, 2024-11-01:
> i have the same issue withe the smart plag. Works fine but do not behave as a repeter. Did you solve this already?
I'm using zigbee2mqtt now, and it works as a repeater without any configuration on the device.

[image]