zigpy/zha-device-handlers #3282: [Device Support Request] TS0601 by _TZE204_wbhaespm 3 phase power meter din

l4red0, 2024-08-03:
### Problem description

https://allegro.pl/oferta/bezpiecznik-3-fazy-regulowany-1-do-63a-zigbee-tuya-licznik-miernik-energii-16097353966

The device is a 3-phase circuit breaker (63A) with power metering, mounted on a DIN rail. It is presented as a Tuya device under the RTX brand, model C63. The device is recognized by ZHA, but no entities are registered. I was able to bind the power summation delivery entity with a custom quirk and identify attributes for each of the three phases (including amps, volts, and watts). However, I am currently stuck with correctly binding that data to ZHA.

### Solution description

Bind data for all of three phases to ZHA.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0702",
            "0x0b04",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [
            "0x0006"
          ],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "_TZE204_wbhaespm",
      "model": "TS0601"
    }
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.7.4",
[... 792 lines cut ...]
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```

</details>

l4red0, 2024-08-04:
I've made some progress. I tinkered a bit and checked how the implementations of other three-phase meters look. I think I'm now mapping the entire data from the three phases + summation delivered. Unfortunately, the sensors in the ZHA device panel only show data from the first phase (even though the data is transmitted and read by ZHA for all three phases; rms_voltage - as a sensor, only shows phase 1, rms_voltage_ph_b - hidden but reads data, similarly with current and power for the other phases). Additionally, the sum of W for the respective phases goes to total_active_power (it's calculated correctly but doesn't display as a sensor).

There are still some other mysteries to solve like switch control or these attributes:
```python
[zhaquirks.tuya] [0xf438:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x020d (command 0x0002)
[zhaquirks.tuya] [0xf438:1:0xef00] Received value [0] for attribute 0x0509 (command 0x0002)
[zhaquirks.tuya] [0xf438:1:0xef00] Received value [0, 0, 2, 126] for attribute 0x0201 (command 0x0002) #summation deliverd
[zhaquirks.tuya] [0xf438:1:0xef00] Received value [48, 48, 48, 48, 48, 48, 48, 48, 48, 48] for attribute 0x0313 (command 0x0002)
```
This is far from an ideal solution but it allows for further development. I would appreciate any guidance.

[image]
<details>
  <summary>Additional screenshots</summary>

[image]

[image]

[image]

</details>
<details>
  <summary>Custom quirk (very messy!)</summary>
  
  ```python
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
import logging
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff
from zigpy.quirks import CustomDevice

_LOGGER = logging.getLogger(__name__)

# Tuya attribute IDs
TUYA_TOTAL_ENERGY_ATTR = 0x0201
#TUYA_CURRENT_ATTR = 0x0212
#TUYA_POWER_ATTR = 0x0267
#TUYA_VOLTAGE_ATTR = 0x0214

RTX_TOTAL_ACTIVE_POWER_ATTR = 0x0304
RTX_TOTAL_CURRENT = 0x0212
RTX_AVG_VOLTAGE = 0x0214
RTX_VCP_P1 = 0x0006
RTX_VCP_P2 = 0x0007
RTX_VCP_P3 = 0x0008

#SWITCH_EVENT = "switch_event"

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),
        RTX_TOTAL_CURRENT: ("current", t.uint32_t, True),
        RTX_AVG_VOLTAGE: ("voltage", t.uint32_t, True),
[... 151 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

  ```
</details>

l4red0, 2024-08-15:
I was able to build the basic functionality I was aiming for, but I am pausing work on this quirk. However, there are still many features and data from the device that remain unhandled, and the ones I managed to implement could be done better.

In this quirk, I've combined power, current, and voltage readings across all three phases to provide a single entity for each metric. The power values from all phases are summed and reflected in the main `active_power` attribute, while the current values are summed and stored in `rms_current`. For voltage, there is calculated the average across all three phases and assigned it to `rms_voltage`.

The original data for phase A **has been overwritten** with these combined values, so entities now represent the overall 3-phase performance rather than just phase A. This gives more accurate and holistic readings for power, current, and voltage while keeping the energy summation unchanged. Worth to mention separate VCP data for all three phases is accessible by 'manage device' option but not as entities.

Quirk provides total power (sum), total current (sum), and average voltage, all while preserving the original energy summation behavior. This implementation improves monitoring of the 3-phase system by giving a complete view rather than per-phase / phase A data.

Any further development (like on/off switch or mystery attributes mentioned in previous comment) will be very welcome.

[image]

<details>
  <summary>Custom quirk (read comment above!)</summary>
  
```python
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering
import logging

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes

_LOGGER = logging.getLogger(__name__)

TUYA_TOTAL_ENERGY_ATTR = 0x0201
RTX_VCP_P1 = 0x0006
RTX_VCP_P2 = 0x0007
RTX_VCP_P3 = 0x0008

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),
        RTX_VCP_P1: ("vcp_ph_1", t.data64, True),
        RTX_VCP_P2: ("vcp_ph_2", t.data64, True),
        RTX_VCP_P3: ("vcp_ph_3", t.data64, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_reported(value / 100)
        elif attrid in [RTX_VCP_P1, RTX_VCP_P2, RTX_VCP_P3]:
            _LOGGER.debug(f"Received phase data: attrid={attrid}, value={value}")
            power = int.from_bytes(value[0:3], byteorder="little")
            current = int.from_bytes(value[3:6], byteorder="little")
            voltage = int.from_bytes(value[6:8], byteorder="little")
            phase_index = attrid - RTX_VCP_P1  
            self.endpoint.electrical_measurement.vcp_reported(voltage, current, power, phase_index)

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):

    cluster_id = ElectricalMeasurement.cluster_id

    phase_attributes = [
        {  # Phase 1
            "voltage": ElectricalMeasurement.AttributeDefs.rms_voltage.id,
[... 116 lines cut ...]
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```
</details>

Noamibr, 2024-09-05:
Thank you for this, highly appreciated.
I have the exact model and have the same issue.
When applying the quirk, it now shows the sensor's information but combined (for example - instead of 240V it shows 720V).
I still don't see any controls though...