zigpy/zha-device-handlers #3287: [Device Support Request] SASWELL SAS980SWT-7-Z01 Smart irrigation, water timer

emmanuelg34, 2024-08-06:
### Problem description

https://www.domadoo.fr/fr/jardin-connecte/5886-saswell-electrovanne-et-programmateur-d-arrosage-connecte-zigbee-mesure-de-consommation.html 
https://www.saswell.com/smart-irrigation-wifi-water-timer-sas980swt-7-z01_p147.html

Wifi smart sprinkler controller
Smart irrigation,Wifi water timer SAS980SWT-7-Z01

same as S0601 - TZE200_81isopgh

"TS0601
by _TZE204_qtnjuoae
Zigbee info
IEEE: a4:c1:38:23:6c:b8:8c:ca
NWK: 0xc713
Device Type: EndDevice
LQI: 30
RSSI: Unknown
Last seen: 2024-08-02T14:26:52
Power source: Battery or Unknown"

### Solution description

Could you please add the device to ZHA ?
TS0601 
by _TZE204_qtnjuoae 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_qtnjuoae",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.7.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
[... 372 lines cut ...]
        }
      }
    }
  }
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

nicktrigger, 2024-10-14:
Seconded.  I guess this is basically the same as the similar valves with slightly different ids.

So hopefully just a case of adding to ZHA with the product name. 

I'm not technically minded enough to get my head around these custom quirks.

prairiesnpr, 2024-11-08:
Same on this one, see if this gives control, then we can add in the other DPs.

```python
from zhaquirks.tuya import TuyaPowerConfigurationCluster4AA
from zhaquirks.tuya.builder import TuyaQuirkBuilder

TuyaQuirkBuilder("_TZE204_qtnjuoae", "TS0601") 
.tuya_onoff(dp_id=1)
.tuya_battery(dp_id=7, power_cfg=TuyaPowerConfigurationCluster4AA)
.skip_configuration()
.add_to_registry()
```

nicktrigger, 2024-11-08:
Awesome this also works! I really appreciate the help with this.

The other dpids that would be useful would be:

"1 ( code: switch , value: False, cloud pull )",
"4 ( code: fault , value: 0, cloud pull )",
"5 ( code: water_once , value: 831, cloud pull )",
"6 ( code: water_total , value: 83, cloud pull )",
"8 ( code: battery_state , value: low, cloud pull )",
"9 ( code: time_use , value: 0, cloud pull )",
"11 ( code: countdown , value: 60, cloud pull )",
"12 ( code: work_state , value: idle, cloud pull )",
"15 ( code: use_time_one , value: 540, cloud pull )",
"17 ( code: timer , value: AA==, cloud pull )"

nicktrigger, 2024-11-09:
I used the same code as your helpful response in the other valve thread and its working great with the 4aa power setting, but not 2aa.

This one reports summation delivered but its always 86L, which is definitely not correct. And irrigation time also seems to have no affect on the watering time.

[image]

closed 2025-12-09 by github-actions[bot]