zigpy/zha-device-handlers #3288: Aquara lumi.motion.ac01 (Presence sensor FP1) missing sensor.name_presence_event in ZHA

akaszubski, 2024-08-07:
### Problem description

The aquara FP1 presence sensor has a sensor to set  occupancy on/off , this takes a long time to activate. In zigbee2mqtt, there is also a sensor for sensor.name_presence_event which triggers immediately enter/leave/away/approach. This can be set to trigger quicker before occupancy activates.

### Solution description

in ZHA for the FP1 presence sensor to include sensor.name_presence_event for the device as per zigbee2mqtt support.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.MainsPowered|RxOnWhenIdle|AllocateAddress: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=127, server_mask=11264, maximum_outgoing_transfer_size=127, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0xfff0",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0406",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "aqara",
  "model": "lumi.motion.ac01",
  "class": "zhaquirks.xiaomi.aqara.motion_ac01.AqaraLumiMotionAc01"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.7.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.4",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "6.6.31-haos-raspi",
    "supervisor": "2024.08.0",
    "host_os": "Home Assistant OS 12.4",
    "docker_version": "26.1.4",
[... 434 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

dmulcahey, 2024-08-07:
Check the logbook there are events for these IIRC

akaszubski, 2024-08-07:
Thank you, you are right its possible to use zha_event to trigger the automation rather than using a sensor state change.

closed 2024-08-07 by akaszubski

dmulcahey, 2024-08-07:
Just as an FYI we are looking at event entities too. So there may be entities for this in the future as well but for today the events are the way to go.