zigpy/zha-device-handlers #3294: [Device Support Request] Aqara FP1e (lumi.sensor_occupy.agl1)

dmulcahey, 2024-08-09:
### Problem description

Everything on this device is implemented on the apple cluster 0xfcc0

### Solution description

Implement the proper quirk providing support for the device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 140,
    "manufacturer_code": 4660,
    "maximum_buffer_size": 108,
    "maximum_incoming_transfer_size": 127,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 127,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0xfff0",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "aqara",
  "model": "lumi.sensor_occupy.agl1",
  "class": "zigpy.device.Device"
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

martinw72, 2024-08-10:
Would love to see this device supported in ZHA! Is there anything I can do? Would like to help, but I’m not a dev!

sandokanalles, 2024-08-10:
I copied the py file from the Aqara FP1, changed a few lines to lumi.sensor_occupy.agl1.

Then created the directory config/zha_quirks under the homeassistant folder
Copied the file motion_agl1.py to that directory

Added the following lines in the configuration.yaml
zha:
  enable_quirks: true
  custom_quirks_path: config/zha_quirks

And after a restart and some time (do not know why) motion was detected in HA.

I guess a lot more needs to be changed in the file but for me it is working and can use it like this until somebody with some actual experience can look at this :) 

-: Begin motion_agl1.py --------------------
"""Quirk for aqara lumi.sensor_occupy.agl1."""
from __future__ import annotations

import logging
from typing import Any

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as types
from zigpy.zcl.clusters.general import Basic, DeviceTemperature, Identify, Ota
from zigpy.zcl.clusters.measurement import OccupancySensing

from zhaquirks.const import (
    COMMAND,
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZHA_SEND_EVENT,
)
from zhaquirks.xiaomi import XiaomiAqaraE1Cluster

OCCUPANCY = 0x0000
PRESENCE = 0x0142
PRESENCE2 = 101
PRESENCE_EVENT = 0x0143
PRESENCE_EVENT2 = 102
MONITORING_MODE = 0x0144
MOTION_SENSITIVITY = 0x010C
APPROACH_DISTANCE = 0x0146
RESET_NO_PRESENCE_STATUS = 0x0157
SENSOR = "sensor"

_LOGGER = logging.getLogger(__name__)

class AqaraPresenceEvents(types.enum8):
    """Aqara presence events."""

    Enter = 0x00
    Leave = 0x01
    Enter_Left = 0x02
    Leave_Right = 0x03
    Enter_Right = 0x04
    Leave_Left = 0x05
    Approach = 0x06
    Away = 0x07
    Unknown = 0xFF

class OppleCluster(XiaomiAqaraE1Cluster):
    """Opple cluster."""

    ep_attribute = "opple_cluster"
    attributes = {
        PRESENCE: ("presence", types.uint8_t, True),
        MONITORING_MODE: ("monitoring_mode", types.uint8_t, True),
        MOTION_SENSITIVITY: ("motion_sensitivity", types.uint8_t, True),
        APPROACH_DISTANCE: ("approach_distance", types.uint8_t, True),
        RESET_NO_PRESENCE_STATUS: ("reset_no_presence_status", types.uint8_t, True),
    }

    def _update_attribute(self, attrid: int, value: Any) -> None:
        super()._update_attribute(attrid, value)
        if attrid == PRESENCE or attrid == PRESENCE2:
            if value != 0xFF:
                self.endpoint.occupancy.update_attribute(OCCUPANCY, value)
        elif attrid == PRESENCE_EVENT or attrid == PRESENCE_EVENT2:
            self.listener_event(ZHA_SEND_EVENT, AqaraPresenceEvents(value).name, {})

#AqaralumiSensorOccupyAgl1
#class AqaraLumiMotionAc01(CustomDevice):
class AqaralumiSensorOccupyAgl1(CustomDevice):
    """lumi.sensor_occupy.agl1 custom device implementation."""

    signature = {
        MODELS_INFO: [("aqara", "lumi.sensor_occupy.agl1")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: 0xFFF0,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OppleCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: 0xFFF0,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    DeviceTemperature.cluster_id,
                    OccupancySensing.cluster_id,
                    OppleCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }

    device_automation_triggers = {
        (AqaraPresenceEvents.Enter.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Enter.name
        },
        (AqaraPresenceEvents.Leave.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Leave.name
        },
        (AqaraPresenceEvents.Enter_Left.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Enter_Left.name
        },
        (AqaraPresenceEvents.Leave_Right.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Leave_Right.name
        },
        (AqaraPresenceEvents.Enter_Right.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Enter_Right.name
        },
        (AqaraPresenceEvents.Leave_Left.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Leave_Left.name
        },
        (AqaraPresenceEvents.Approach.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Approach.name
        },
        (AqaraPresenceEvents.Away.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Away.name
        },
        (AqaraPresenceEvents.Unknown.name, SENSOR): {
            COMMAND: AqaraPresenceEvents.Unknown.name
        },
    }

dmulcahey, 2024-08-10:
it’s similar to the existing FP1 but the event data appears to be sent on different attributes. I need to find a block of time to sit with the device and decode it.

elmocito, 2024-08-16:
Make it a +1 that a support would be awesome 🤩

hendrikboldt, 2024-08-20:
+1,

 I am currently trying to use the Senro, but I figured it still needs to be supported.

merdok, 2024-08-26:
Also waiting for support for this device...

gruberdieter, 2024-08-26:
+1 That would be really great. Thanks

dontmint, 2024-09-12:
+1 - just got this from really cheap TaoBao store

mattpugh, 2024-09-19:
Confirm https://github.com/klolik/zha-device-handlers/commit/49d66b678ff07f58c0a14e065241b29efea4ec82 is working with my FP1E Aqara Presence Sensor.

dontmint, 2024-09-24:
> Confirm [klolik@49d66b6](https://github.com/klolik/zha-device-handlers/commit/49d66b678ff07f58c0a14e065241b29efea4ec82) is working with my FP1E Aqara Presence Sensor.

It works for me as well

sandokanalles, 2024-09-24:
For me presence is detected, thx!

dontmint, 2024-09-27:
Somehow my sensor only detects motions, not presence. If I sit for a while, the sensor change to `clear`

dontmint, 2024-09-27:
Somehow my sensor only detect motions, not presence. If things not moving, after few seconds it's status is back to `away`
[image]

jeverley, 2024-09-29:
I've made some tweaks to the code in @sandokanalles quirk to expose additional parameter exposed by the device:

- DETECTION_RANGE: allows you to specify a maximum detection range between 0 and 600mm (6 metres)
- MOTION: values are Idle, Large, Small
- MOTION_DISTANCE: distance to the detected motion in mm

I've also removed functions that do not appear to be supported in the FP1**E** variant (it seems somewhat simplified).

It feels like this might be a good candidate for the quirks v2 format to expose the non-standard configuration and reporting values in the UI, in a similar way that z2m can achieve.

I might make an attempt at this.

```
"""Quirk for aqara lumi.sensor_occupy.agl1."""
from __future__ import annotations

import logging
from typing import Any

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as types
from zigpy.zcl.clusters.general import Basic, DeviceTemperature, Identify, Ota
from zigpy.zcl.clusters.measurement import OccupancySensing

from zhaquirks.const import (
    COMMAND,
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZHA_SEND_EVENT,
)
from zhaquirks.xiaomi import XiaomiAqaraE1Cluster

DETECTION_RANGE = 0x015B # UINT32 The configurable maximum detection distance in millimeters (default 600 = 6 meters).
MOTION = 0x0160 # UINT8 Detected motion magnitude 0x02 = no movement, 0x03 = large movement, 0x04 = small movement
MOTION_DISTANCE = 0x015f # UINT32 Distance to the detected motion (mm)
MOTION_SENSITIVITY = 0x010C # UINT8 The configurable sensitivity 0x01 = low, 0x02 = medium, 0x03 = high (default 0x03 = High)
PRESENCE = 0x0142 #UINT8 Presence detected 0x0 = none, 0x01 = detected
RESET_NO_PRESENCE_STATUS = 0x0157 # Trigger spatial learning

_LOGGER = logging.getLogger(__name__)

class AqaraMotionSensitivity(types.enum8):
    """Aqara motion sensitivity."""

    Low = 0x01
    Medium = 0x02
    High = 0x03

class AqaraMotion(types.enum8):
    """Aqara motion."""

    Idle = 0x02
    Large = 0x03
    Small = 0x04

class OppleCluster(XiaomiAqaraE1Cluster):
    """Aqara manufacturer specific settings."""

    attributes = {
        DETECTION_RANGE: ("detection_range", types.uint32_t, True),
        MOTION:  ("motion", AqaraMotion, True),
        MOTION_DISTANCE: ("motion_distance", types.uint32_t, True),
        MOTION_SENSITIVITY: ("motion_sensitivity", types.uint8_t, True),
        PRESENCE: ("presence", types.uint8_t, True),
        RESET_NO_PRESENCE_STATUS: ("reset_no_presence_status", types.uint8_t, True),
    }

    def _update_attribute(self, attrid: int, value: Any) -> None:
        super()._update_attribute(attrid, value)
        if attrid == PRESENCE:
            if value != 0xFF:
                self.endpoint.occupancy.update_attribute(
                    OccupancySensing.AttributeDefs.occupancy.id,
                    OccupancySensing.Occupancy(value)
                )

#AqaralumiSensorOccupyAgl1
#class AqaraLumiMotionAc01(CustomDevice):
class AqaralumiSensorOccupyAgl1(CustomDevice):
    """lumi.sensor_occupy.agl1 custom device implementation."""

    signature = {
        MODELS_INFO: [("aqara", "lumi.sensor_occupy.agl1")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: 0xFFF0,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OppleCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: 0xFFF0,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    DeviceTemperature.cluster_id,
                    OccupancySensing.cluster_id,
                    OppleCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

jeverley, 2024-09-29:
[image]

I've been able to expose the additional attributes using the quirks v2 API - unfortunately not all of the entities have a meaningful label yet as we'll need strings added to HA/ZHA 

```
"""Quirk for aqara lumi.sensor_occupy.agl1."""
from __future__ import annotations

import logging
from typing import Any

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.quirks.v2 import QuirkBuilder
from zigpy.quirks.v2.homeassistant import EntityPlatform, EntityType
import zigpy.types as types
from zigpy.zcl.clusters.general import Basic, DeviceTemperature, Identify, Ota
from zigpy.zcl.clusters.measurement import OccupancySensing
from zigpy.zcl.foundation import ZCLAttributeDef

from zhaquirks.xiaomi import XiaomiAqaraE1Cluster

DETECTION_DISTANCE_ATTR_ID = 0x015B # UINT32 The configurable maximum detection distance in millimeters (default 600 = 6 meters).
DETECTION_SENSITIVITY_ATTR_ID = 0x010C # UINT8 The configurable detection sensitivity (0x01 = low, 0x02 = medium, 0x03 = high, default 0x03 = High)
MOTION_ATTR_ID = 0x0160 # UINT8 Detected motion (0x02 = no movement, 0x03 = large movement, 0x04 = small movement)
MOTION_DISTANCE_ATTR_ID = 0x015f # UINT32 Distance to the detected motion (mm)
OCCUPANCY_ATTR_ID = 0x0142 #UINT8 Occupancy detected 0x0 = absence, 0x01 = presence
RESET_NO_PRESENCE_STATUS_ATTR_ID = 0x0157 # Trigger AI spatial learning

_LOGGER = logging.getLogger(__name__)

class AqaraMotionSensitivity(types.enum8):
    """Aqara motion sensitivity."""

    Low = 0x01
    Medium = 0x02
    High = 0x03

class AqaraMotion(types.enum8):
    """Aqara motion."""

    Unknown = 0x00
    Unknown_1 = 0x01
    Idle = 0x02
    Moving = 0x03
    Still = 0x04

class OppleCluster(XiaomiAqaraE1Cluster):
    """Aqara manufacturer cluster."""

    class AttributeDefs(XiaomiAqaraE1Cluster.AttributeDefs):
        """Aqara occupancy sensor manufacturer specific attributes."""

        detection_distance = ZCLAttributeDef(
            id=DETECTION_DISTANCE_ATTR_ID,
            type=types.uint32_t,
            is_manufacturer_specific=True,
        )

        detection_sensitivity = ZCLAttributeDef(
            id=DETECTION_SENSITIVITY_ATTR_ID,
            type=types.uint8_t,
            is_manufacturer_specific=True,
        )

        motion = ZCLAttributeDef(
            id=MOTION_ATTR_ID,
            type=types.uint8_t,
            is_manufacturer_specific=True,
        )

        motion_distance = ZCLAttributeDef(
            id=MOTION_DISTANCE_ATTR_ID,
            type=types.uint32_t,
            is_manufacturer_specific=True,
        )

        occupancy = ZCLAttributeDef(
            id=OCCUPANCY_ATTR_ID,
            type=types.uint8_t,
            is_manufacturer_specific=True,
        )

        reset_no_presence_status = ZCLAttributeDef(
            id=RESET_NO_PRESENCE_STATUS_ATTR_ID,
            type=types.uint8_t,
            is_manufacturer_specific=True,
        )

    def _update_attribute(self, attrid: int, value: Any) -> None:
        """Update the OccupancySensing cluster."""
        super()._update_attribute(attrid, value)
        if attrid == OCCUPANCY_ATTR_ID:
            if value != 0xFF:
                self.endpoint.occupancy.update_attribute(
                    OccupancySensing.AttributeDefs.occupancy.id,
                    OccupancySensing.Occupancy(value)
                )

(
    QuirkBuilder("aqara", "lumi.sensor_occupy.agl1")
    .adds(DeviceTemperature)
    .adds(OccupancySensing)
    .replaces(OppleCluster)
    # Detection distance
    .number(
        OppleCluster.AttributeDefs.detection_distance.name,
        OppleCluster.cluster_id,
        min_value=0,
        max_value=600,
        multiplier=0.01,
        step=0.01,
    )
    # Detection sensitivity
    .enum(
        OppleCluster.AttributeDefs.detection_sensitivity.name,
        AqaraMotionSensitivity,
        OppleCluster.cluster_id,
    )
    # Motion
    .enum(
        OppleCluster.AttributeDefs.motion.name,
        AqaraMotion,
        OppleCluster.cluster_id,
        entity_platform=EntityPlatform.SENSOR,
        entity_type=EntityType.STANDARD,
    )
    # Motion distance
    .sensor(
        OppleCluster.AttributeDefs.motion_distance.name,
        OppleCluster.cluster_id,
        multiplier=0.01,
    )
    # Trigger AI spatial learning 
    .write_attr_button(
        OppleCluster.AttributeDefs.reset_no_presence_status.name,
        1,
        OppleCluster.cluster_id,
    )
    .add_to_registry()
)
```

jeverley, 2024-09-29:
> Somehow my sensor only detect motions, not presence. If things not moving, after few seconds it's status is back to `away` <img alt="Screenshot 2024-09-27 at 21 36 20" width="303" src="https://private-user-images.githubusercontent.com/34891994/371575122-67983efb-ec29-45a2-bd66-948df5132468.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc2MzkxMDEsIm5iZiI6MTcyNzYzODgwMSwicGF0aCI6Ii8zNDg5MTk5NC8zNzE1NzUxMjItNjc5ODNlZmItZWMyOS00NWEyLWJkNjYtOTQ4ZGY1MTMyNDY4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI5VDE5NDAwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM3OGMyZGE0YWJhYzY0ZjRlMzUxOWE2ZTc1N2U0YjZkZWI0ODJiMTE2NmRmYjZmYTBiNmRkYmVlMzJlMjNmMDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.WcvJp3TjxnnDpkj68fQWNm7JC_ISNns8t5ybaO0n_4w">

The 'Presence status reset' button a exposed by the quirk code I've attached in https://github.com/zigpy/zha-device-handlers/issues/3294#issuecomment-2381449768 may help you (it did for me), this triggers the device to use its 'AI Spatial Learning' function.
You should do this when the area is unoccupied and ensure nothing moves in the detection zone for at least 30 seconds.

jeverley, 2024-10-04:
A recent change has been introduced that allows v2 quirks to specify a fallback name for entities that don't yet have a common translation.

[image]

@dmulcahey, @puddly There still appear to be some bugs around naming distance device class entities, in addition to their unit not being applied correctly. I'll take a closer look at the logs around these and raise an issue documenting my findings.

```
"""Quirk for aqara lumi.sensor_occupy.agl1."""

from __future__ import annotations

import logging
from typing import Any

from zigpy.quirks.v2 import (
    QuirkBuilder,
    NumberDeviceClass,
    SensorDeviceClass,
    SensorStateClass,
)
from zigpy.quirks.v2.homeassistant import EntityPlatform, EntityType, UnitOfLength
import zigpy.types as types
from zigpy.zcl.clusters.general import DeviceTemperature
from zigpy.zcl.clusters.measurement import OccupancySensing
from zigpy.zcl.foundation import ZCLAttributeDef

from zhaquirks.xiaomi import XiaomiAqaraE1Cluster

APPROACH_DISTANCE_ATTR_ID = 0x015B  # UINT32 The configurable maximum detection distance in millimeters (default 600 = 6 meters).
MOTION_ATTR_ID = 0x0160  # UINT8 Detected motion (0x02 = no movement, 0x03 = large movement, 0x04 = small movement)
MOTION_DISTANCE_ATTR_ID = 0x015F  # UINT32 Distance to the detected motion (mm)
MOTION_SENSITIVITY_ATTR_ID = 0x010C  # UINT8 The configurable detection sensitivity (0x01 = low, 0x02 = medium, 0x03 = high, default 0x03 = High)
OCCUPANCY_ATTR_ID = 0x0142  # UINT8 Occupancy detected 0x0 = absence, 0x01 = presence
RESET_NO_PRESENCE_STATUS_ATTR_ID = 0x0157  # UINT8 Trigger AI spatial learning
RESTART_DEVICE_ATTR_ID = 0x00E8  # BOOL Trigger device restart

_LOGGER = logging.getLogger(__name__)

class AqaraMotionSensitivity(types.enum8):
    """Aqara motion sensitivity."""

    Low = 0x01
    Medium = 0x02
    High = 0x03

class AqaraMotion(types.enum8):
    """Aqara motion."""

    Unknown_0 = 0x00
    Unknown_1 = 0x01
    Idle = 0x02
    Moving = 0x03
    Still = 0x04

class OppleCluster(XiaomiAqaraE1Cluster):
    """Aqara manufacturer cluster for the FP1E presence sensor."""

    class AttributeDefs(XiaomiAqaraE1Cluster.AttributeDefs):
        """Aqara occupancy sensor manufacturer specific attributes."""

        approach_distance = ZCLAttributeDef(
            id=APPROACH_DISTANCE_ATTR_ID,
            type=types.uint32_t,
            access="rw",
            is_manufacturer_specific=True,
        )

        motion = ZCLAttributeDef(
            id=MOTION_ATTR_ID,
            type=types.uint8_t,
            access="rp",
            is_manufacturer_specific=True,
        )

        motion_distance = ZCLAttributeDef(
            id=MOTION_DISTANCE_ATTR_ID,
            type=types.uint32_t,
            access="rp",
            is_manufacturer_specific=True,
        )

        motion_sensitivity = ZCLAttributeDef(
            id=MOTION_SENSITIVITY_ATTR_ID,
            type=types.uint8_t,
            access="rw",
            is_manufacturer_specific=True,
        )

        occupancy = ZCLAttributeDef(
            id=OCCUPANCY_ATTR_ID,
            type=types.uint8_t,
            access="rp",
            is_manufacturer_specific=True,
        )

        reset_no_presence_status = ZCLAttributeDef(
            id=RESET_NO_PRESENCE_STATUS_ATTR_ID,
            type=types.uint8_t,
            access="w",
            is_manufacturer_specific=True,
        )

        restart_device = ZCLAttributeDef(
            id=RESTART_DEVICE_ATTR_ID,
            type=types.Bool,
            access="w",
            is_manufacturer_specific=True,
        )

    def _update_attribute(self, attrid: int, value: Any) -> None:
        """Update the OccupancySensing cluster."""
        super()._update_attribute(attrid, value)
        if attrid == OCCUPANCY_ATTR_ID:
            self.endpoint.occupancy.update_attribute(
                OccupancySensing.AttributeDefs.occupancy.id,
                OccupancySensing.Occupancy(value),
            )

(
    QuirkBuilder("aqara", "lumi.sensor_occupy.agl1")
    .adds(DeviceTemperature)
    .adds(OccupancySensing)
    .replaces(OppleCluster)
    .number(
        OppleCluster.AttributeDefs.approach_distance.name,
        OppleCluster.cluster_id,
        min_value=0,
        max_value=6,
        step=0.1,
        unit=UnitOfLength.METERS,
        multiplier=0.01,
        device_class=NumberDeviceClass.DISTANCE,
        fallback_name="Approach distance",
    )
    .enum(
        OppleCluster.AttributeDefs.motion.name,
        AqaraMotion,
        OppleCluster.cluster_id,
        entity_platform=EntityPlatform.SENSOR,
        entity_type=EntityType.STANDARD,
        fallback_name="Motion",
    )
    .sensor(
        OppleCluster.AttributeDefs.motion_distance.name,
        OppleCluster.cluster_id,
        unit=UnitOfLength.METERS,
        multiplier=0.01,
        device_class=SensorDeviceClass.DISTANCE,
        state_class=SensorStateClass.MEASUREMENT,
        fallback_name="Motion distance",
    )
    .enum(
        OppleCluster.AttributeDefs.motion_sensitivity.name,
        AqaraMotionSensitivity,
        OppleCluster.cluster_id,
    )
    .write_attr_button(
        OppleCluster.AttributeDefs.reset_no_presence_status.name,
        1,
        OppleCluster.cluster_id,
    )
    .write_attr_button(
        OppleCluster.AttributeDefs.restart_device.name,
        0,
        OppleCluster.cluster_id,
        # entity_type=EntityType.DIAGNOSTIC,
        fallback_name="Restart device",
    )
    .add_to_registry()
)
```

DemianTinkiel, 2024-10-07:
I tried @jeverley quirk at https://github.com/zigpy/zha-device-handlers/issues/3294#issuecomment-2394712555 first because I liked the entites shown in the screenshot but it did not work for me: only the identify button was exposed as entity

https://github.com/klolik/zha-device-handlers/commit/49d66b678ff07f58c0a14e065241b29efea4ec82 is working with my FP1E

jeverley, 2024-10-07:
> I tried @jeverley quirk at https://github.com/zigpy/zha-device-handlers/issues/3294#issuecomment-2394712555 first because I liked the entites shown in the screenshot but it did not work for me: only [...]

Thanks for testing, did you get any errors logged at all? Also which version of home assistant core are you running?

DemianTinkiel, 2024-10-07:
No errors that I could see (xiaomi_miio is causing a lot of noise at the moment, sorry!)
core is  2024.9.0

I also tried your v2.0 in https://github.com/zigpy/zha-device-handlers/issues/3294#issuecomment-2381449768, that works!

DemianTinkiel, 2024-10-07:
> Ahh that would explain it, I think the support for the fallback_name argument was only introduced in 2024.10 (it likely fails on entity creation on previous versions) 🙂

ok, I'll update my HA instance later on in the week and get back to you :+1:

ScottCLT, 2024-10-12:
I'm a HA user for years, but I'm not sure where to put this quirk information. I've copied the code, but have no idea where to paste it. Can you guy give me some brief direction on this?

Thanks,

-Scott

brofeelgood, 2024-10-12:
> I'm a HA user for years, but I'm not sure where to put this quirk information. I've copied the code, but have no idea where to paste it. Can you guy give me some brief direction on this?

Here's a decent write-up on how to set up quirks.

[https://community.home-assistant.io/t/zigbee-guide-how-to-add-setup-local-custom-zha-device-handlers-also-known-as-quirks-in-the-zha-integration/683473](
https://community.home-assistant.io/t/zigbee-guide-how-to-add-setup-local-custom-zha-device-handlers-also-known-as-quirks-in-the-zha-integration/683473)

Mayhem-SWE, 2024-11-06:
@jeverley your lovely quirks worked perfectly fine, up until just now when updating to HA Core 2024.11.0 seems to have broken something. All entities on the device are now listed as unavailable because they are no longer provided by the integration?! Unfortunately middle of the night here so will have to figure out what went wrong tomorrow and possibly roll back HA Core if required (and even possible)…

jviha, 2024-11-07:
I see the same thing, the update broke the functions of the sensor. Everything is unavailable. I tried to re-pair, restart, but no luck.

jeverley, 2024-11-07:
Hi all, the revised quirk code below should correct the issue introduced in the latest core release (it had made translation_key mandatory):

```
"""Quirk for aqara lumi.sensor_occupy.agl1."""

from __future__ import annotations

import logging
from typing import Any

from zigpy.quirks.v2 import (
    QuirkBuilder,
    NumberDeviceClass,
    SensorDeviceClass,
    SensorStateClass,
)
from zigpy.quirks.v2.homeassistant import EntityPlatform, EntityType, UnitOfLength
import zigpy.types as types
from zigpy.zcl.clusters.general import DeviceTemperature
from zigpy.zcl.clusters.measurement import OccupancySensing
from zigpy.zcl.foundation import ZCLAttributeDef

from zhaquirks.xiaomi import XiaomiAqaraE1Cluster

APPROACH_DISTANCE_ATTR_ID = 0x015B  # UINT32 The configurable maximum detection distance in millimeters (default 600 = 6 meters).
MOTION_ATTR_ID = 0x0160  # UINT8 Detected motion (0x02 = no movement, 0x03 = large movement, 0x04 = small movement)
MOTION_DISTANCE_ATTR_ID = 0x015F  # UINT32 Distance to the detected motion (mm)
MOTION_SENSITIVITY_ATTR_ID = 0x010C  # UINT8 The configurable detection sensitivity (0x01 = low, 0x02 = medium, 0x03 = high, default 0x03 = High)
OCCUPANCY_ATTR_ID = 0x0142  # UINT8 Occupancy detected 0x0 = absence, 0x01 = presence
RESET_NO_PRESENCE_STATUS_ATTR_ID = 0x0157  # UINT8 Trigger AI spatial learning
RESTART_DEVICE_ATTR_ID = 0x00E8  # BOOL Trigger device restart

_LOGGER = logging.getLogger(__name__)

class AqaraMotionSensitivity(types.enum8):
    """Aqara motion sensitivity."""

    Low = 0x01
    Medium = 0x02
    High = 0x03

class AqaraMotion(types.enum8):
    """Aqara motion."""

    Unknown_0 = 0x00
    Unknown_1 = 0x01
    Idle = 0x02
    Moving = 0x03
    Still = 0x04

class OppleCluster(XiaomiAqaraE1Cluster):
    """Aqara manufacturer cluster for the FP1E presence sensor."""

    class AttributeDefs(XiaomiAqaraE1Cluster.AttributeDefs):
        """Aqara occupancy sensor manufacturer specific attributes."""

        approach_distance = ZCLAttributeDef(
            id=APPROACH_DISTANCE_ATTR_ID,
            type=types.uint32_t,
            access="rw",
            is_manufacturer_specific=True,
        )

        motion = ZCLAttributeDef(
            id=MOTION_ATTR_ID,
            type=types.uint8_t,
            access="rp",
            is_manufacturer_specific=True,
        )

        motion_distance = ZCLAttributeDef(
            id=MOTION_DISTANCE_ATTR_ID,
            type=types.uint32_t,
            access="rp",
            is_manufacturer_specific=True,
        )

        motion_sensitivity = ZCLAttributeDef(
            id=MOTION_SENSITIVITY_ATTR_ID,
            type=types.uint8_t,
            access="rw",
            is_manufacturer_specific=True,
        )

        occupancy = ZCLAttributeDef(
            id=OCCUPANCY_ATTR_ID,
            type=types.uint8_t,
            access="rp",
            is_manufacturer_specific=True,
        )

        reset_no_presence_status = ZCLAttributeDef(
            id=RESET_NO_PRESENCE_STATUS_ATTR_ID,
            type=types.uint8_t,
            access="w",
            is_manufacturer_specific=True,
        )

        restart_device = ZCLAttributeDef(
            id=RESTART_DEVICE_ATTR_ID,
            type=types.Bool,
            access="w",
            is_manufacturer_specific=True,
        )

    def _update_attribute(self, attrid: int, value: Any) -> None:
        """Update the OccupancySensing cluster."""
        super()._update_attribute(attrid, value)
        if attrid == OCCUPANCY_ATTR_ID:
            self.endpoint.occupancy.update_attribute(
                OccupancySensing.AttributeDefs.occupancy.id,
                OccupancySensing.Occupancy(value),
            )

(
    QuirkBuilder("aqara", "lumi.sensor_occupy.agl1")
    .friendly_name(model="Presence Sensor FP1E", manufacturer="Aqara")
    .adds(DeviceTemperature)
    .adds(OccupancySensing)
    .replaces(OppleCluster)
    .number(
        OppleCluster.AttributeDefs.approach_distance.name,
        OppleCluster.cluster_id,
        min_value=0,
        max_value=6,
        step=0.1,
        #unit=UnitOfLength.METERS,
        multiplier=0.01,
        device_class=NumberDeviceClass.DISTANCE,
        translation_key="approach_distance",
        fallback_name="Approach distance",
    )
    .enum(
        OppleCluster.AttributeDefs.motion.name,
        AqaraMotion,
        OppleCluster.cluster_id,
        entity_platform=EntityPlatform.SENSOR,
        entity_type=EntityType.STANDARD,
        translation_key="motion",
        fallback_name="Motion",
    )
    .sensor(
        OppleCluster.AttributeDefs.motion_distance.name,
        OppleCluster.cluster_id,
        #unit=UnitOfLength.METERS,
        multiplier=0.01,
        device_class=SensorDeviceClass.DISTANCE,
        state_class=SensorStateClass.MEASUREMENT,
        translation_key="motion_distance",
        fallback_name="Motion distance",
    )
    .enum(
        OppleCluster.AttributeDefs.motion_sensitivity.name,
        AqaraMotionSensitivity,
        OppleCluster.cluster_id,
        translation_key="motion_sensitivity",
        fallback_name="Motion sensitivity",
    )
    .write_attr_button(
        OppleCluster.AttributeDefs.reset_no_presence_status.name,
        1,
        OppleCluster.cluster_id,
        translation_key="reset_no_presence_status",
        fallback_name="Presence status reset",
    )
    .write_attr_button(
        OppleCluster.AttributeDefs.restart_device.name,
        0,
        OppleCluster.cluster_id,
        #entity_type=EntityType.DIAGNOSTIC,
        translation_key="restart_device",
        fallback_name="Restart device",
    )
    .add_to_registry()
)

```

Mayhem-SWE, 2024-11-18:
@jeverley if you haven't yet, you should submit this quirk to be included with ZHA.

closed 2024-11-26 by TheJulianJES

Mayhem-SWE, 2024-12-06:
For anyone who had this custom quirk installed, be aware that upgrading to the just released HA 2024.12.0 breaks ZHA. With an updated version of the quirk itself now being shipped with HA, it seems ZHA is not equipped to handle multiple matching quirks and refuses to start.

As this was the first custom quirk I ever installed I have no idea if this is really the intended behaviour, but it definitely is user un-friendly and ought to be changed. Removing the custom quirk and rebooting fixes things and the presence sensor still works with the now official quirk. Thanks @jeverley !

```
2024-12-06 10:29:01.736 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SkyConnect for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 132, in async_setup_entry
    async with radio_mgr.connect_zigpy_app() as app:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 182, in connect_zigpy_app
    app = await self.radio_type.controller.new(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        app_config, auto_form=False, start_radio=False
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 239, in new
    await app._load_db()
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 111, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.13/site-packages/zigpy/appdb.py", line 684, in load
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/__init__.py", line 43, in get_device
    return _DEVICE_REGISTRY.get_device(device)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/registry.py", line 125, in get_device
    raise MultipleQuirksMatchException(
        f"Multiple matches found for device {device}: {matches}"
    )
zigpy.exceptions.MultipleQuirksMatchException: Multiple matches found for device <Device model='lumi.sensor_occupy.agl1' manuf='aqara' nwk=0x24B7 ieee=54:ef:44:10:00:db:c7:67 is_initialized=True>: [QuirksV2RegistryEntry(quirk_file=PosixPath('/usr/local/lib/python3.13/site-packages/zhaquirks/xiaomi/aqara/motion_agl1.py'), quirk_file_line=141, manufacturer_model_metadata=(ManufacturerModelMetadata(manufacturer='aqara', model='lumi.sensor_occupy.agl1'),), friendly_name=FriendlyNameMetadata(model='Presence Sensor FP1E', manufacturer='Aqara'), device_alerts=(), filters=(), custom_device_class=None, device_node_descriptor=None, skip_device_configuration=False, adds_metadata=(AddsMetadata(cluster=<class 'zigpy.zcl.clusters.general.DeviceTemperature'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({})), AddsMetadata(cluster=<class 'zhaquirks.xiaomi.aqara.motion_agl1.OccupancySensingLocal'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({})), AddsMetadata(cluster=<class 'zhaquirks.xiaomi.aqara.motion_agl1.IasZoneLocal'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({}))), removes_metadata=(), replaces_metadata=(ReplacesMetadata(remove=RemovesMetadata(cluster_id=64704, endpoint_id=1, cluster_type=<ClusterType.Server: 0>), add=AddsMetadata(cluster=<class 'zhaquirks.xiaomi.aqara.motion_agl1.OppleCluster'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({}))),), replaces_cluster_occurrences_metadata=(), entity_metadata=(NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=64704, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='approach_distance', fallback_name='Approach distance', attribute_name='approach_distance', reporting_config=None, min=0, max=6, step=0.1, unit=<UnitOfLength.METERS: 'm'>, mode=None, multiplier=0.01, device_class=<NumberDeviceClass.DISTANCE: 'distance'>), ZCLSensorMetadata(entity_platform=<EntityPlatform.SENSOR: 'sensor'>, entity_type=<EntityType.STANDARD: 'standard'>, cluster_id=64704, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='motion_distance', fallback_name='Motion distance', attribute_name='motion_distance', reporting_config=None, divisor=1, multiplier=0.01, unit=<UnitOfLength.METERS: 'm'>, device_class=<SensorDeviceClass.DISTANCE: 'distance'>, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>), ZCLEnumMetadata(entity_platform=<EntityPlatform.SELECT: 'select'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=64704, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='motion_sensitivity', fallback_name='Motion sensitivity', enum=<enum 'AqaraMotionSensitivity'>, attribute_name='motion_sensitivity', rep [... cut ...] _name='reset_no_presence_status', attribute_value=1), WriteAttributeButtonMetadata(entity_platform=<EntityPlatform.BUTTON: 'button'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=64704, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='restart_device', fallback_name='Restart device', attribute_name='restart_device', attribute_value=0)), device_automation_triggers_metadata=frozendict.frozendict({}))]
```

jeverley, 2024-12-07:
> For anyone who had this custom quirk installed, be aware that upgrading to the just released HA 2024.12.0 breaks ZHA. With an updated version of the quirk itself now being shipped with HA, it seems [...]

Thanks for the heads up @Mayhem-SWE, looks like someone's created an issue for this in https://github.com/zigpy/zigpy/issues/1508