zigpy/zha-device-handlers #3295: [BUG] Custom Quirk for TS1201 stopps working after update to 2024.8

TrackGeeks, 2024-08-10:
### Bug description

After updating to core 2024.8 control of IR devices stopped. Traced it to an error code when executing the script to send the string to the IR blaster. Someone had already posted a fix for the quirk on the HA forum. It was suggested to also post it as a bug here. 

Custom quirk here - https://github.com/ferehcarb/zha-device-handlers/blob/794f52130d6a20817b5ef22128b7503365c10717/zhaquirks/tuya/ts1201.py

Modified entry on line 382 in custom quirk from:
last_learned_ir_code = t.CharacterString()
To:
last_learned_ir_code = t.CharacterString(value="")

After saving and reloading the quirk, IR control was restored. 

### Steps to reproduce

Execute any IR command

### Expected behavior

Expected IR command to be transmitted from IR blaster. No commands were transmitted. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
Logger: homeassistant.components.script.formuler_center
Bron: helpers/script.py:525
integratie: Script (documentatie, problemen)
Eerst voorgekomen: 16:40:11 (1 gebeurtenissen)
Laatst gelogd: 16:40:11

Formuler Center: Error executing script. Unexpected error for call_service at pos 1: 2
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 525, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 764, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2763, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2806, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1102, in _async_admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/websocket_api.py", line 1389, in issue_zigbee_cluster_command
    await zha_device.issue_cluster_command(
  File "/usr/local/lib/python3.12/site-packages/zha/zigbee/device.py", line 888, in issue_cluster_command
    response = await getattr(cluster, commands[command].name)(
                                      ~~~~~~~~^^^^^^^^^
KeyError: 2
```

</details>

Anotheronewritesthedust, 2024-08-11:
Here is the link to the HA forum referenced in the bug report.  This link includes a suggested code fix for the issue

https://community.home-assistant.io/t/zha-moes-universal-ir-remote/576968/24

TheJulianJES, 2024-08-11:
Reference to PR of the quirk:
- https://github.com/zigpy/zha-device-handlers/pull/2336

closed 2025-08-11 by github-actions[bot]