zigpy/zha-device-handlers #3298: [Device Support Request] Sonoff SWV-BSP Smart Water Valve

andy-81, 2024-08-11:
### Problem description

Currently only opening and closing the valve is possible however the Sonoff hub and Zigbee2MQTT support pulling in if there is an issue with the valve and the flow data from the valve which would be fantastic to be able to use in ZHA.

### Solution description

It would be great if a custom quirk could be created however I have struggled to follow along with the details of how to create the custom device handlers. I was hoping the work from Z2M may help point to how to pull the information but the information does not look similar to the Zwave standard where you require the address details which has further complicated me understanding how to write the custom driver. Any help in someone creating this custom handler would be appreciated.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

https://youtu.be/ofzIoxliNJo?si=9JpD2oPFi9j76muC

this video shows the details of what is shown in ZHA vs what is shown in Zigbee2MQTT

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

brodock, 2024-08-13:
Cluster `0xfc11` seems to be known by zigbee2mqtt project: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/sonoff.ts#L52-L71

brodock, 2024-08-19:
schedule seems to be handled as part of the custom cluster above in the following attribute `0x5008`: 

the logic to serialize/de-serialize its content can be seen here: 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/sonoff.ts#L297-L370

andy-81, 2024-08-22:
I did find reference in the zigbee2mqtt handler about the SWZ valve.

It does mention in the list about the water flow and the status which shows if it is leaking, no pressure... 

It can be found here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/sonoff.ts#L1104-L1138

jamesshannon, 2024-08-23:
This is the CL that added much of the code. Notice that they also had to edit the `fromZigbee.ts` file to add the `msFlowMeasurement`... uh... "thing" (I don't know much about zigbee). Not sure if equivalent work would need to be done here.

soyxan, 2024-08-28:
I am also interested in this device support. I own the device and can provide any information or perform any test if needed. Thanks!

benbancroft, 2024-08-30:
Hi all,

I've just created PR https://github.com/zigpy/zha/pull/187 for zha, which implements at least the flow rate sensor part of this request. I plan to look at implementing a device handler for `0xfc11` cluster features (i.e. issue status) next if I have some time in coming days/weeks, unless somebody else has already made some progress here?

benbancroft, 2024-08-31:
I've just pushed up https://github.com/zigpy/zha-device-handlers/pull/3340 and https://github.com/zigpy/zha/pull/189, which implements device quirk and Home Assistant sensors for valve leak status and water supply status. Other than some device weirdness around having to ensure manufacturer_specific is not set on attribute read, this wasn't too bad to add in the end.

This is working nicely for my irrigation setup so far:
[image]

That should hopefully be everything that was initially requested for this request. There are however other features supported in Zigbee2MQTT (scheduling, shutoff timer etc) that probably could be added as a future enhance, which again I might look at if I get some time.

soyxan, 2024-09-01:
Thanks a lot, that is a good improvement. Now a newbie question, when will this quirck be available in Home Assistant? Can I manually upgrade ZHA or do I have to wait for a new HA release?

On the other hand, regarding the other features like scheduling, I think that is quite important to have them available. Rely in HA for scheduling the irrigation maybe is not the best option as any failure will affect our plants wellbeing :) Been able to define and save the schedules in the SWV device will keep the irrigation working even if HA or the Zigbee network is completely dead. I do not have the skills to do it by my own, but I am available for testing as I own the SWV.

aaronegger, 2024-09-02:
Is there any chance to get the amount of water (total liters)? That would be super usefull :D

jamesshannon, 2024-09-03:
My two cents regarding scheduling, specifically turn-on scheduling. The device will be fairly simple (e.g., time of day, though it might have another way to schedule). It definitely won't have sun-based scheduling, forecast-based scheduling, etc. So it's kinda a weird User Experience if it supports some on-device scheduling but not others.

Now... if you're talking about "time / flow *limits*" then yes, I agree that's super important. When I turn on the faucet (via HA) then I want to be able to send a "turn off in X" parameter as part of the "on" message. That way if something happens to the network then the device still turns off. I'm a little concerned about my plants, but very concerned about my water running for days. The issue could be something as simple as me rebooting HA at the time that my "turn off" automation is supposed to run.

TeDeVPrime, 2024-09-12:
nice! but how do we get updates for the ZHA in Hass? i don't see these options in my list

TeDeVPrime, 2024-09-24:
guys anyone who can tell us how to get the latest updates?

brodock, 2024-09-24:
it seems the related PRs haven't been merged yet

andy-81, 2024-09-30:
Any idea when this pr will be merged?

Would be great to gain access to the features @benbancroft added already. 😊

TeDeVPrime, 2024-10-06:
guys the new version from HASS came up but still my Sonoff Smart Water Valve just shows on / off switch. nothing more.

no i need to do something?

jamesshannon, 2024-10-09:
The pull requests haven't been approved. After they're approved and merged into the production branch then the updated version of zigpy will need to be included in ZHA, and then ZHA will need to be included in HA. In theory, all these things could happen quickly, but they require approval and merging.

TeDeVPrime, 2024-10-09:
i find it crazy that a pull request would need months.
ZHA should be able to be updated on it's own and not as part of HA. 
what if we need a hotfix?

we have had two HA releases since this was fixed. now i have to wait for November (if i am lucky) in order to be able to control my watering safely.

two times it has failed me until now and the water left running.

jamesshannon, 2024-10-09:
I can't speak to the approval process. It might have just gotten forgotten about.

I don't know why you're having so many problems. If you have an HA automation set up that turns it on, then waits, then turns it off, you should only get failures if:
a) The zigbee connection is really flaky. But if that's the case then you probably have failed turn-ons, too?
b) HA dies (or gets restarted, etc) during the on time and then the automation doesn't finish to turn it off.

For:
a)  I believe that Zigbee has a bi-directional mechanism and includes confirmation, right? In which case you should be able to build a check-and-retry into your automation. Ie, "repeat: turn off. wait 10 seconds. if it's off then exit" 

b) You could build a "failsafe off" automation to run, e.g., every 15 minutes and turn the water off. The more frequently it runs and the longer your watering period the more likely you are to have issues with this (ie, you'll turn off while it should still be one). If that's a problem, you could add a datetime helper or something as a second check.

TeDeVPrime, 2024-10-09:
i have two failures 

1 was because my HASS crashed..... lucky eh?
and 1 was because the device became "unavailable" for whatever reason.

i also have a friend who had this issue with bad signal

so i don't want to be dependent on critical things like that. since the device supports natively the solution to my problem.... and since it is fixed... i want that hotfix. because to me... this is a hotfix.

so having all these steps for "approval" for something that i should be able to access as beta or early-release..... it's annoying. especially when HASS has update for every single thing.... except ZHA.

soyxan, 2024-10-09:
First of all I totally agree with the comments regarding the long waiting till we get the update, but, in addition to that and AFAIK, the PR https://github.com/zigpy/zha-device-handlers/pull/3340 only covers basic feedback sensors from the device (flow measurement and leakage status) but no specific commands like "turn on with shutdown timer" as some of you are expecting. Please correct me if I am wrong.

TeDeVPrime, 2024-10-09:
you are correct. i am hoping that someone will soon figure it out. from what i have read, sonoff made it in a way that when you send the turn on command... you can also pass parameters for water flow and time.

the sensors that "exists" at the moment will give me an extra way of checking the system on event changes. 
it will also give me notifications as to whether i have running water or not at my house (it's a weekend house, so i am not there every single day).

i like the simplicity of ZHA but man.... Z2M had these features since day one. 

i have never gotten to the technical details between these two (ZHA and Z2M). i just found the ZHA easier and faster to use. 
Hopefully we will see updates on this device soon

fgsch, 2024-10-10:
Can we please move the discussion elsewhere, e.g., GH discussions?
If you want to be notified of updates just subscribe to the PR.  Protesting here doesn't help.

marcforn, 2024-11-07:
I'm also waiting for this to be merged

micz, 2025-03-04:
I created PR #3930 that adds these features:

- Water Leakage binary sensor
- Water Shortage binary sensor
- `on_time` attribute, representing the number of seconds after opening before the valve automatically closes.
- A custom behaviour for the On command:
   1. If `on_time == 0` it's a normal On command (`0x01`)
   2. If `on_time != 0` it sends a on_with_timed_off (`0x42`), with the `on_time` numbers of seconds sets in the attribute

In my environment everything works, could someone help testing it?
Thank you!

Stothed775, 2025-03-20:
Has an “open for” function already been implemented? So that the valve closes again automatically after a set time?

In my opinion, this is a very important safety feature

micz, 2025-03-20:
With PR #3930, I tried to implement that. It has been labeled "waiting for reviewer response".

I'm setting up the system for the spring and plan to use it as a custom quirk.

PrimesoftCopilot, 2025-03-20:
it is good seeing someone trying to make this device work properly.
ZHA "zigpy" has been a mess around this device since day one.
Z2M had full support since day one.
6-7 months after product release and we are still waiting for "approvals".

i must be missing a lot of knowledge or information because the fact that we have to wait for someone's "approval - review"... and then wait for the next version of HASS that will include the changes.... it is simply nuts.

Micz thank you for your effort. I don't know if we are going to magically see the changes when they happen or we have to reconfigure the device but at least you made a progress.

etienne678, 2025-03-20:
i took the quirk code from PR #3930 and edited it to hardcode a maximum of 20 minutes valve open time and use it always when the device is turned on, so that even if i forget or connection is lost it turns off after 20 mins. I have this running in my garden with 4 of those devices. it works well for me like this, it should be a setting in the UI. unfortunately setting the kwargs["on_time"] through the UI did not work in my case.

micz, 2025-03-20:
@etienne678 which firmware version do you have on the SWVs?

etienne678, 2025-03-20:
@micz  i tried to check, zha only gives firmware "unknown". i updated it through the ewelink app last week before i switched them over to HA / ZHA. if you let me know how to look it up i can post it here.

micz, 2025-03-20:
I have 0x00001004, maybe that's why it's not working. I never tried it with 1003.

etienne678, 2025-03-20:
[image]
no newer version available, at least in HA.

micz, 2025-03-21:
Before starting the work on this quirk I set up a test network with a spare coordinator using zigbee2mqtt to see how the device works there. So the device was updated to 1004.
Then I setup a test network with ZHA dev environment to work on the quirk, but the device was already using 1004.

Antriel, 2025-05-27:
What needs to happen to get this done, and get the new firmware to be available via ZHA?

austwhite, 2025-07-17:
I am using ZHA and updated to Firmware 1004 now.   The device was fine on 1003, but Itead / SOnOff did add the option to disable/change the auto-shutoff on low water flow.   This was an update on 1004
This option is available in Zigbee2MQTT, but not available in ZHA yet.
I saw about a quirk above, but not 100% sure how to implement it.

petrikd, 2025-07-20:
Hi All!
Maybe this is not the right thread - But does anyone know why is "Unknown" the SWV firmware in ZHA (for me)?
I'm really interested in this topic - can wait to possibility to set the auto close time :D

austwhite, 2025-07-20:
@petrikd 
Likely firmware settings in your configuration.yaml file, but this is not the place to discuss it.
This is a request for a modification to a driver (which is ongoing) and not a discussion forum. 
There should be plenty of discussion forums if you do a google search.   That's how I found the correct lines to add to my file   :)

TeDeVPrime, 2025-07-23:
is there any news for this?
i don't even see an update for this device in HASS... i don't see any other options other than the basic....

X-Ryl669, 2026-05-23:
Notice that for my SWV-ZFE devices, by default:
1. Switching the valve on auto switches off after 10mn (there's a default 10mn timer off in the device)
2. Using OnTime / Offtime (0x4001, 0x4002) attributes of cluster id 0x06 (OnOff) isn't supported by the device, opposite to what the quirks file in #4927 is proposing. The device answer with an Unsupported Attribute error
3. This 10mn timer is a default, but I haven't found a way to disable it. Seems like @ausfas found out that using a "On with Timed off" command (0x42) on cluster id 0x6 works, but in a unusual way: you need to set the "On/Off control" to 0 and set both the "On time" and "Off time"). So switching the valve on with "On" command (0x01) doesn't work as expected (it turns the valve on, but auto turns off for a non-user selected time which breaks irrigation schedules for example). It is **mandatory** to set a timer for using this device and none of the quirks above do that.

Looking at [Zigbee protocol](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf) page 3-56, and opposite to @ausfas is doing [here](https://github.com/zigpy/zha-device-handlers/pull/3930#issuecomment-4250761470), the `ontime` and `offtime` are supposed to be in 1/10 of seconds (so a value of 600 is 60s not 10mn).