zigpy/zha-device-handlers #3301: [Device Support Request] Tuya 214C Ultrasonic water meter with valve

netsoft-ruidias, 2024-08-14:
### Problem description

This is a Smart Ultrasonic Water Flow Meter with Tuya Zigbee App Control.
The device can be paired with ZHA, but no sensors are available.

The device:
https://www.alibaba.com/product-detail/Smart-Ultrasonic-Water-Flow-Meter-With_1600722839075.html

### Solution description

Add support for this device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_vuwtqx0t",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

NdS-Research-Facilities, 2024-08-20:
same for https://www.aliexpress.com/item/1005007353486146.html
213E device is similar:

`[0xed9d] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)`

NdS-Research-Facilities, 2024-08-21:
`  "data": {
    "ieee": "**REDACTED**",
    "nwk": 55451,
    "manufacturer": "_TZE200_zlwr0raf",
    "model": "TS0601",
    "name": "_TZE200_zlwr0raf TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": 112,
    "rssi": -72,
    "last_seen": "2024-08-20T21:03:01",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_zlwr0raf",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "update.tze200_zlwr0raf_ts0601_firmware",
        "name": "_TZE200_zlwr0raf TS0601"
      }
    ],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],`

NdS-Research-Facilities, 2024-08-21:
part of solution might be here: https://github.com/Koenkk/zigbee2mqtt/issues/21255

alrubaa, 2024-09-28:
Has anyone managed to get a ZHA quirk working for the 213E Tuya meter (TS0601 TZE200_zlwr0raf )?
I have been trying to adapt a quirk working by trial and error due to lack of experience in quirks, but I have not got it working, all I am after is the total water used and battery status, but could not get this working. 
Any help will be greatly appreciated.
Thank you.
AJ Ruba

develtop, 2024-12-18:
There is a implementation for this device for Tuya Local.
Maybe this could help for ZHA as well:

https://github.com/make-all/tuya-local/blob/main/custom_components/tuya_local/devices/zigbee_ultrasonic_watermeter.yaml

prairiesnpr, 2024-12-18:
> There is a implementation for this device for Tuya Local. Maybe this could help for ZHA as well:
> 
> https://github.com/make-all/tuya-local/blob/main/custom_components/tuya_local/devices/zigbee_ultrasonic_watermeter.yaml
> 
> Issue: [make-all/tuya-local#2412](https://github.com/make-all/tuya-local/issues/2412)

That's a nice DP listing, should be easy to build out a v2 quirk from that. See https://github.com/zigpy/zha-device-handlers/wiki/Tuya-%E2%80%90-v2-Quirk-with-TuyaQuirkBuilder and https://github.com/zigpy/zha-device-handlers/blob/dev/tuya.md also the current quirks here would be a good starting point https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_valve.py

develtop, 2025-01-15:
Unfortunately i don't have the skills for this. Just an old school developer with no clou of python or zha  :-(
I own the device and would be happy to test.

jurekjure, 2025-02-11:
Just received my Zemismart 214C today, and was disappointed to see that all I get through ZHA is a Firmware entity :(

I would love to test if anybody has time to build ...

jurekjure, 2025-05-22:
I have it working but sadly only through Tuya. It is the only device I dont have integrated with HA. It works as promised, I have it set to report the consumption every 12 hours but It can be set to as low as 1 hour. Its clean, well built, with no effect on water as far as I can tell.

PR #3888 (2025-02-21): Add Tuya ultrasonic valve `_TZE200_zlwr0raf`
## Proposed change

Add Tuya ultrasonic valve `_TZE200_zlwr0raf`

Based on: https://github.com/make-all/tuya-local/blob/main/custom_components/tuya_local/devices/zigbee_ultrasonic_watermeter.yaml

## Additional information

Closes: #3301

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works