zigpy/zha-device-handlers #3302: [Device Support Request] Tuya Smart Air Detector

lfarkas, 2024-08-16:
### Problem description

this air sensor: TS0601 by _TZE204_yvx5lh6k
https://www.aliexpress.com/item/1005005123624201.html?spm=a2g0o.order_list.order_list_main.5.1ff11802snuqlb
is not in the list:
https://github.com/zigpy/zha-device-handlers/blob/156aaa6bb704a13fe9500b833c293405fc5ffd79/zhaquirks/tuya/air/ts0601_air_quality.py#L37
almost the same included but would be useful to add this device too

### Solution description

add:
("_TZE204_yvx5lh6k", "TS0601"),

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

therealchunkyd, 2024-08-28:
Also following this waiting on support

closed 2024-10-20 by TheJulianJES

vinxion, 2024-11-05:
I linked this one but do not see the sensors, how to fix this?
With Z2M it is visible but I prefer ZHA.

PR #3440 (2024-10-20): Add Tuya air sensor `_TZE204_yvx5lh6k` variant
## Proposed change
add air sensor: TS0601 by _TZE204_yvx5lh6k

## Additional information
Fixes #3302

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works