zigpy/zha-device-handlers #3308: [Device Support Request] Sonoff SNZB-04p Zigbee Door/Window Sensor

lernerich, 2024-08-21:
### Problem description

The device successfully pairs, and the primary binary sensor operates as expected.
However, the secondary binary sensor (tamper switch) does not function correctly; it remains in a "closed" state regardless of tampering or not.

### Solution description

The tamper switch (rectius: secondary binary sensor) should detect and report any tampering by changing state from "closed" to "open" when triggered (or other proper states).

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4742,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 255,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 255,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0500",
        "0xfc11",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "SNZB-04P",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

[... 426 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

The official Sonoff webpage shows the tamper alert doesn't work in ZHA: https://sonoff.tech/product/gateway-and-sensors/snzb-04p/

lernerich, 2024-09-04:
Despite the fact that two binary sensors appear on the ZHA, the second non-functional sensor is not the tamper.
The tamper has a manufacturer-specific cluster (0xFC11).
In another sensor's quirk (Sonoff SNZB-06), this cluster is named as follows:
SONOFF_CLUSTER_FC11_ID = 0xFC11
There, a custom cluster replaces the original ManufacturerSpecificCluster.
With the same technique, it is possible to read the sensor state, as an attribute, in the system. But I was unable to configure a sensor in Home Assistant with these binary values.

Nbird67, 2024-12-06:
Some news about the support ?
Thank you

mguaylam, 2025-01-02:
Right now I can see 1 or 2 issue but should not be incredibly hard to fix.

- Reporting look not right for the opening. It does report but there does not seem to be anything periodic.
- We need to gather attributes information since some stuff seem to be implemented in manufacturer specific clusters.
  - According to this code : https://github.com/zigbeefordomoticz/z4d-certified-devices/commit/c9725170f7504fe29e9367cfe462122ca75d11b7 the cluster for the tamper is fc11 and the attribute should be 2000.
- There is a manufacturer specific cluster for 0xFC57 but no indication what it is for.
- We need to find what's creating the second opening entity and it's probably not the tamper. But there is a IAS cluster and _ I think_  it should at least try to create a contact entity from it.

I have written to ITead support to have some information on this device.
Tamper will probably be done via the FC11 with a v2 quirk but we will need to implement reporting for it in ZHA.

mguaylam, 2025-01-02:
I tried to configure reporting on the IAS zone state but it does not look promising.

mguaylam, 2025-01-02:
Hey there @liangjia2019, sorry to bother you.

I saw you implemented the ZigBee2MQTT part for this device for ITead, would you mind sharing cluster details about this device so I can properly handle it? Especially 0xFC57 and 0xFC11?

**More importantly**, would it be possible to update the device to do reporting on the IAS cluster for every 10-15 minutes like the SNZB-05P?
The device does not report again when there is a network failure and is a security issue for the house.

liangjia2019, 2025-01-03:
> Hey there @liangjia2019, sorry to bother you.
>
> I saw you implemented the ZigBee2MQTT part for this device for Itead, would you mind sharing cluster details about this device so I can properly handle [...]

Hi mguaylam , I'm glad to receive your message.

Here are some relevant information I know:
tamper    cluster:0xfc11   attribute:0x2000   datatype:0x20   value:  0x00 installation 0x01 removal

Cluster: 0xFC57  was Works with All Hubs (WWAH) Cluster,He currently only works on Amazon Hub,You can ignore.

He is currently unable to report to the cluster within 10-15 minutes, but he may achieve it in the near future.

mguaylam, 2025-01-03:
Thank you for your fast reply @liangjia2019! Really appreciate your help on this matter.
I just finished integrating the tamper sensor and the reporting.

Unfortunately I see **two major issues** :
- No reporting interval of less than an hour on `cluster 0x0500` ➡️ `attribute 0x0002` (IAS)
- Reporting configuration is **refused** on `cluster:0xfc11` ➡️ `attribute:0x2000` (tamper) 
  - result : `success: false` ➡️ currently at `min_interval: 1`, `max_interval: 65534` and `reportable_change: 0`

Things that need to be fixed in the firmware :

- Reporting `max interval` for `cluster 0x0500` ➡️ `attribute 0x0002` to `900` (15 minutes or something reasonable)
- Allow reporting configuration for `cluster 0xfc11` ➡️ `attribute 0x2000` (no constraint)

We would love to have your help on this matter in order to make this sensor a great recommendation for ZHA. Would you guys be interested into fixing those issues?

Thank you for your help.

liangjia2019, 2025-01-03:
> Thank you for your fast reply @liangjia2019! Really appreciate your help on this matter. I just finished integrating the tamper sensor and the reporting.
>
> **Unfortunately I see two major issues :** [...]

They will solve it in the near future. Thank you for your feedback.

PR #3665 (2025-03-22): Add Sonoff SNZB-04P contact sensor
## Proposed change

Implement the tamper sensor of the device and disable the entity generation from the on/off cluster (it does not work).

## Additional information

Fixes #3308
~~Requires https://github.com/zigpy/zha/pull/345~~

## Checklist

- [X] The changes are tested and work correctly
- [X] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-03-22 by TheJulianJES