zigpy/zha-device-handlers #3309: [Device Support Request] TZE204_ex3rcdha presence and lux sensor

agf102, 2024-08-22:
### Problem description

I've just bought a couple of these from Aliexpress (https://www.aliexpress.com/item/1005007335302133.html) and have been playing around to see if they work in HA / ZHA.  I copied a quirks file I found on the HA forum and added the model references in.  

I can now see controls for detection delay; fading time; max / min range; sensitivity and can see sensors for illuminance (shows unknown), occupancy (shows clear) and firmware (shows up to date)

### Solution description

Correctly operating illuminance and occupancy sensors

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
35
36
37
38
39
40
41
42
43
44
45
46
47
48
[... 1491 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

</details>

dermaxxx213, 2024-08-26:
Sorry for my question, but works the sensor with this quirk? 
must i copy the hole code or only a part? 
i try both, but it didnt work.

agf102, 2024-09-02:
> Sorry for my question, but works the sensor with this quirk? must i copy the hole code or only a part? i try both, but it didnt work.

Using the whole quirk included in this post I was able to get partial functionality but the illuminance and occupancy sensors don't work and I can't therefore test whether the controls are working either

lajoshanko, 2024-09-11:
Seems to be the z2mqtt version is working, maybe it can speed up the zha support too:
https://github.com/Koenkk/zigbee2mqtt/issues/23705#issuecomment-2344137902

przemek-baranowski, 2024-10-09:
signature in ZHA looks like this:

`
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_ex3rcdha",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
`

esrfkos, 2024-10-17:
Hey Guys is there any update on this issue?
Would be wonderful if someone could share their progress.

esrfkos, 2024-10-17:
Here is a custom quirk i tried to put together with chatGPT:

`from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Scenes, Ota, Time
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaMCUCluster

class TuyaCustomCluster(TuyaMCUCluster):
    """Tuya Manufacturer-specific cluster for 0xEF00."""

    # This handles attribute reporting and command mapping for the Tuya cluster
    attributes = {
        0x0000: ('tuya_attribute_1', 'Type', bool),
        0x0001: ('tuya_attribute_2', 'Type', int),
    }
    client_commands = {}
    server_commands = {}

class TZE204EX3RCDHA(CustomDevice):
    """Custom device representing Tuya-based TS0601 device."""

    signature = {
        "endpoints": {
            1: {
                "profile_id": 0x0104,
                "device_type": 0x0051,
                "input_clusters": [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCustomCluster.cluster_id,
                ],
                "output_clusters": [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
        "manufacturer": "_TZE204_ex3rcdha",
        "model": "TS0601",
    }

    replacement = {
        "endpoints": {
            1: {
                "profile_id": 0x0104,
                "device_type": 0x0051,
                "input_clusters": [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCustomCluster,
                ],
                "output_clusters": [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
`

tonyhollaar, 2024-10-24:
I have not got your script @esrfkos working. I have found the below script which at least shows in Home Assistant the correct buttons but does not actually work for this motion sensor. Is there anybody that can solve this e.g. find the correct data_point_handlers and fix the code? 

```
"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""
import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)

from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaManufCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

ZONE_TYPE = 0x0001

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""

    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""
[... 904 lines cut ...]
            111: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMinRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            112: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

wesleysee, 2024-11-15:
Here's a custom quirk I put together that seems to be working as far as I can tell.

```
import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
    MultistateInput,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    # TuyaDPType,
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaMmwPresenceState (TuyaAttributesCluster, MultistateInput):
    """Tuya local AnalogInput cluster."""

    def __init__(self, *args, **kwargs):
[... 273 lines cut ...]
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarBreathMotionMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarBreathMotionMinRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {                                    
                PROFILE_ID: 41440,                                                
                DEVICE_TYPE: 97,                                                      
                INPUT_CLUSTERS: [],                                        
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],                     
            }, 
        }
    }
```

lajoshanko, 2024-11-15:
For me it does not work the occupancy always clear no matter what

mdshw5, 2024-11-15:
@wesleysee That quirk works great for some of these devices that I just received today! Thanks for sharing and helping me get started with these cheap zigbee devices!

tonyhollaar, 2024-11-17:
@wesleysee works as well for me! Thank you so much...

lajoshanko, 2024-11-17:
Update: the light sensor works without a problem, the occupancy seemingly works sometime, but cannot modifie any of the provided values:
[image]

wesleysee, 2024-11-17:
The above error seem to be a more widespread issue, I'm seeing it across a variety of my Tuya devices. Also see it reported in this issue: https://github.com/home-assistant/core/issues/130218

lajoshanko, 2024-11-18:
After adding it a few times the occupancy sensor also started to work. Thanks fornthe quirk!

sportluffi, 2024-12-07:
> The above error seem to be a more widespread issue, I'm seeing it across a variety of my Tuya devices. Also see it reported in this issue: [home-assistant/core#130218](https://github.com/home-assistant/core/issues/130218)

This issue has been merged and released. It is now possible to change the fading time and the Motion max range from the HA UI.

Madmorgan007, 2024-12-08:
Hi I'm new to home assistant so how do I add custom quirk's ???

tonyhollaar, 2024-12-08:
> Hi I'm new to home assistant so how do I add custom quirk's ???

1) Open your folder of home assistant, i use  the seperately downloaded and installed add-on Samba to access my files from my PC to access my home assistant yellow drive/folder.
2) Make a new folder in your home assistant directory. In my case the folder is named “custom_zha_quirks”;
3) Make a new file named e.g. MYFILE.PY
(Can be any filename)

Paste the code from this forum of the custom quirk into it.

4) Open configuration. yaml with a text editor e.g. notepad++
add the following lines:
zha:
  enable_quirks: true
  custom_quirks_path: /config/
custom_zha_quirks
5) Restart home assistant from Settings menu
6) remove and re-add your device in home assistant as an e.g. zigbee device and then  press the black button on your human presence sensor untill it blinks
7) verify it works and use an automation to turn on/off the light

Madmorgan007, 2024-12-09:
Hello again, The script has worked and showing more settings now but the device's light is aways off and the occupancy shows as clear and does not change.

Rob4ik77, 2025-02-23:
@wesleysee  You are amazing! thank you mate! works like a charm

PR #3887 (2025-02-26): Add Tuya mmWave radar 5.8GHz, ZY_HPS01 `_TZE204_ex3rcdha`
## Proposed change

Adds Tuya mmWave radar 5.8GHz, ZY_HPS01 `_TZE204_ex3rcdha`

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/6c3d7e2aa1fba6f6342baea21e5b3bbd3688cb0e/src/devices/tuya.ts#L13228

## Additional information

Closes: #3309

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works