zigpy/zha-device-handlers #3316: [Device Support Request] Tuya water timer (TS0049 by _TZ3000_gjpgagal)

gunarser, 2024-08-25:
### Problem description

Please can a handler be created to support a Tuya water timer (TS0049). The device can be added to home assistant via ZHA but not all entities (like timer which by default is set to 10 minutes) are available to control the valve.

### Solution description

Creation of a new handler that supports the device

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_gjpgagal",
  "model": "TS0049",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.8.2",
    "dev": false,
[... 488 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I have a Tuya Zigbee gateway at my disposal that could be used for further troubleshooting if necessary.

danpeig, 2024-09-01:
Look at issue #2377

gunarser, 2024-09-02:
> Look at issue #2377

If that was meant for me then I already tried the quirk and commented on the results here: https://github.com/zigpy/zha-device-handlers/issues/2377#issuecomment-2258177255
It looks like there are some incompatibility issues, i.e. the device is quite different from the one the quirk was made for.

danpeig, 2024-09-02:
#2377 is the thread for TS0049. Developers that work with this type of device are subscribed to that issue.

As far as I am aware of, Tuya has only one specification for TS0049.

Most user problems are related  with binding the quirk to the device because different manufacturers have different signatures.

gunarser, 2024-09-02:
OK. Perhaps I was doing something wrong. I changed the manufacturer as per my comment but the quirk does not get loaded. What is the best way to troubleshoot such cases?

danpeig, 2024-09-02:
Do you have other custom quirks installed and working? I think the first step is ensure the system is actually loading the code.

My recommendation is use a quirk for a ZigBee device you already have working like the TS0001 or TS0011 (wall plugs and light switches) and see if it loads properly. The custom quirk will appear in the device information page.

Once the quirk are loading, then you can force the 0049 to bind to your specific device.

gunarser, 2024-09-02:
I have Nous light switches that require quirk and some time back I installed it manually and it was correctly loaded. But for some time the quirk is included in HA and added automatically. So I removed the custom quirk.
I don't have any other device that need custom quirk. Can I use those light switches for a test even though the quirk is already in HA?

This is how swithes look like in HA:

[image]

danpeig, 2024-09-02:
If you have a custom quirk it will be loaded instead of the built in. This is just to test the environment.

If it loads, then it's just about fingerprinting your device properly.

You can change the TS001x quirk (known to be working) to see if you manage to attach it to the water valve. The valve will obviously no work, but you will know the binding code.

Regards

gunarser, 2024-09-03:
I added the ts001x.py file to the config/custom_zha_quirks/ directory and it looks like the custom quirk is indeed loaded for light switches instead of the built-in one:
[image]

I modified the ts001x.py file by changing model ID of 3-gang switch from TS0013 to TS0049 but the quirk does not get loaded after HA restart. Should I re-pair the water timer to get the quirk loaded or is there something else that needs to be done in order to get the ts001x loaded?

danpeig, 2024-09-03:
The quirk should still be loading for the other devices. If it is not, you have a syntax problem with the code.

If it loads but not binds yo the valve, you have a problem with the "signature". The signature must match your valve device in order for it to bind.  The device name is only one parameter of the signature (see https://github.com/zigpy/zha-device-handlers).

gunarser, 2024-09-04:
I changed almost all the parameters in the signature both in TS0049.py and ts001x.py but none of the quirks gets loaded (at least they do not appear under Zigbee info tab).
How much of the signature must match in order to get quirk loaded?

This is the signature section of the modified TS0049 quirk:

```
class TuyaIrrigationValve(EnchantedDevice):
    """Tuya green irrigation valve device."""
    signature = {
        MODELS_INFO: [("_TZ3000_gjpgagal", "TS0049")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0
            # device_version=1
            # input_clusters=[0, 1, 3, 4, 5, 6, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Power.cluster_id,
                    Identity.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Null.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Power.cluster_id,
                    Identity.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    Null.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }`

```

This is how the device signature looks like in 'Manage Zigbee device' window:
```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_gjpgagal",
  "model": "TS0049",
  "class": "zigpy.device.Device"
}
```

danpeig, 2024-09-04:
If you use the text below in the signature it should bind:

`json
"profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
`

gunarser, 2024-09-04:
I finally managed to load and attach a custom quirk. I used this [quirk-generator](https://github.com/zigpy/quirk-generator) 
For beginners it is a great tool to get started.
Now the device info looks like this:

[image]

Now the question is how to add the missing attributes like timer setting to the quirk?

danpeig, 2024-09-04:
Great! Now you just need to get the replacements to work correctly. The timer will be set using the "Manage Zigbee Device" -> "Valve family cluster" -> irrigation_time. The On-Off switch will be exposed in the interface.

gunarser, 2024-09-05:
Should the TS0049 quirk provide the needed "Valve family cluster" or there is other quirk that should be used as an example? This is what I see under clusters (no value is read from the irrigation_time attribute):

[image]
[image]
[image]

danpeig, 2024-09-05:
Yes. It is probably not mapped correctly.

gunarser, 2024-09-05:
What would be the procedure to get the right mappings? Do I need Tuya Zigbee gateway for that?

danpeig, 2024-09-05:
The attribute mapping is the final step to get the valve working. There is no need to use a Tuya brand ZigBee router.

This link has the attribute map specs for the valve: https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/Zigbee-Water-valve-controller-access-standard?id=Kbahvojbqzvgl

gunarser, 2024-09-08:
I understand that I can additionally map/read/control the following attributes:
DP4 Fault report
DP11 Irrigation time
DP12 Work state
and perhaps DP15 Once using time

But I don't quite understand how that mapping is done via the quirk. Which lines variables need to be adjusted?

danpeig, 2024-09-08:
The data points are numbers, in the original code: 26, 101, 110, 111...etc.

Clusters are hexadecimal: 0xEF01, 0xEF02...

First you must set the clusters, then the DP.

gunarser, 2024-09-08:
I modified the following lines in the quirk:

````
class TuyaValveFamilyCluster(TuyaMCUCluster):
    """On/Off Tuya family cluster with extra device attributes"""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("countdown", t.uint32_t, True),
            0xEF02: ("work_state", t.uint32_t, True),
            0xEF03: ("fault", t.uint32_t, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        4: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "fault",
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "countdown",
        ),
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "work_state",
        ),
    }

    data_point_handlers = {
        4: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }
````

But I get no values in the corresponding attributes:

[image]

Value is none for all.

gunarser, 2024-09-11:
What am I doing wrong? Perhaps cluster numbers are defined wrong? I am total newbie to quirks, don't know how these mappings work.

> I modified the following lines in the quirk:
>
> ```
> class TuyaValveFamilyCluster(TuyaMCUCluster):
>     """On/Off Tuya family cluster with extra device attributes"""
>
>     attributes = [...]

danpeig, 2024-09-11:
Is your valve supported by ZigBee 2MQTT? For the other model, we used the mapping information from this tool. Otherwise it is trial and error.

gunarser, 2024-09-12:
I don't know. I found the following forum posts that might relate to my valve:
https://github.com/Koenkk/zigbee2mqtt/issues/23191
https://github.com/Koenkk/zigbee2mqtt/issues/16671
https://github.com/Koenkk/zigbee2mqtt/issues/22950
https://github.com/Koenkk/zigbee2mqtt/issues/21788

Should I install Z2M in order to understand whether my specific device is supported?

> Is your valve supported by ZigBee 2MQTT? For the other model, we used the mapping information from this tool. Otherwise it is trial and error.

danpeig, 2024-09-12:
I just got the Z2M source code.

gunarser, 2024-09-13:
Good! Tell if anything is needed from me. As I mentioned before I have a spare Tuya Zigbee gateway - could do some additional troubleshooting if necessary.

> I just got the Z2M source code.

gunarser, 2024-09-14:
In the meantime, I installed Z2M Docker container and paired the valve to it:

[image]

[image]

[image]

Should I collect some more info from Z2M in order to create ZHA quirk?

danpeig, 2024-09-14:
Z2M has this device properly mapped. You need to translate it to ZHA. This is what we did with the other valve. Unfortunately I don't remember how exactly that worked.

gunarser, 2024-09-14:
OK. Who might know?

danpeig, 2024-09-14:
Whoever committed this valve map to Z2M.

gunarser, 2024-09-14:
I connected the valve to Tuya Zigbee gateway and retrieved DP IDs as described in the following document: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

They were:
````
{"1":"Switch",
"4":"Failure to report",
"7":"Battery",
"9":"Accumulated usage time",
"10":"Weather Delay",
"11":"Irrigation time",
"12":"work state",
"15":"Once using time",
"16":"Cycle irrigation",
"17":"Normal  timer",
"29":"Next irrigation time"}
````

Basically they are the same as described in Tuya document: https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/Zigbee-Water-valve-controller-access-standard?id=Kbahvojbqzvgl

So the DPs are correct. The question is why the quirk does not map them correctly. What am I missing?!

gunarser, 2024-09-14:
Here is the complete modified quirk:

```
from typing import Any, Dict, Optional, Union

import zigpy.types as t
#from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TUYA_SEND_DATA, TuyaLocalCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaMCUCluster,
#    TuyaOnOff,
)
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Time, PowerConfiguration, Scenes

#class TuyaValveFamilyBattery(TuyaLocalCluster, DoublingPowerConfigurationCluster):
#    _values = [10, 50, 90]
#    _CONSTANT_ATTRIBUTES = {
#        PowerConfiguration.attributes_by_name["battery_quantity"].id: 4,
#        PowerConfiguration.attributes_by_name["battery_size"].id: PowerConfiguration.BatterySize.AAA
#    }
#
#    def _update_attribute(self, attrid, value):
#        if attrid == self.BATTERY_PERCENTAGE_REMAINING:
#            value = self._values[value]
#        super()._update_attribute(attrid, value)

class TuyaValveFamilyCluster(TuyaMCUCluster):
    """On/Off Tuya family cluster with extra device attributes"""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("countdown", t.uint32_t, True),
            0xEF02: ("work_state", t.enum8, True),
            0xEF03: ("fault", t.uint32_t, True),
        }
    )

    async def command(
            self,
            command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
            *args,
            manufacturer: Optional[Union[int, t.uint16_t]] = None,
            expect_reply: bool = True,
            tsn: Optional[Union[int, t.uint8_t]] = None,
            **kwargs: Any,
    ):
        """Override the default Cluster command."""
        self.debug("Setting the NO manufacturer id in command: %s", command_id)
        return await super().command(
            TUYA_SEND_DATA,
            *args,
            manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID,
            expect_reply=expect_reply,
            tsn=tsn,
            **kwargs,
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        4: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "fault",
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "countdown",
        ),
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "work_state",
        ),
    }

    data_point_handlers = {
        4: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class TuyaIrrigationValve(EnchantedDevice):
    """Tuya green irrigation valve device."""
    signature = {
        MODELS_INFO: [("_TZ3000_gjpgagal", "TS0049")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0
            # device_version=1
            # input_clusters=[0, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    0xe001,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaValveFamilyCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```

I commented out attribute definitions that are there already out-of-box like on/off switch.

joseruisa, 2025-02-14:
[image]

What this means?

prairiesnpr, 2025-02-14:
> What this means?

You should ask in the zigbee2mqtt repo.

closed 2025-08-22 by github-actions[bot]

ummarbhutta, 2025-09-29:
Hi @gunarser ,

Just an update. I installed `ZHA Toolkit` v 1.1.29 and based on documentation [here](https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/Zigbee-Water-valve-controller-access-standard?id=Kbahvojbqzvgl#title-12-DP11%20Irrigation%20time), I managed to change the pre-configured auto shutoff time, which was set to 10 minutes in my case also.

See example below to set auto shutoff time to 60 seconds. You can execute it from **Developer tools > Actions** tab.

### Request
```
action: zha_toolkit.zcl_cmd
data:
  ieee: "aa:aa:aa:aa:aa:bb:bb:bb"
  cluster: 0xE001
  cmd: 0xFE
  endpoint: 1
  args: [11,0,0,0,60]
```

### Response
```
zha_toolkit_version: v1.1.29
zigpy_version: 0.82.2
zigpy_rf_version: 0.14.1
ieee_org:
  - 44
  - 44
  - 44
  - 44
  - 44
  - 111
  - 111
  - 111
ieee: aa:aa:aa:aa:aa:bb:bb:bb
command: zcl_cmd
command_data: null
start_time: "2025-09-28T15:15:42.288588+00:00"
errors: []
params:
  cmd_id: 254
  endpoint_id: 1
  cluster_id: 57345
  dir: 0
  tries: 1
  expect_reply: true
  args:
    - 11
    - 0
    - 0
    - 0
    - 60
  kwargs: {}
  read_before_write: true
  read_after_write: true
cmd_reply:
  - 254
  - 0
success: true
```

Just that you know this is a manufacturer specific command (Client -> Server), and all quirks deal with exposing attributes. I know commands can be exposed as well, but I still don't have luck regarding that.

ummarbhutta, 2025-11-16:
Hi @gunarser 

Yes I have partial success. Built following quirk.

```
from __future__ import annotations
import struct
import logging
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Identify, Groups, Scenes, OnOff, PowerConfiguration, Ota
from zigpy.zcl import foundation
from zigpy.zcl.clusters import Cluster
from zigpy.exceptions import DeliveryError
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    OUTPUT_CLUSTERS,
    MODELS_INFO,
    PROFILE_ID,
)

from typing import Any, Dict, Optional, Union

import zigpy.types as t
#from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TUYA_SEND_DATA, TuyaLocalCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaMCUCluster,
#    TuyaOnOff,
)
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Time, PowerConfiguration, Scenes

# Tuya Data Points (DP) for Water Valve Controller
TUYA_DP_SWITCH = 1          # On/Off state
TUYA_DP_FAULT = 2           # Device fault status
TUYA_DP_WEATHER_DELAY = 9   # Weather delay setting
TUYA_DP_IRRIGATION_TIME = 11 # Irrigation time in seconds
TUYA_DP_MODE = 12           # Operation mode
TUYA_DP_REMAINING_TIME = 13 # Remaining irrigation time

# Weather delay options
WEATHER_DELAY_MAP = {
    0: "none",
    1: "24_hours", 
    2: "48_hours",
    3: "72_hours"
}

# Operation modes
OPERATION_MODE_MAP = {
    0: "schedule",
    1: "cycle", 
    2: "manual"
}
[... 196 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaElectricianPrivateCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

###############################################################
```
After this I see following as output in `Manage Zigbee device` section

[image]

[image]

[image]