zigpy/zha-device-handlers #3326: [Device Support Request] TS0601 by _TZE204_xnbkhhdr

ctechmat, 2024-08-28:
### Problem description

This device is currently not recognized, it only shows LQI and RSSI entities..
It is a smart thermostat from Avatto.

Link: https://www.aliexpress.com/i/1005005821437205.html

This device is supported by Z2MQTT:
* https://www.zigbee2mqtt.io/devices/ZWT198_ZWT100-BH.html
* https://github.com/Koenkk/zigbee2mqtt/issues/23611

### Solution description

Would be great to have a quirk for this.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_xnbkhhdr",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

bauderline, 2024-09-22:
Also, would like to see this supported, I can get it to at least display the temp if I add the signature to the existing Avatto quirk but unfortunately setting the temperature does not work.

bszdev, 2024-10-03:
I modified a previous custom quirk file (ZWT198 TRV) for _TZE204_xnbkhhdr.
Not the best code but now the following functions works well on my device:
- auto datetime sync
- thermostat functions (display and changes)
- temperature calibration
- display the current HVAC function (idle/heating)

The battery state does not work.
```python
import logging
from typing import Optional, Union

import zigpy.types as t
from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaManufCluster,
    TuyaManufClusterAttributes,
    TuyaPowerConfigurationCluster,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaTimePayload,
    TuyaUserInterfaceCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogOutput,
    Basic,
    Groups,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.hvac import Thermostat

_LOGGER = logging.getLogger(__name__)

ZWT198_TARGET_TEMP_ATTR = 0x0202  # target room temp (degree)
ZWT198_TEMPERATURE_ATTR = 0x0203  # current room temp (degree/10)
ZWT198_MODE_ATTR = 0x0404  # [0] schedule [1] manual
ZWT198_SYSTEM_MODE_ATTR = 0x0101  # device [0] off [1] on
ZWT198_HEAT_STATE_ATTR = 0x0465  # [0] heating icon off [1] heating icon on
ZWT198_CHILD_LOCK_ATTR = 0x0109  # [0] unlocked [1] locked
ZWT198_TEMP_CALIBRATION_ATTR = 0x0213  # temperature calibration (degree)
ZWT198ManufClusterSelf = {}

class CustomTuyaOnOff(LocalDataCluster, OnOff):
    """Custom Tuya OnOff cluster."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.thermostat_onoff_bus.add_listener(self)

    # pylint: disable=R0201
    def map_attribute(self, attribute, value):
        """Map standardized attribute value to dict of manufacturer values."""
        return {}

[... 364 lines cut ...]
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    ZWT198ManufCluster,
                    ZWT198Thermostat,
                    ZWT198UserInterface,
                    TuyaPowerConfigurationCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [ZWT198ChildLock],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.CONSUMPTION_AWARENESS_DEVICE,
                INPUT_CLUSTERS: [ZWT198TempCalibration],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

Mikodzi, 2024-10-30:
> I modified a previous custom quirk file (ZWT198 TRV) for _TZE204_xnbkhhdr. Not the best code but now the following functions works well on my device:
> 
>     * auto datetime sync
> 
>     * thermostat functions (display and changes)
> 
>     * temperature calibration
> 
>     * display the current HVAC function (idle/heating)
> 
>     * child lock function
> 
> 
> The battery state does not work.
> 
> ```python
> import logging
> from typing import Optional, Union
> 
> import zigpy.types as t
> from zhaquirks import Bus, LocalDataCluster
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
> )
> from zhaquirks.tuya import (
>     NoManufacturerCluster,
>     TuyaManufCluster,
>     TuyaManufClusterAttributes,
>     TuyaPowerConfigurationCluster,
>     TuyaThermostat,
>     TuyaThermostatCluster,
>     TuyaTimePayload,
>     TuyaUserInterfaceCluster,
> )
> from zhaquirks.tuya.mcu import EnchantedDevice
> from zigpy.profiles import zha
> from zigpy.zcl import foundation
> from zigpy.zcl.clusters.general import (
>     AnalogOutput,
>     Basic,
>     Groups,
>     OnOff,
>     Ota,
>     Scenes,
>     Time,
> )
> from zigpy.zcl.clusters.hvac import Thermostat
> 
> _LOGGER = logging.getLogger(__name__)
> 
> ZWT198_TARGET_TEMP_ATTR = 0x0202  # target room temp (degree)
> ZWT198_TEMPERATURE_ATTR = 0x0203  # current room temp (degree/10)
> ZWT198_MODE_ATTR = 0x0404  # [0] schedule [1] manual
> ZWT198_SYSTEM_MODE_ATTR = 0x0101  # device [0] off [1] on
> ZWT198_HEAT_STATE_ATTR = 0x0465  # [0] heating icon off [1] heating icon on
> ZWT198_CHILD_LOCK_ATTR = 0x0109  # [0] unlocked [1] locked
> ZWT198_TEMP_CALIBRATION_ATTR = 0x0213  # temperature calibration (degree)
> ZWT198ManufClusterSelf = {}
> 
> 
> class CustomTuyaOnOff(LocalDataCluster, OnOff):
>     """Custom Tuya OnOff cluster."""
> 
>     def __init__(self, *args, **kwargs):
>         """Init."""
[... 380 lines cut ...]
>                     Scenes.cluster_id,
>                     ZWT198ManufCluster,
>                     ZWT198Thermostat,
>                     ZWT198UserInterface,
>                     TuyaPowerConfigurationCluster,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             2: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
>                 INPUT_CLUSTERS: [ZWT198ChildLock],
>                 OUTPUT_CLUSTERS: [],
>             },
>             3: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.CONSUMPTION_AWARENESS_DEVICE,
>                 INPUT_CLUSTERS: [ZWT198TempCalibration],
>                 OUTPUT_CLUSTERS: [],
>             },
>         }
>     }
> ```

Cheers, works great!

emonfejm, 2024-11-16:
Thank you for the code.

Termostat control work fine
Child lock  work fine
Temp calibration  work fine

Battery do not work - Unknown
Pi heating demand do not work - Unknown
Setpoint change source dk not work - Unknown
Time stamp do not work - Unknown

Any tips how to fix that would be much appreciated.

mojrs10, 2024-11-27:
Great work

emonfejm, 2024-12-05:
Tem setup works occasionally.

cibermike20, 2025-01-04:
Hello, same here. Is there any (fully) working custom quirk?

Thanks!

epserra, 2025-01-05:
I have the ZWT198/ZWT100-BH thermostat configured in Zigbee2mqtt but it does not show the battery status, I think it is necessary to be able to change the battery before it runs out, can this function be included in a future revision?

emonfejm, 2025-01-05:
This is this is for zha not mqtt.

emonfejm, 2025-01-14:
that is totally rubbish for can not really on that at all. Not updating the status of cure temperature. It is restarting to default setings l, can not setup days temperature. Just waste of money.

closed 2025-01-25 by TheJulianJES

emonfejm, 2025-01-30:
It was close but could you be so kind and explain where that was implemented and how to add this quirk to HA please?

PR #3748 (2025-01-25): Add Avatto thermostats (Tuya)
## Proposed change

Adds Tuya ZWT198/ZWT100-BH Avatto wall thermostats

- _TZE204_lzriup1j
- _TZE200_viy9ihs7
- _TZE204_xnbkhhdr
- _TZE284_xnbkhhdr

Closes: #3627
Closes: #3212
Closes: #2589
Closes: #3482
Closes: #3326

## Additional information

Based on: https://github.com/Koenkk/zigbee2mqtt/discussions/25752
and https://github.com/Koenkk/zigbee-herdsman-converters/blob/40e72bc2515deb67ef35dddab04437342b7b3eb9/src/devices/tuya.ts#L7027
## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using [... cut]