zigpy/zha-device-handlers #3330: [Device Support Request] Legrand Wireless Radiant Switch

bobyangyf, 2024-08-29:
### Problem description

I'm pretty new to Home Assistant and ZHA overall, so excuse me if this is the wrong place or request.

I'm trying to pair Legrand Radiant Wireless Switch using ZHA (sonoff dongle) via home assistant
https://www.legrand.us/wiring-devices/designer-switches-and-outlets/radiant-wireless-smart-switch-with-netatmo-white/p/wnrl23wh

The closest issue I found was #2098 but that was resolved just by repairing. I've tried that multiple times but to no avail (repairing and deleting).

### Details

I'm pairing the device with ZHA set to channel 11. 
I've successfully paired the device according to ZHA, which shows up as a binary input. However, no matter on/off paddle I hit, I observe no events nor changes in state from Home Assistant ZHA Device view. I do see "checkin" events being fired every now and then, so the device is certainly connected.

### Solution description

I would like this device to function in ZHA

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 1,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4129,
    "maximum_buffer_size": 89,
    "maximum_incoming_transfer_size": 63,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 63,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0104",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0xfc01"
      ],
      "output_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Remote switch",
  "class": "zigpy.device.Device"
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

bobyangyf, 2024-11-13:
Some updates here.

I managed to get the switch to pair properly by pressing keys to keep it alive. Now there's multiple end points bound.

However, I don't see any zha_events no matter what I press on the switch via the HA event listeners. Not sure if there's something else I'm missing here.

bobyangyf, 2024-11-15:
I managed to get paired successfully after deleting, rebooting HA. 

I now get "On" and "Off" events in the logs, but these don't translate to any HA automation triggers.
See the logs:
[image]

Available triggers in HA:
[image]

Final signature
`{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 1,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4129,
    "maximum_buffer_size": 89,
    "maximum_incoming_transfer_size": 63,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 63,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0104",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0xfc01"
      ],
      "output_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Remote switch",
  "class": "zigpy.device.Device"
}`

junosuarez, 2024-12-24:
Since we're seeing the raw events, I was able to get this device working for automation triggers via low-level `event` triggers:

Example automation:
```
alias: Office Wall Switch
description: ""
triggers:
  - trigger: event
    event_type: zha_event
    event_data:
      device_id: 6de8443177456075ab1b2a74aab89be1
conditions: []
actions:
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ trigger.event.data.command == \"off\" }}"
        sequence:
          - action: light.turn_off
            metadata: {}
            data: {}
            target:
              area_id: office
      - conditions:
          - condition: template
            value_template: "{{ trigger.event.data.command == \"on\" }}"
        sequence:
          - action: light.turn_on
            metadata: {}
            data: {}
            target:
              area_id: office
mode: single
```

Note, `device_id` is the Home Assistant device id, as displayed in the URL from the device detail page.

This works as a proof-of-concept, but it is a very poor editing experience in the Automations UI, and it would still be best to support this device in ZHA.

I'm willing to try as time allows, but no promises on timeline.

PR #3970 (2025-05-24): Add device triggers for Legrand radiant wireless switch
## Proposed change

Adds on/off press and long press event support for the [Legrand radiant wireless switch](https://www.legrand.us/wiring-devices/designer-switches-and-outlets/radiant-wireless-smart-switch-with-netatmo-white/p/wnrl23wh)

[image]
[image]

## Additional information

- fixes #3330 

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-05-24 by TheJulianJES