zigpy/zha-device-handlers #3332: [Device Support Request] Futurehome Termostat

esuomi, 2024-08-29:
### Problem description

I'm re-reporting https://github.com/zigpy/zha-device-handlers/issues/2985 as the information in that request is quite muddled, and I wanted to provide it in more legible format with some additional info. That said, I also have this thermostat, two of them in fact, so there is actual need beyond just polishing someone else's text :)

---

The device is [Futurehome Termostat](https://www.futurehome.io/en_no/shop/thermostat-w) (_note the missing h, the product is Norwegian_) and shows up as Router in Home Assistant. Like reported in the other issue, all the details match for me as well, such as the device type reporting itself as Router and no entities available.

This device is already supported by Zigbee2MQTT [Futurehome TS0601_futurehome_thermostat](https://www.zigbee2mqtt.io/devices/TS0601_futurehome_thermostat.html#futurehome-ts0601-futurehome-thermostat) although there is an open issue of missing support for power measurement: https://github.com/Koenkk/zigbee2mqtt/issues/23139

### Solution description

The device should function like a thermostat :) Feature parity with Zigbee2MQTT's support would be great.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_e5hpkc6d",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>
[... 681 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

esuomi, 2024-09-10:
As update (and to avoid staleness), modifying [this ZHA quirk](https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto.py) by adding the model to MODELS_INFO of either the Avatto entry or creating a copy of the Beok class and changing the model accordingly. and all controls become available with at least at glance reasonable values.

More official support would of course be nice, but this works as a great stopgap until such can be produced.

928driver, 2024-09-29:
@esuomi , thanks for finding a solution for this :)
But I am not able to get it to work for some reason. I downloaded the ts0601_thermostat_avatto.py file, added _TZE200_4hbx5cvx which is my futurehome thermostat, under models and modifed configuration.yaml to fetch the quirk. After rebooting, logs indicate that the quirk is picked up but when re-add the thermostat, it just shows up as a router. Do you have any idea why it is not working for me?

esuomi, 2024-09-29:
@928driver For what it's worth, I didn't re-add the device, I simply loaded the quirk, rebooted and it jumped up and showed All The Things. It's survived multiple reboots and updates already (I'm the kind of person who keeps everything always updated to latest) so can't really help with that.

My modified quirk uses the lattest thing I mentioned, effectively I added this at the bottom of the quirk:
```python
class Futurehome(EnchantedDevice, TuyaThermostat):
    """Futurehome Thermostatic radiator valve."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.thermostat_onoff_bus = Bus()
        self.AvattoTempCalibration_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        #  endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 5, 61184]
        #  output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_e5hpkc6d", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    AvattoManufCluster,
                    AvattoThermostat,
                    AvattoUserInterface,
                    TuyaPowerConfigurationCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [AvattoChildLock],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.CONSUMPTION_AWARENESS_DEVICE,
                INPUT_CLUSTERS: [AvattoTempCalibration],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```
which is mostly a copy-paste from the `Beok2` class above, and it just worked, so...yay? If I had to guess, there's some specifics about device pairing which don't work correctly with this quirk but do for the generic handling, but it is beyond my knowledge to figure out what exactly could be the cause.

...of course, we could try emailing Futurehome as well...

jpakanen2, 2024-10-10:
@esuomi Do you know if OTA updates are working with ZHA using the configuration above?

esuomi, 2024-10-10:
@jpakanen2 The device firmware reports itself as Unknown so probably not.

esuomi, 2024-10-16:
Another minor update, seems Futurehome has added information about available channels and their data to their support page: https://support.futurehome.no/hc/en-no/articles/4834513064605-Channels

To quote,

[image]
> We see the thermostat entering the system with 5 channels:
>
>    Channel 0 = Physical device for configurations and the like.
>    Channel 1 = Thermostat.
>    Channel 2 = Internal room sensor.
>    Channel 3 = External room sensor.
>    Channel 4 = Floor sensor.
>
> Channel 0 should never be used for anything other than settings (parameter on Z-wave) or associations, i.e. it should not be placed in any room in the app.
>
>Channel 1 (ch_1) represents the thermostat function on the device. Channel 1 is used for the “core function” of the device. On a dimmer, the actual dimming function will be on channel 1, while on the thermostat, temperature adjustment and energy measurement are on channel 1. This is the case in almost all situations, which is why we always place channel 1 in the room.
>
> The next channels are often more device specific than channel 1, which is universal. On this thermostat, they each represent a temperature sensor. Therefore, place the sensor you want to use in the room to control it.

One thing I'd like to have added from here is actually ch_4 / floor sensor, but I have no idea how to do that. Anyway, interesting information nonetheless.

EuroEager2008, 2025-08-24:
Tested Develco HAN sensor ("Futurehome") with z2m, seems to work perfect (except for crazy values at startup)

esuomi, 2025-09-04:
@EuroEager2008 That works better than my from earlier in the thread, so thanks. Although now I suspect that HA's default thermometer card and related logic is a bit flawed, as I think it doesn't react to the floor temperature, although in device settings the floor temperature can be seen as expected...ah well, it's still better than previously, so that's great.

PapsMan-No, 2025-09-05:
@TheOneAndOnlyPP Very good that you posted this – I also eventually found that out. But as for you, it took me a looong time, so excellent to share this!
@EuroEager2008 If it would somehow be possible to access the energy data for the Futurehome Thermostats, that would be really useful.
Under the Futurehome umbrella, the energy data provided by the thermostats were an essential part of my energy management, i.e. to keep consumption per hour below certain thresholds. (I have three of these thermostats in rooms with floor heating.)