zigpy/zha-device-handlers #3337: [Device Support Request] TS0601 by _TZE204_dwcarsat Smart Air quality sensor (Zigbee)

knipknap, 2024-08-30:
### Problem description

The device is found via Zigbee, but does not provide any sensors.

[image]

A similar device with the same name is already supported according to #1313, but has a different model identification.
I don't know if it is relevant, but I found only one message in the system logs related to the sensor:

[image]

### Solution description

The device should provide the sensor values.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_dwcarsat",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Logs

<details><summary>Logs</summary>

I don't know if it is relevant, but I found only one message in the system logs related to the sensor:

[image]

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

sarask95, 2024-09-20:
I have the same problem a solution would be great

xmrxmr00, 2024-09-25:
Same here: TS0601 by _TZE204_dwcarsat

AnnoyingTechnology, 2024-10-01:
You can use the _TZE200 modified with a manufacturerid 204. 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5889

knipknap, 2024-10-01:
Sounds great, but how? I tried to add this to configuration.yaml:

```yaml
zha:
  custom_quirks_path: /config/custom_zha_quirks/
  device_config:
    aa:bb:cc:dd:ee:ff:00:11-1:    # format: {ieee}-{endpoint_id}
      manufacturer: "_TZE200_dwcarsat"
```

but it seems "manufacturer" is not allowed:

`Invalid config for 'zha' at configuration.yaml, line 40: 'manufacturer' is an invalid option for 'zha', check: zha->device_config->aa:bb:cc:dd:ee:ff:00:11-1->manufacturer, please check the docs at https://www.home-assistant.io/integrations/zha
`

Perrudi, 2024-10-08:
Same here: TS0601 by _TZE204_dwcarsat

jackoncept, 2024-10-12:
Hello Perrudo, I am in the exact same situation, if you find how to move forward I will be interested  !

knipknap, 2024-10-26:
Thanks for including this!

I have been trying to find out in which release it might be available, but cannot figure it out. I see that it is included in zha-device-handlers 0.0.123, but do not find any reference to it in the Home Assistant requirements.txt (nor any git submodules), so I am not sure how the two repositories relate.

Any hint?

Atomindul, 2024-11-03:
> Thanks for including this!
>
> I have been trying to find out in which release it might be available, but cannot figure it out. I see that it is included in zha-device-handlers 0.0.123, but do not find [...]

Hey knipknap - I'm a complete ammeter at this, but per home assistant website
"New versions of Home Assistant are released on the first Wednesday of every month. The exact dates can be seen in the upcoming events calendar on the [Home Assistant Developers](https://developers.home-assistant.io/) website."
https://www.home-assistant.io/faq/release/

I wonder can we expect to see this update be incorporated into the upcoming We 06 Nov 2024 Home Assistant update, since it's the first update since ticket #3337 was closed?

knipknap, 2024-11-07:
Jep, that's great, works here too, Thank you!

Just the formaldehyde value seems wrong:
[image]

According to
https://www.cdph.ca.gov/Programs/CCDPHP/DEODC/OHB/HESIS/CDPH%20Document%20Library/formaldehyde.pdf

> "The eyes, nose, and throat are irritated by formaldehyde vapors at levels as low as about 0.3 part formaldehyde per million parts of air (0.3 part per million, or 0.3 “ppm""
> 
> "The Permissible Exposure Limit for formaldehyde is 0.75 part of formaldehyde per million parts of air (0.75 part per million, or 0.75 ppm)."

So I guess if the shown value is correct I would be unable to write this comment.

Atomindul, 2024-11-07:
I would treat all but the temp and humidity readings with skepticism tbh.
https://www.youtube.com/watch?v=dxVUxYIrawU

Atomindul, 2024-11-07:
My formaldehyde readings are similar to yours, knip.

[image]

Perrudi, 2024-11-08:
in comparison to my readings here:
[image]

Perrudi, 2024-11-09:
I noticed, that today my values of "Formaldehyd" and "CO2" and the values are swapped. Should this be investigated?
[image]

juha-pmg, 2024-11-29:
Just received a similar device. All of my values are off, but that was more than expected from these cheap sensors. Formaldehyde reading is ~400 ppm that sounds like co2 value to me. Co2 reading feels to be pm2.5 reading measured in ug/m3 Voc is plausible as is, PM2.5 value could be Formaldehyde measured in ppbillion. 

The temp and humidity sensor aht30 is only one that has markings and it should be fairly accurate. There is one gas sensor without markings, it's probably VOC sensor of some sort, I am not aware of any multigas sensors on that package type, though that doesn't tell anything as I don't really work with these. One PM sensor that can be sourced from aliexpress for less than $5. These pm sensors should be fairly on close on general use. 

I feel that co2 and formaldehyde values might be calculated (or faked if you want to be blunt).  Don't take my word for it tough. My values for reference
[image]

snippetsBySam, 2024-12-10:
I've got the exact same device but only the VOCs sensor is showing. Anyone have ideas why?

When I download diagnostics, I can see some of the other sensor attributes in the json file but they aren't showing

[image]

```
"0x040d": {
            "endpoint_attribute": "carbon_dioxide_concentration",
            "attributes": {
              "0xfffd": {
                "attribute": "ZCLAttributeDef(id=0xFFFD, name='cluster_revision', type=<class 'zigpy.types.basic.uint16_t'>, zcl_type=<DataTypeId.uint16: 33>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0002": {
                "attribute": "ZCLAttributeDef(id=0x0002, name='max_measured_value', type=<class 'zigpy.types.basic.Single'>, zcl_type=<DataTypeId.single: 57>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0000": {
                "attribute": "ZCLAttributeDef(id=0x0000, name='measured_value', type=<class 'zigpy.types.basic.Single'>, zcl_type=<DataTypeId.single: 57>, access=<ZCLAttributeAccess.Read|Report: 9>, mandatory=True, is_manufacturer_specific=False)",
                "value": 4e-06
              },
              "0x0001": {
                "attribute": "ZCLAttributeDef(id=0x0001, name='min_measured_value', type=<class 'zigpy.types.basic.Single'>, zcl_type=<DataTypeId.single: 57>, access=<ZCLAttributeAccess.Read: 1>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0xfffe": {
                "attribute": "ZCLAttributeDef(id=0xFFFE, name='reporting_status', type=<enum 'AttributeReportingStatus'>, zcl_type=<DataTypeId.enum8: 48>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0003": {
                "attribute": "ZCLAttributeDef(id=0x0003, name='tolerance', type=<class 'zigpy.types.basic.Single'>, zcl_type=<DataTypeId.single: 57>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              }
            },
            "unsupported_attributes": []
          }```

JesterIsDead, 2024-12-24:
Same problem, formaldehyde and C02 appeared to be swapped. I'd be dead if I were breathing that much formaldehyde this long.

RiX012, 2025-01-14:
This seems to be further discussed in: https://github.com/home-assistant/core/issues/130270

TheJulianJES, 2025-01-26:
Reopening as this is still an issue. In the future, please just open a new issue and don't discuss in closed ones.

closed 2025-01-28 by TheJulianJES