zigpy/zha-device-handlers #3338: [Device Support Request] _TZ3218_ya5d6wth switch with 4 channels and temperature sensor

kskalski, 2024-08-31:
### Problem description

I have a device (model the same as https://www.zigbee2mqtt.io/devices/TYZGTH4CH-D1RF.html) that connects over Zigbee and adds 4 control switches, but the temperature sensor attached to the device is missing from ZHA.

### Solution description

Temperature sensor entity should be included.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
[... 602 lines cut ...]
        }
      }
    }
  }
}
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>
I don't know what quirks are and how to use them, but this thread https://community.home-assistant.io/t/zigbee-temperature-sensor-switch-only-detected-as-light/748950/9 mentions the code:

```python
dp_to_attribute: dict[int, DPToAttributeMapping] = {
    102: DPToAttributeMapping(
        TuyaTemperatureMeasurement.ep_attribute,
        "measured_value",
        converter=lambda x: x * 10,  # decidegree to centidegree
    ),
}
```

</details>

Ulukai0, 2024-10-05:
Hi, 
I have the same problem, and it's my first try with custom quirks, so I don't know if it's a config problem or the quirk file problem...
I copied & pasted your code, deleted, reboot HA and re-add the device but doesn't work, do I have to modify a part of this code?

kskalski, 2024-10-06:
I use the exact same code, placed into 
`/home/homeassistant/.homeassistant/custom_zha_quirks/TZ3218_ya5d6wth.py`

You can check that home assistant log has this line when loading:
`WARNING (ImportExecutor_0) [zhaquirks] Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers`

and on Zigbee Home Automation integration you should be able to pair it as a new device

Ulukai0, 2024-10-06:
Thanks for the answer, I forgot the ".py" in file name.....
Now the Quirk is working !!! 
I changed the device type to DeviceType.ON_OFF_SWITCH to control my floor heating valve.
Winter is coming !!! and I'm ready thanks to you !