zigpy/zha-device-handlers #3339: [Device Support Request] Legrand contactor PN: 412171 - 412191 - 199122

aauzi, 2024-08-31:
### Problem description

The device is natively detected as a zha.DeviceType.ON_OFF_PLUG_IN_UNIT by ZHA but the automatically provided switch does not control the OnOf cluster it exposes.

On, Off and Toggle OnOff cluster commands result in UNSUPPORTED_CLUSTER_COMMAND failures.

### Solution description

The device implements two manufacturer specific clusters:
- the cluster 0xfc01 implement a mode selector attribute that allow to select Automatic mode or Switch mode.
- the cluster 0xfc41 implementent de Automatic mode control of a three states switch Off/Auto/On.

When the device is in Switch mode, it operates normally the OnOff cluster.

However, when the device is in Auto mode, on, off and toggle OnOff custer's
commands are not supported.

The factory defaut is Auto mode. This explains the falty behavior. 

The joined custom quirk implements redirection of OnOff commands and on_off attribute to/from the AutoOnOff cluster (id: 0xfc41).

Ideally a complete support would allow to configure the mode and would expose 3 states control  and 4 states display: Off (manual), Off (auto), On (auto) and On (manual)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

(https://assets.legrand.com/pim/NP-FT-GT/F03037EN-05%20(Connected%20contactor).pdf)

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 1,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4129,
    "maximum_buffer_size": 89,
    "maximum_incoming_transfer_size": 63,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 63,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000f",
        "0x0b04",
        "0xfc01",
        "0xfc41"
      ],
      "output_clusters": [
        "0x0000",
        "0x0005",
[... 1473 lines cut ...]
                    Ota.cluster_id,
                    LegrandContactorMode,
                ],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.COMBO_BASIC,
                INPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    def __init__(self, *args, **kwargs):
        """Init."""
        self.mode_bus = Bus()
        self.auto_status_bus = Bus()
        super().__init__(*args, **kwargs)
```

</details>

### Additional information

Please refer to custom quirk inline documentation.

aauzi, 2024-09-07:
Hi,

after more digging in the resources available, mostly around Quirks V2.
I came up with a refined custom quirk.

### Custom quirk

<details><summary>Custom quirk</summary>

```python
"""Module for Legrand Contactor, Drivia with Netatmo 20AX - 230V~ - 50Hz

PN: 412171 - 412191 - 199122

Documentation (en): https://assets.legrand.com/pim/NP-FT-GT/F03037EN-05%20(Connected%20contactor).pdf
Documentation (fr): https://assets.legrand.com/pim/NP-FT-GT/F03037FR-05%20(Contacteur%20Connect%C3%A9).pdf

Tested with PN: 199122
FW version: 0x005a45ff

signature:
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 1,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4129,
    "maximum_buffer_size": 89,
    "maximum_incoming_transfer_size": 63,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 63,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000f",
        "0x0b04",
        "0xfc01",
        "0xfc41"
      ],
      "output_clusters": [
        "0x0000",
        "0x0005",
        "0x0006",
        "0x0019",
        "0xfc01"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0066",
      "input_clusters": [
        "0x0021"
      ],
      "output_clusters": [
        "0x0021"
      ]
    }
[... 643 lines cut ...]
            use_cache: false
            allow_create: false

- alias: legrand contactor force off
  description: ''
  trigger:
    - platform: sun
      event: sunset
      offset: "-01:00:00"
  condition: []
  action:
    - action: switch.turn_off
      entity_id: switch.legrand_contactor_commutateur

- alias: legrand contactor reset auto
  description: ''
  trigger:
    - platform: sun
      event: sunrise
      offset: "+02:00:00"
  condition: []
  action:
    - action: button.press
      entity_id: button.legrand_contactor_reset_auto
```

mycanaletto, 2024-10-01:
@aauzi 

Many thanks for your work, which makes life easier for a lot of people, starting with me!

I haven't really grasped the significance of automation and I'm not sure I've got it all right. I'm always interested in updates!

Merci !

[image]

MiGaNuTs, 2024-10-02:
Thanks ! Good job !

jeverley, 2024-11-07:
Apologies ignore those two references, they were created in error.

Wanick95, 2024-11-07:
Thanks good job,
Works in 2024.10 but stop working in 2024.11
`Unexpected exception importing custom quirk 'legrand'

Traceback (most recent call last):

  File "/usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py", line 479, in setup

    spec.loader.exec_module(module)

  File "<frozen importlib._bootstrap_external>", line 995, in exec_module

  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed

  File "/config/custom_quirks/legrand.py", line 751, in <module>

    .enum("mode", LegrandMode, LegrandContactorMode.cluster_id,

     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 666, in enum

    ZCLEnumMetadata(

  File "<attrs generated init zigpy.quirks.v2.ZCLEnumMetadata>", line 16, in __init__

    __attr_validator_fallback_name(self, __attr_fallback_name, self.fallback_name)

  File "/usr/local/lib/python3.12/site-packages/attr/validators.py", line 101, in __call__

    raise TypeError(

TypeError: ("'fallback_name' must be <class 'str'> (got None that is a <class 'NoneType'>).", Attribute(name='fallback_name', default=NOTHING, validator=<instance_of validator for type <class 'str'>>, repr=True, eq=True, eq_key=None, order=True, order_key=None, hash=None, init=True, metadata=mappingproxy({}), type='str', converter=None, kw_only=True, inherited=True, on_setattr=None, alias='fallback_name'), <class 'str'>, None)`

aauzi, 2024-11-08:
@Wanick95 thanks for the early notice, I was a bit late with the upgrade to 2024.11.

The custom_quirk needed to take into account some nice evolutions of quirks V2.

For the people who use the previous version and would like to check the changes, here is what changed in diff format.

<details><summary>Custom quirk diff</summary>

```diff
diff --git a/contactor.py b/contactor.py
index 923eec3..2585ed9 100644
--- a/contactor.py
+++ b/contactor.py
@@ -68,7 +68,7 @@ signature:
 import logging
 from enum import Enum
 from zigpy.profiles import zgp, zha
-from zigpy.quirks.v2 import QuirkBuilder, CustomDeviceV2, EntityType, EntityPlatform
+from zigpy.quirks.v2 import QuirkBuilder, CustomDeviceV2, EntityType, EntityPlatform, ReportingConfig
 from zigpy.quirks import CustomCluster
 import zigpy.types as t
 from zigpy.zcl import ClusterType
@@ -638,18 +638,52 @@ acts as a proxy to LegrandContactorAutoOnOff in Auto mode.
         self.reporting_bus = Bus()
         super().__init__(application, ieee, nwk, replaces, quirk_metadata)
 
+
+REPORTING_WHEN_CHANGED = ReportingConfig(
+    min_interval = 0,
+    max_interval = 0,
+    reportable_change = 1)
+
 (
-   QuirkBuilder(f" {LEGRAND}", " Contactor")
-   .device_class(LegrandContactorV2)
-   .replaces(LegrandContactorMode)
-   .replaces(LegrandContactorAutoOnOff)
-   .replaces(LegrandContactorSwitchOnOff)
-   .enum("mode", LegrandMode, LegrandContactorMode.cluster_id,
-         ClusterType.Server, 1, EntityType.CONFIG, EntityPlatform.SELECT, False, True, 'operating_mode')
-   .command_button("override", LegrandContactorAutoOnOff.cluster_id, (AutoOverride.Automatic, ), None,  
-                   ClusterType.Server, 1, EntityType.STANDARD, False, 'reset_auto')
-   .enum("status", LegrandContactorAutoStatus, LegrandContactorAutoOnOff.cluster_id,
-         ClusterType.Server, 1, EntityType.DIAGNOSTIC, EntityPlatform.SENSOR, False, True, 'auto_status')
+    QuirkBuilder(f" {LEGRAND}", " Contactor")
+    .device_class(LegrandContactorV2)
+    .replaces(LegrandContactorMode)
+    .replaces(LegrandContactorAutoOnOff)
+    .replaces(LegrandContactorSwitchOnOff)
+    .enum(attribute_name = "mode",
+          enum_class = LegrandMode,
+          cluster_id = LegrandContactorMode.cluster_id,
+          cluster_type = ClusterType.Server,
+          endpoint_id = 1,
+          entity_platform = EntityPlatform.SELECT,
+          entity_type = EntityType.CONFIG,
+          initially_disabled = False,
+          attribute_initialized_from_cache = True,
+          reporting_config = REPORTING_WHEN_CHANGED,
+          translation_key='operating_mode',
+          fallback_name='Mode')
+    .command_button(command_name = "override",
+                    cluster_id = LegrandContactorAutoOnOff.cluster_id,
+                    command_args = (AutoOverride.Automatic, ),
+                    command_kwargs = None,  
+                    cluster_type = ClusterType.Server,
+                    endpoint_id = 1,
+                    entity_type = EntityType.STANDARD,
+                    initially_disabled = False,
+                    translation_key='reset_auto',
[... 693 lines cut ...]
          entity_type = EntityType.DIAGNOSTIC,
          initially_disabled = False,
          attribute_initialized_from_cache = True,
          reporting_config = REPORTING_WHEN_CHANGED,
          translation_key='auto_status',
          fallback_name='Auto Status')
   #.switch("led_dark", LegrandContactorMode.cluster_id,
   #        ClusterType.Server, 1, False, None, 0, 1, EntityPlatform.SWITCH, False, True, 'led_dark')
   #.switch("led_on", LegrandContactorMode.cluster_id,
   #        ClusterType.Server, 1, False, None, 0, 1, EntityPlatform.SWITCH, False, True, 'led_on')
   .add_to_registry()
)
```
</details>

What's nice with this evolution is the enforcement of definition of not only the _fallback_name_ parameter but also the  _reporting_config_ one.

_fallback_name_ at some point will be used to give a sensible name to the entities. +1
And  _reporting_config_ defines automatic reporting to the attribute bound to the entity.

The **ReportingConfig** is particularly interesting because it obsoletes the poll automation I was previously using.

I'm not really sure if the definition of **REPORTING_WHEN_CHANGED** I've done is 100% adequate but it's awesome to have this ability.

Thanks again.

Wanick95, 2024-11-08:
Thank you @aauzi , works like a charm
I was reading the documentation to try to solved it by my self, but you were so fast.

ricobal, 2024-12-09:
Thank you @aauzi for this quirck.
I have just installed it on HA version 2024.11.3 and I have a strange behaviour for the switch. When switched on, switch status changes to "on" and after a few minutes, it changes back to off in HA whereas the Legrand switch stays on.
Would you have any idea of where this could come from ? 
Many thanks.
Eric.

[image]

aauzi, 2024-12-31:
@ricobal thanks for letting me know your issue and my apologies for the late answer.

My guess is that an attribute change report form the OnOff cluster is handled while in Auto mode.

The result is a change of the HA Switch state whereas the actual device does not change (due to the fact it is in Auto mode)

To be honest I haven't exactly reproduced the problem you describe but your observation made me realize such reports weren't properly ignored in Auto mode.

So I've changed the following.

For the people who use the previous version and would like to check the changes, here is what changed in diff format.

<details><summary>Custom quirk diff</summary>

```diff
diff --git a/contactor.py b/contactor.py
index eb05858..d5318f2 100644
--- a/contactor.py
+++ b/contactor.py
@@ -502,6 +502,8 @@ NOTE: The name and ep_attribute class attributes are NOT changed to benefit
 
         super()._update_attribute(self.ON_OFF_ID, value)
 
+        _LOGGER.debug(f'LegrandContactorSwitchOnOff.on_off_reported done.')
+
     async def _read_attributes(self, attr_ids, *args, **kwargs):
         """Read attributes.
 
@@ -581,6 +583,20 @@ FIXME: something should execute the LegrandContactorAutoOnOff.override(AutoOverr
 
         return await super().command(command_id, *args, manufacturer=manufacturer, expect_reply=expect_reply, tsn=tsn)
 
+    def _update_attribute(self, attrid, value):
+        """Legrand switch OnOff attribute update.
+
+Ignores on_off  attribute update when the device is Auto mode."""
+
+        _LOGGER.debug(f'LegrandContactorSwitchOnOff._update_attribute: attrid={attrid}, value={value}')
+
+        # await self._update_contactor_mode()
+
+        if not self._contactor_is_switch:
+            _LOGGER.debug('LegrandContactorSwitchOnOff._update_attribute ignored in auto mode.')
+            return
+
+        super()._update_attribute(attrid, value)
 
 class LegrandContactorV2(CustomDeviceV2):
     """Legrand Contactor device.
```
</details>

And of course, the full modified version tested with 2024.12.

<details><summary>Custom quirk</summary>

```python
"""Module for Legrand Contactor, Drivia with Netatmo 20AX - 230V~ - 50Hz

PN: 412171 - 412191 - 199122

Documentation (en): https://assets.legrand.com/pim/NP-FT-GT/F03037EN-05%20(Connected%20contactor).pdf
Documentation (fr): https://assets.legrand.com/pim/NP-FT-GT/F03037FR-05%20(Contacteur%20Connect%C3%A9).pdf

Tested with PN: 199122
FW version: 0x005a45ff

signature:
{
  "node_descriptor": {
[... 762 lines cut ...]
2024-12-31 10:49:29.511 DEBUG (MainThread) [contactor] LegrandContactorAutoOnOff._update_attribute: attrid=0, value=<AutoStatus.ForcedOn: 1>
2024-12-31 10:49:29.511 DEBUG (MainThread) [contactor] LegrandContactorAutoOnOff._update_attribute: attrid=1, value=<Bool.true: 1>
2024-12-31 10:49:29.511 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff.on_off_reported: value=<Bool.true: 1>
2024-12-31 10:49:29.512 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff.on_off_reported done.
2024-12-31 10:49:29.512 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff._update_attribute: attrid=0, value=<Bool.true: 1>
2024-12-31 10:49:29.512 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff._update_attribute ignored in auto mode.
2024-12-31 10:49:31.714 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff.command
2024-12-31 10:49:31.714 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff._update_contactor_mode.
2024-12-31 10:49:31.802 DEBUG (MainThread) [contactor] LegrandContactorMode._update_attribute: attrid=0, value=[4, 0]
2024-12-31 10:49:31.802 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff.contactor_is_switch_reported: value=False
2024-12-31 10:49:31.802 DEBUG (MainThread) [contactor] LegrandContactorAutoOnOff.turn_off
2024-12-31 10:49:31.897 DEBUG (MainThread) [contactor] LegrandContactorAutoOnOff._update_attribute: attrid=0, value=<AutoStatus.ForcedOn: 1>
2024-12-31 10:49:31.898 DEBUG (MainThread) [contactor] LegrandContactorAutoOnOff.command: id=0
2024-12-31 10:49:32.089 DEBUG (MainThread) [contactor] LegrandContactorAutoOnOff._update_attribute: attrid=0, value=<AutoStatus.ForcedOff: 0>
2024-12-31 10:49:32.089 DEBUG (MainThread) [contactor] LegrandContactorAutoOnOff._update_attribute: attrid=1, value=<Bool.false: 0>
2024-12-31 10:49:32.090 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff.on_off_reported: value=<Bool.false: 0>
2024-12-31 10:49:32.090 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff.on_off_reported done.
2024-12-31 10:49:32.090 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff._update_attribute: attrid=0, value=<Bool.false: 0>
2024-12-31 10:49:32.090 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff._update_attribute ignored in auto mode.
2024-12-31 10:52:15.788 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff._update_attribute: attrid=0, value=<Bool.false: 0>
2024-12-31 10:52:15.788 DEBUG (MainThread) [contactor] LegrandContactorSwitchOnOff._update_attribute ignored in auto mode.
```
</details>

Thanks again.

ricobal, 2024-12-31:
@aauzi thank you very much for your answer. I will test the new custom quirk version and let you know if the issue is solved on my configuration. Thanks.

ricobal, 2025-01-02:
@aauzi. Just to confirm that problem is solved with the new custom quirk. Thanks a lot.

aauzi, 2025-03-09:
Please be advised that I have now opened the PR #3941 
The quality checks and test coverage helped me to fine-tune details in the quirk so one may consider using the source of the PR.
Questions and observations are still warmly welcome in this thread.
But in order to move on, review comments in the PR are also kindly requested.

closed 2025-09-07 by aauzi