zigpy/zha-device-handlers #3354: [Device Support Request]

lpt2007, 2024-09-08:
### Problem description

First problem:
Device shows as light.
Second problem:
If I switch any of four relays to on, automatically goes almost immediately to off state

### Solution description

Device shows as switch and relay stay on when I turn it on.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

https://github.com/user-attachments/assets/604711a0-4176-4962-ae4b-c26bfe4210f0

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_imaccztn",
  "model": "TS0004",
  "class": "zhaquirks.tuya.ts000x.Switch_4G_GPP"
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

seanw18, 2024-09-09:
Any chance this is a  "mode" set on the relay? I.e. inching mode or momentary? 
I have the 4 channel zigbee version of this https://www.aliexpress.com/item/1005006190483000.html working okay.
TS0004 by _TZ3000_imaccztn   using default Quirk: zhaquirks.tuya.ts000x.Switch_4G_GPP

lpt2007, 2024-09-09:
No I don't haveoption to set "mode".

Here is my settings:
[image]

@seanw18 Are your settings like mine?

seanw18, 2024-09-09:
Yes, except, I've renamed the relay entities.
I can't change the "power on state". 
Just wanted to make you to hadn't connected the device to a tuya zigbee hub, it seems like the tuya app allows you to change the "mode".

lpt2007, 2024-09-09:
I don't have tuya zigbee hub. And even if I head it i don't know what happens when I try to reconnect it to zha back. I must do factory reset to put it in pairing mode.

seanw18, 2024-09-09:
Okay, sorry for the red herring.

lpt2007, 2024-09-10:
I found what is wrong:) 

There is physical switch on device to switch between "on/off" or "itching" or "one relay on and other off".

closed 2024-09-10 by lpt2007