zigpy/zha-device-handlers #3356: [Device Support Request] Tuya ultrasonic level sensor (TS0601 by _TZE284_kyyu8rbj)

hickey, 2024-09-10:
### Problem description

Recently bought the Tuya ultrasonic liquid level sensor (https://www.aliexpress.us/item/3256807266442468.html?spm=a2g0o.order_list.order_list_main.11.21ef1802lD5999&gatewayAdapt=glo2usa) and while it will pair with ZHA, it only reports LQI and RSSI. 

I am very new (i.e. just started reading about ZHA quirks) and would like to assist/work on this quirk. Mostly I will probably need some guidance and basic information for building the Python script for the quirk. Part of the reason I would like to work on this quirk is to better understand the Zigbee protocols, add to the volunteers contributing quirks and write a quirk for the Tuya water quality sensor I also got (this has many more values to read and will be much more interesting project). 

I am making the assumption from the diagnostic output below that the `reporting_status` is the attribute that contains the level value. So it seems that this should be a pretty straight forward quirk to implement. 

### Solution description

I am envisioning a custom quirk that reports back a single value to ZHA which is the last level measured. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_kyyu8rbj",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

[... 689 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

brooksaw, 2024-10-05:
@hickey - Any luck with this? Just got one and am trying to get it going and have made some progress but wouldn't mind some input

brooksaw, 2024-10-05:
{
    "alarm": "normal",
    "alert": "triggered",
    "linkquality": 75,
    "status": "off",
    "water_level": -1124073472
}

This is what I have with state now

brooksaw, 2024-10-07:
{
    "battery_percentage": 0,
    "installation_height": null,
    "linkquality": 114,
    "liquid_level_percent": 0,
    "max_set": null,
    "mini_set": null,
    "relay_switch": true,
    "status": "lower_alarm",
    "water_level": 0
}

with support from supplier I am almost there

Leon3310, 2024-10-11:
I bought the same Zigbee peripheral. I'm very interested in the developments.

syc0n, 2024-11-15:
Can you share your quirk?

syc0n, 2024-11-30:
`TUYA_TANK_LEVEL_PERCENT_DP = 22
TUYA_TANK_DEPTH_DP = 2
TUYA_TANK_STATE_DP = 1
TUYA_INSTALLATION_HEIGHT_DP = 19
TUYA_MAX_SET_DP = 7
TUYA_MIN_SET_DP = 8
TUYA_DEPTH_MAX_DP = 21

import zigpy.types as t
from zhaquirks.tuya.builder import TuyaPowerConfigurationCluster2AAA, TuyaQuirkBuilder
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant import UnitOfLength, UnitOfTime
from zigpy.quirks.v2 import EntityPlatform, EntityType

(
    TuyaQuirkBuilder("_TZE284_kyyu8rbj", "TS0601")
    .applies_to("_TZE284_kyyu8rbj", "TS0601")
    #.tuya_temperature(dp_id=TUYA_TANK_DEPTH_DP)
    .tuya_sensor(
        dp_id=TUYA_TANK_DEPTH_DP,
        attribute_name="TUYA_TANK_DEPTH_DP",
        type=t.int16s,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        #unit=UnitOfLength.METERS,
        #unit=UnitOfTime.SECONDS,
        entity_type=EntityType.STANDARD,
        translation_key="TUYA_TANK_DEPTH_DP",
        fallback_name="TUYA_TANK_DEPTH_DP",
    )
    .tuya_sensor(
        dp_id=TUYA_TANK_DEPTH_DP,
        attribute_name="TUYA_TANK_DEPTH_DP",
        type=t.int16s,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        #unit=UnitOfLength.METERS,
        #unit=UnitOfTime.SECONDS,
        entity_type=EntityType.STANDARD,
        translation_key="TUYA_TANK_DEPTH_DP",
        fallback_name="TUYA_TANK_DEPTH_DP",
    )
    .tuya_sensor(
        dp_id=TUYA_INSTALLATION_HEIGHT_DP,
        attribute_name="TUYA_INSTALLATION_HEIGHT_DP",
        type=t.int16s,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        #unit=UnitOfLength.METERS,
        #unit=UnitOfTime.SECONDS,
        entity_type=EntityType.STANDARD,
        translation_key="TUYA_INSTALLATION_HEIGHT_DP",
        fallback_name="TUYA_INSTALLATION_HEIGHT_DP",
    )
    .tuya_sensor(
        dp_id=TUYA_DEPTH_MAX_DP,
        attribute_name="TUYA_DEPTH_MAX_DP",
        type=t.int16s,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        #unit=UnitOfLength.METERS,
        #unit=UnitOfTime.SECONDS,
        entity_type=EntityType.STANDARD,
        translation_key="TUYA_DEPTH_MAX_DP",
        fallback_name="TUYA_DEPTH_MAX_DP",
    )
    .tuya_sensor(
        dp_id=TUYA_MAX_SET_DP,
        attribute_name="TUYA_MAX_SET_DP",
        type=t.int16s,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        #unit=UnitOfLength.METERS,
        #unit=UnitOfTime.SECONDS,
        entity_type=EntityType.STANDARD,
        translation_key="TUYA_MAX_SET_DP",
        fallback_name="TUYA_MAX_SET_DP",
    )
    .tuya_sensor(
        dp_id=TUYA_MIN_SET_DP,
        attribute_name="TUYA_MIN_SET_DP",
        type=t.int16s,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        #unit=UnitOfLength.METERS,
        #unit=UnitOfTime.SECONDS,
        entity_type=EntityType.STANDARD,
        translation_key="TUYA_MIN_SET_DP",
        fallback_name="TUYA_MIN_SET_DP",
    )

    .skip_configuration()
    .add_to_registry()
)`

its working, and fine for me currenlty.

I had issues with UnitOfLength, but that should be fixed

prairiesnpr, 2024-11-30:
Building off of @syc0n, this might expose a bit more functionality, untested, don't have the device.

```python
import zigpy.types as t
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant import PERCENTAGE, UnitOfLength
from zigpy.quirks.v2 import EntityType

class TuyaLiquidState(t.enum8):
    """Tuya Liquid State Enum."""

    Normal = 0x00
    Low = 0x01
    High = 0x02

(
    TuyaQuirkBuilder("_TZE284_kyyu8rbj", "TS0601")
    .tuya_enum(
        dp_id=1,
        attribute_name="liquid_state",
        enum_class=TuyaLiquidState,
        translation_key="liquid_state",
        fallback_name="Liquid state",
    )
    .tuya_sensor(
        dp_id=2,
        attribute_name="liquid_depth",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        entity_type=EntityType.STANDARD,
        converter=lambda x: x / 1000,
        translation_key="liquid_depth",
        fallback_name="Liquid depth",
    )
    .tuya_sensor(
        dp_id=22,
        attribute_name="liquid_level_percent",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        unit=PERCENTAGE,
        entity_type=EntityType.STANDARD,
        converter=lambda x: x / 1000,
        translation_key="liquid_level_percent",
        fallback_name="Liquid level ratio",
    )
    .tuya_number(
        dp_id=7,
        attribute_name="max_set",
        type=t.uint16_t,
        unit=PERCENTAGE,
        min_value=0,
        max_value=100,
        step=1,
        translation_key="max_set",
        fallback_name="Liquid max percentage",
    )
    .tuya_number(
        dp_id=8,
        attribute_name="mini_set",
        type=t.uint16_t,
        unit=PERCENTAGE,
        min_value=0,
        max_value=100,
        step=1,
        translation_key="mini_set",
        fallback_name="Liquid minimal percentage",
    )
    .tuya_number(
        dp_id=19,
        attribute_name="installation_height",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.MILLIMETERS,
        min_value=100,
        max_value=4000,
        step=1,
        translation_key="installation_height",
        fallback_name="Height from sensor to tank bottom",
    )
    .tuya_number(
        dp_id=21,
        attribute_name="liquid_depth_max",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.MILLIMETERS,
        min_value=100,
        max_value=2000,
        step=1,
        translation_key="liquid_depth_max",
        fallback_name="Height from sensor to liquid level",
    )
    .skip_configuration()
    .add_to_registry()
)
```

gmerciel, 2024-12-09:
Based on previous comments this is what I've got so far. Basically I moved the state to Diagnosis, removed PERCENTAGE units since values doesn't show if this is specified and corrected the distance units to match the actual values.

```python
from zigpy.quirks.v2 import EntityType
from zigpy.quirks.v2.homeassistant import PERCENTAGE, UnitOfLength
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t

from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaLiquidState(t.enum8):
    """Tuya Liquid State Enum."""

    Normal = 0x00
    Low = 0x01
    High = 0x02

(
    TuyaQuirkBuilder("_TZE284_kyyu8rbj", "TS0601")
    .tuya_enum(
        dp_id=1,
        attribute_name="liquid_state",
        enum_class=TuyaLiquidState,
        entity_type=EntityType.DIAGNOSTIC,
        translation_key="liquid_state",
        fallback_name="Liquid state",
    )
    .tuya_sensor(
        dp_id=2,
        attribute_name="liquid_depth",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        entity_type=EntityType.STANDARD,
        converter=lambda x: x / 100,
        translation_key="liquid_depth",
        fallback_name="Liquid depth",
    )
    .tuya_sensor(
        dp_id=22,
        attribute_name="liquid_level_percent",
        type=t.uint16_t,
        state_class=SensorStateClass.MEASUREMENT,
        entity_type=EntityType.STANDARD,
        translation_key="liquid_level_percent",
        fallback_name="Liquid level ratio",
    )
    .tuya_number(
        dp_id=7,
        attribute_name="max_set",
        type=t.uint16_t,
        min_value=0,
        max_value=100,
        step=1,
        translation_key="max_set",
        fallback_name="Liquid max percentage",
    )
    .tuya_number(
        dp_id=8,
        attribute_name="mini_set",
        type=t.uint16_t,
        min_value=0,
        max_value=100,
        step=1,
        translation_key="mini_set",
        fallback_name="Liquid minimal percentage",
    )
    .tuya_number(
        dp_id=19,
        attribute_name="installation_height",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.CENTIMETERS,
        min_value=10,
        max_value=400,
        step=1,
        translation_key="installation_height",
        fallback_name="Height from sensor to tank bottom",
    )
    .tuya_number(
        dp_id=21,
        attribute_name="liquid_depth_max",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.CENTIMETERS,
        min_value=10,
        max_value=400,
        step=1,
        translation_key="liquid_depth_max",
        fallback_name="Height from sensor to liquid level",
    )
    .skip_configuration()
    .add_to_registry()
)
```

prairiesnpr, 2024-12-10:
> Based on previous comments this is what I've got so far. Basically I moved the state to Diagnosis, removed PERCENTAGE units since values doesn't show if this is specified and corrected the distance units to match the actual values.
> 

Is this quirk working, if so will you be submitting a PR?

gmerciel, 2024-12-10:
It is working, altough, I'm making more tests to figure out why is the PERCENTAGE unit not working. Once I figure this out I can submit a PR.

prairiesnpr, 2024-12-12:
The percent issue is in ZHA, https://github.com/zigpy/zha/issues/327

gmerciel, 2024-12-14:
I'm having trouble with the `liquid_level_percent` attribute, it is constantly reporting 100.0, even when `liquid_depth` is showing the tank at 0.5m (out of 1.8m). Changing the value of `liquid_depth_max` doesn't seem to take any effect (I would assume it is used to calculate the fill percentage). It seems like this configuration doesn't exist on the sensor, if you look at the configuration options in the Tuya App, it shows only _Set Max_, _Set Min_ and _Installation height_.

prairiesnpr, 2024-12-14:
> I'm having trouble with the `liquid_level_percent` attribute, it is constantly reporting 100.0, even when `liquid_depth` is showing the tank at 0.5m (out of 1.8m). 

If you could enable debug logging and grab the logs when `liquid_level_percent` updates, it would be interesting to see what value and type the device is actually reporting.

closed 2025-01-20 by TheJulianJES

ausfas, 2025-04-23:
This definitely needs improvement. 
https://github.com/zigpy/zha-device-handlers/issues/4022

kapabel, 2025-06-04:
I have the same issue as @gmerciel 
My liquid_level_percent shows constantly 100% even when it clearly isn't the case. At one stage for some reasons it output a reading of 188% for the Liquid level ratio.

PR #3744 (2025-01-20): Add Tuya ultrasonic level sensors
## Proposed change

Adds _TZE284_kyyu8rbj and _TZE200_lvkk0hdg

Closes: #3474 
Closes: #3397
Closes: #3356

## Additional information

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works