zigpy/zha-device-handlers #3365: [BUG] Lixee Zlinky_TIC v14

Nbird67, 2024-09-16:
### Bug description

Hello, since firmware upgrade from v13 to v14 and a "quirk" in Home Assistant (ZHA integration) I reappaired (and try with reset too) my Linxee and since my counters are all false.
Instead of retrieving kWh they retrieve I think Wh, but display kWh.
Example: 1 605 825,0 kWh instead of 1605 kWh
Linky in historical mode.
This made my energy panel all wrong...

Thank you for your help

### Steps to reproduce

Go to the Lixee zigby device (manage device) check the value of the counter metering => current_tier1_summ_delivered (id: 0x0100) value for exemple 1610442 (Wh) but in the sensor panel the values are displayed as kWh and not divide by 1000

### Expected behavior

Value in kWh ou Wh, but note Wh value displayed as kWh.

I have created a sensor template to divide values by 1000

### Screenshots/Video

[image]
[image]

### Device signature

  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4151,
    "maximum_buffer_size": 127,
    "maximum_incoming_transfer_size": 200,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 200,
    "descriptor_capability_field": 0
  },
  "endpoints": {    
      "1": {
      "profile_id": "0x0104",
      "device_type": "0x0053",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0702",
        "0x0b01",
        "0x0b04",
        "0xef00",
        "0xff66"
      ],
      "output_clusters": [
        "0x0019",
        "0xef00"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [
        "0x0021"
      ],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LiXee",
  "model": "ZLinky_TIC",
  "class": "zigpy.device.Device"
}

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

blauret, 2024-10-21:
I believe the core issue is Zlinky_TIC signature does not match the quirks anymore.

```
2024-10-21 11:16:08.645 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LiXee ZLinky_TIC (00:15:8d:00:05:be:83:47)
2024-10-21 11:16:08.645 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lixee.zlinky.ZLinkyTICFWV12'>
2024-10-21 11:16:08.646 DEBUG (MainThread) [zigpy.quirks] Fail because input cluster mismatch on at least one endpoint
2024-10-21 11:16:08.646 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lixee.zlinky.ZLinkyTIC'>
2024-10-21 11:16:08.646 DEBUG (MainThread) [zigpy.quirks] Fail because input cluster mismatch on at least one endpoint
```

The result is the divider number is not applied anymore

After modifying the quirk:

```

2024-10-21 11:20:43.017 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LiXee ZLinky_TIC (00:15:8d:00:05:be:83:47)
2024-10-21 11:20:43.017 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lixee.zlinky.ZLinkyTICFWV12'>
2024-10-21 11:20:43.017 DEBUG (MainThread) [zigpy.quirks] Device matches filter signature - device ieee[00:15:8d:00:05:be:83:47]: filter signature[{'models_info': [('LiXee', 'ZLinky_TIC')], 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.METER_INTERFACE: 83>, 'input_clusters': [0, 1, 1, 3, 10, 1794, 2817, 2820, 61184, 65382], 'output_clusters': [25, 61184]}, 242: {'profile_id': 41440, 'device_type': <DeviceType.PROXY_BASIC: 97>, 'input_clusters': [33], 'output_clusters': [33]}}}]
2024-10-21 11:20:43.017 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:15:8d:00:05:be:83:47: <class 'zhaquirks.lixee.zlinky.ZLinkyTICFWV12'>
```

I have a rough patch:

```diff
diff --git a/zhaquirks/lixee/zlinky.py b/zhaquirks/lixee/zlinky.py
index 3891fe0..57c7a0d 100644
--- a/zhaquirks/lixee/zlinky.py
+++ b/zhaquirks/lixee/zlinky.py
@@ -11,6 +11,7 @@ from zigpy.zcl.clusters.general import (
     Identify,
     Ota,
     PowerConfiguration,
+    Time,
 )
 from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement, MeterIdentification
 from zigpy.zcl.clusters.smartenergy import Metering
@@ -157,13 +158,16 @@ class ZLinkyTIC(CustomDevice):
                 DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                 INPUT_CLUSTERS: [
                     Basic.cluster_id,
+                    PowerConfiguration.cluster_id,
                     Identify.cluster_id,
+                    Time.cluster_id,
                     Metering.cluster_id,
                     MeterIdentification.cluster_id,
                     ElectricalMeasurement.cluster_id,
+                    0xef00,
                     ZLinkyTICManufacturerCluster.cluster_id,
                 ],
-                OUTPUT_CLUSTERS: [Ota.cluster_id],
+                OUTPUT_CLUSTERS: [Ota.cluster_id, 0xef00],
             },
             242: {
                 PROFILE_ID: zgp.PROFILE_ID,
@@ -182,6 +186,7 @@ class ZLinkyTIC(CustomDevice):
                     Basic.cluster_id,
                     PowerConfiguration.cluster_id,
                     Identify.cluster_id,
+                    Time.cluster_id,
                     ZLinkyTICMetering,
                     MeterIdentification.cluster_id,
                     ElectricalMeasurement.cluster_id,

```

I will clean this up and make a PR shortly. I also noticed it's not a quirk V2. I might update that later.

blauret, 2024-10-21:
Although, there's something weird in the log. The signature matches ZLinkyTICFWV12, which as a result of the patch a twice the PowerConfiguration cluster.

PR #3448 (2024-10-23): Add ZLinky variant for firmware v14.0
## Proposed change

Add Signature for the [latest Firmware release](https://github.com/fairecasoimeme/Zlinky_TIC/releases/tag/v14.0) of the ZLinky_TIC

## Additional information

The latest firmware add new clusters which prevent it from being associated with the quirks and result in incorrect data being pushed in HA.

Fixes #3365 

## Checklist

- [X] The changes are tested and work correctly
- [X] `pre-commit` checks pass / the code has been formatted using Black
- [X] Tests have been added to verify that the new code works
closed 2024-10-23 by TheJulianJES