zigpy/zha-device-handlers #3370: [Device Support Request] Aqara Smart Radiator Thermostat E1

GamerClassN7, 2024-09-19:
### Problem description

Device cna be only calibrated over ZigBee not manualy this functionality is missing to my awernease

### Solution description

Z2M has sthis functionality would be possible to ge correct zigbeecommand there and implement it in to the ZHA 

### Screenshots/Video

[image]
since it canot be calibrated  valve return problem 
[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 127,
    "maximum_incoming_transfer_size": 100,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 100,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0201",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019",
        "0x0201",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.airrtc.agl001",
  "class": "zhaquirks.xiaomi.aqara.thermostat_agl001.AGL001"
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

GamerClassN7, 2024-09-19:
from what i fount in Z2M config it should be this command but i am unable to found it un "Manage Zigbee Device"
[image]

puddly, 2024-09-19:
It looks like the device has a manufacturer-specific attribute `0x0270` with type `uint8_t` that controls calibration. Since it's manufacturer-specific, there's no way for ZHA to know that it exists, so you should manually add it to the quirk so it can be exposed.

closed 2025-08-11 by github-actions[bot]